OpenIPC/firmware #920: Compiling hi3516dv300 Error

zhengchunxian-ai, 2023-07-21:
Run  ./Building.sh hi3516dv300_ lite
Error reporting ”curl: (33) HTTP server doesn't seem to support byte ranges. Cannot resume.“
The same applies to multiple operations

Nuts-Underline, 2023-09-15:
can be network problem (github coonection for downloading stuff)
what URL (must be near in logs)

zerog2k, 2023-11-15:
```
jens@mars:~/build/openipc-firmware$ ./building.sh BOARD=hi3518ev100
Starting OpenIPC builder.
Building OpenIPC for BOARD=hi3518ev100

  SoC: BOARD=hi3518ev100
Board: BOARD=hi3518ev100_lite

Found cache directory.
Found existing Buildroot directory.
Found existing Buildroot downloads directory.
Copying Buildroot downloads to cache directory ...
cp -rvf buildroot-2023.02.1/dl/* /tmp/buildroot_dl
'buildroot-2023.02.1/dl/buildroot-2023.02.1.tar.gz' -> '/tmp/buildroot_dl/buildroot-2023.02.1.tar.gz'
Done.

Cleaning source directory.
make distclean
Done.

Downloading Buildroot sources to cache directory ...
curl --continue-at - --output /tmp/buildroot_dl/buildroot-2023.02.1.tar.gz https://buildroot.org/downloads/buildroot-2023.02.1.tar.gz
** Resuming transfer from byte position 7181664
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (33) HTTP server doesn't seem to support byte ranges. Cannot resume.
```

It seems buildroot.org (or whatever mirrors they are currently using) don't support byte range requests, so cannot handle resuming a transfer.

Nuts-Underline, 2023-11-16:
as fact ./building.sh without parameters **loves** to _make distclean_ and reload ### **all** stuff from internet
workaround for fast rebuild is _./building.sh BOARD=hi3518ev100 build_ or even _make BOARD=hi3518ev100 build_ 
but this way there can be some problem with removing packages

PR #1143 (2023-11-16): fix build curl byte range error
fixes #920
```
Downloading Buildroot sources to cache directory ...
curl --continue-at - --output /tmp/buildroot_dl/buildroot-2023.02.1.tar.gz https://buildroot.org/downloads/buildroot-2023.02.1.tar.gz
** Resuming transfer from byte position 7181664
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (33) HTTP server doesn't seem to support byte ranges. Cannot resume.
```
se [... cut]
closed 2023-11-16 by FlyRouter