OpenIPC/firmware #922: Missing includes and Makefile.param in sony_imx335_2L

assimilat, 2023-07-22:
Attempting a build for hi3518ev300_ultimate via the documented methods, initially got this error:
```
make[3]: Entering directory '/tmp/firmware/output/build/hisilicon-opensdk/libraries/sensor/hi3516ev200/sony_imx335_2L'
Makefile:6: ../../../../../../Makefile.param: No such file or directory
make[3]: *** No rule to make target '../../../../../../Makefile.param'.  Stop.
```

I removed this from the Makefile in the sony_imx335_2L directory and tried again:

```
ifeq ($(PARAM_FILE), )
    PARAM_FILE:=../../../../../../Makefile.param
    include $(PARAM_FILE)
endif

ifeq ($(MPP_BUILD), y)
ifeq ($(CBB_PARAM_FILE), )
    CBB_PARAM_FILE:=../../../../../Makefile.param
    include $(CBB_PARAM_FILE)
endif
endif
```
That resulted in errors for missing includes for sony_imx335_2L.
I checked for the existence of sony_imx335_2L's includes via this:

```
for i in $(grep include /tmp/firmware/output/build/hisilicon-opensdk/libraries/sensor/hi3516ev200/sony_imx335_2L/* | grep "[\#]" | cut -f2 -d'#' | grep "[\"]" | cut -f2 -d'"');
 do
     if [ -z "$(find . -type f | grep ${i})" ];
     then
         echo $i;
    fi;
done
```

Which returned these missing includes:

```
hi_comm_sns.h
hi_comm_video.h
hi_sns_ctrl.h
mpi_ae.h
mpi_awb.h
hi_comm_video.h
hi_sns_ctrl.h
gpioi2c_ex.h
hi_i2c.h
```

liamdiprose, 2023-07-22:
Thank you for your support. 

Running the script without any params:
```
./building.sh
Starting OpenIPC builder.
Usage: ./building.sh <variant>
Variants: v83x_lite v83x_ultimate s2l22m_lite s2l33m_lite s3l_lite ak3916ev300_lite ak3918ev300_lite fh8833v100_lite fh8852v100_lite fh8852v200_lite fh8852v210_lite fh8856v100_lite fh8856v200_lite fh8856v210_lite fh8858v200_lite fh8858v210_lite gk7102_lite gk7102s_lite gk7202v300_lite gk7205v200_fpv gk7205v200_lite gk7205v200_lte gk7205v200_original gk7205v200_ultimate gk7205v210_lite gk7205v300_fpv gk7205v300_lite gk7205v300_lte gk7205v300_ultimate gk7605v100_lite gm8135_lite gm8136_lite hi3516av100_lite hi3516av100_ultimate hi3516av200_lite hi3516av200_ultimate hi3516av300_lite hi3516cv100_lite hi3516cv200_lite hi3516cv300_lite hi3516cv300_mini hi3516cv300_ultimate hi3516cv500_lite hi3516dv100_lite hi3516dv100_ultimate hi3516dv200_lite hi3516dv300_lite hi3516ev100_lite hi3516ev200_fpv hi3516ev200_lite hi3516ev200_lte hi3516ev200_ultimate hi3516ev300_dev hi3516ev300_fpv hi3516ev300_glibc hi3516ev300_lite hi3516ev300_lte hi3516ev300_ultimate hi3518cv100_lite hi3518ev100_lite hi3518ev200_lite hi3518ev200_mini hi3518ev200_ultimate hi3518ev201_lite hi3518ev300_lite hi3518ev300_ultimate hi3519v101_lite hi3520dv200_lite hi3536cv100_lite hi3536dv100_fpv hi3536dv100_lite t10_lite t20_lite t21_lite t30_lite t31_fpv t31_lite t31_ultimate t40_lite t40_ultimate nt98562_lite nt98566_lite rv1109_lite rv1126_lite msc313e_lite msc316dc_lite msc316dm_lite ssc30kq_lite ssc325_lite ssc333_lite ssc335de_lite ssc335_initramfs ssc335_lite ssc337de_lite ssc337de_ultimate ssc337_lite ssc338q_initramfs ssc338q_lite ssc338q_ultimate dm36x_lite xm510_lite xm530_lite xm550_lite
```

And I select gk7205v300_lite by running the script again with the correct param: 
```
./building.sh gk7205v300_lite
```

Unfortunately this gives the same result as above.

cronyx, 2023-07-22:
> Thank you for your support.
> 
> Running the script without any params:
> 
> ```
> ./building.sh
> Starting OpenIPC builder.
> Usage: ./building.sh <variant>
> Variants: v83x_lite v83x_ultimate s2l22m_lite s2l33m_lite s3l_lite ak3916ev300_lite ak3918ev300_lite fh8833v100_lite fh8852v100_lite fh8852v200_lite fh8852v210_lite fh8856v100_lite fh8856v200_lite fh8856v210_lite fh8858v200_lite fh8858v210_lite gk7102_lite gk7102s_lite gk7202v300_lite gk7205v200_fpv gk7205v200_lite gk7205v200_lte gk7205v200_original gk7205v200_ultimate gk7205v210_lite gk7205v300_fpv gk7205v300_lite gk7205v300_lte gk7205v300_ultimate gk7605v100_lite gm8135_lite gm8136_lite hi3516av100_lite hi3516av100_ultimate hi3516av200_lite hi3516av200_ultimate hi3516av300_lite hi3516cv100_lite hi3516cv200_lite hi3516cv300_lite hi3516cv300_mini hi3516cv300_ultimate hi3516cv500_lite hi3516dv100_lite hi3516dv100_ultimate hi3516dv200_lite hi3516dv300_lite hi3516ev100_lite hi3516ev200_fpv hi3516ev200_lite hi3516ev200_lte hi3516ev200_ultimate hi3516ev300_dev hi3516ev300_fpv hi3516ev300_glibc hi3516ev300_lite hi3516ev300_lte hi3516ev300_ultimate hi3518cv100_lite hi3518ev100_lite hi3518ev200_lite hi3518ev200_mini hi3518ev200_ultimate hi3518ev201_lite hi3518ev300_lite hi3518ev300_ultimate hi3519v101_lite hi3520dv200_lite hi3536cv100_lite hi3536dv100_fpv hi3536dv100_lite t10_lite t20_lite t21_lite t30_lite t31_fpv t31_lite t31_ultimate t40_lite t40_ultimate nt98562_lite nt98566_lite rv1109_lite rv1126_lite msc313e_lite msc316dc_lite msc316dm_lite ssc30kq_lite ssc325_lite ssc333_lite ssc335de_lite ssc335_initramfs ssc335_lite ssc337de_lite ssc337de_ultimate ssc337_lite ssc338q_initramfs ssc338q_lite ssc338q_ultimate dm36x_lite xm510_lite xm530_lite xm550_lite
> ```
> 
> And I select gk7205v300_lite by running the script again:
> 
> ```
> ./building.sh gk7205v300_lite
> ```
> 
> Unfortunately this gives the same result as above.

Need to install fzf and try again:

`sudo apt install fzf`

widgetii, 2023-07-22:
Just checked run `./building.sh gk7205v300_lite` against master, everything is ok on my machine

```
- uImage: [1827KB/2048KB]
- rootfs.squashfs: [4996KB/5120KB]
- Build time: 42:30
renamed './output/images/uImage' -> './output/images/uImage.gk7205v300'
renamed './output/images/rootfs.squashfs' -> './output/images/rootfs.squashfs.gk7205v300'
renamed './output/images/rootfs.cpio' -> './output/images/rootfs.gk7205v300.cpio'
renamed './output/images/rootfs.tar' -> './output/images/rootfs.gk7205v300.tar'

The start-stop times
Sat Jul 22 01:51:13 PM MSK 2023
Sat Jul 22 02:33:43 PM MSK 2023
```

viktorxda, 2023-07-22:
Try the following:
```
cd firmware
make deps
make all
```

liamdiprose, 2023-07-23:
Thanks for the `make deps` tip. I'm not on a Debian derivative, but I'm sure I have all the packages installed in that command now. I'm still getting the same error as before:  `Makefile.param: no such file or directory`. 

Where does this file originate? I cannot find it mentioned anywhere else.

viktorxda, 2023-07-23:
I looks like a problem with the makefile, but it only occurs when I directly try to build sony_imx335_2L.
```
Makefile:6: ../../../../../../Makefile.param: No such file or directory
make: *** No rule to make target '../../../../../../Makefile.param'.  Stop.
```
Maybe something is different on your distriubtion as it tries to explicitly build this driver.
You can try to search for `sony_imx335_2L` on the output directory and delete the folder, after that try to resume the build.

widgetii, 2023-07-23:
> I'm not on a Debian derivative,

What's your distro?

liamdiprose, 2023-07-24:
I'm using NixOS.

<details>
  <summary><code>shell.nix</code></summary>
  
  ```nix
  { buildFHSUserEnv }:
  
  (buildFHSUserEnv {
    name = "openipc-br-env";
    targetPkgs = pkgs: (with pkgs; [
      automake 
      # autotools-dev 
      bc 
      # build-essential 
      gcc
      cpio
      curl 
      file 
      fzf 
      git 
      ncurses.dev 
      libtool 
      lzop 
      gnumake 
      rsync 
      unzip 
      wget
      perl
    ]);
  
    profile = ''
      export M="gk7205v300_lite"
    '';
  
    runScript = "bash";
  }).env
  ```
</details>

liamdiprose, 2023-07-24:
Removing the `sony_imx335_2L` folder does fix the problem. So I wonder if a missing dependency was causing the script to "fall-through".

`make BOARD=$M all` works now, I have a rootfs.squashfs waiting for my camera when it arrives :smiley:

widgetii, 2023-07-24:
I suggest using Debian or Ubuntu-base distro for compiling code (or fix compilation by yourself). Nobody in our team uses NixOS currently

viktorxda, 2023-07-24:
I added a possible fix: https://github.com/OpenIPC/openhisilicon/pull/15

widgetii, 2023-07-25:
> I added a possible fix: [OpenIPC/openhisilicon#15](https://github.com/OpenIPC/openhisilicon/pull/15)

Merged

liamdiprose, 2023-07-25:
```
make[1]: Leaving directory '/home/liam/projects/firmware/output/buildroot-2023.02.2'
- uImage: [1827KB/2048KB]
- rootfs.squashfs: [4996KB/5120KB]
- Build time: 21:59
renamed './output/images/uImage' -> './output/images/uImage.gk7205v300'
renamed './output/images/rootfs.squashfs' -> './output/images/rootfs.squashfs.gk7205v300'
renamed './output/images/rootfs.cpio' -> './output/images/rootfs.gk7205v300.cpio'
renamed './output/images/rootfs.tar' -> './output/images/rootfs.gk7205v300.tar'

The start-stop times
Wed Jul 26 08:19:10 AM NZST 2023
Wed Jul 26 08:41:10 AM NZST 2023
```

I can confirm `sony_imx335_2L` is back. This build is now working on NixOS (using the shell.nix above). Thank you everyone!

closed 2023-07-26 by viktorxda