OpenIPC/firmware #933: Image quality, Wi-Fi and GPIO issues - Hi3518EV200 w/ SC2235

Flynt56, 2023-08-02:
Thank you for an amazing project! I've successfully flashed 1 out of 4 of my [TonTon TC55B-2MP](https://www.amazon.com/Tonton-Wireless-Waterproof-Security-Recording/dp/B0811C1K5Z?th=1) cameras. The original firmware had been using the N1 protocol (I don't really know what it is) with a Flash Player web frontend for video playback. Now the flashed camera is able to transmit a useful RTSP stream.

Unfortunately, the setup is not there yet. The flashed camera has the following issues:
- Noisy camera image
  - This probably requires a lot more fiddling with the ISP's settings and IQ profile
  - The original firmware loaded a _custom_ ISP configuration but it's entries don't exactly match up to any of the IQ profiles found in `/etc/sensors/iq` - here's the profile file: [isp_cfg_sc2235_1.txt](https://github.com/OpenIPC/firmware/files/12241205/isp_cfg_sc2235_1.txt)
- Slower startup times
  - There are a lot of weird `dmesg` log entries for `himci_control_cclk`, `himci_wait_cmd` and `himci_set_cclk`
- Wi-Fi chip present on the board but not available in OpenIPC
  - The original firmware logged the following:
```txt
RTL871X: module init start
RTL871X: rtl8188fu v4.3.23.4_19019.20160811
RTL871X: build time: Dec  9 2016 17:23:04
RTL871X: hal_com_config_channel_plan chplan:0x23
RTL871X: rtw_ndev_init(wlan0) if1 mac_addr=9c:a3:a9:5f:89:da
RTL871X: rtw_ndev_init(wlan1) if2 mac_addr=9e:a3:a9:5f:89:da
usbcore: registered new interface driver rtl8188fu
RTL871X: module init ret=0
```
- It has an IR LED array, IR cut and PIR motion sensor
  - Is there a way to get the correct GPIO pin number for these things so that Majestic can work with them?
  - How would I know if I got them right?
- It can get pretty hot
  - OpenIPC's temperature ticker can sometimes show values over 80°C. Is this normal?

Luckily, I've managed to obtain root access to the remaining cameras via Telnet. The remaining cameras are all on stock firmware so there's probably a way to get the OpenIPC one working right by sharing _the good stuff_. I'm not afraid to dig into it more myself but I'm out of ideas and lost, hence the GitHub issue.  Once the OpenIPC camera is at least on par with the rest, I'll flash the rest.

Again, many thanks for an amazing project.

# Snapshot A - Original firmware
[image]

# Snapshot B - OpenIPC
[image]

# ipctool
```yaml
chip:
  vendor: HiSilicon
  model: 3518EV200
board:
  vendor: OpenIPC
  version: 2.3.07.31
  possible-IR-cut-GPIO: 1,2
ethernet:
  mac: "00:9a:32:8f:f8:c0"
  u-mdio-phyaddr: 1
  phy-id: 0x001cc816
  d-mdio-phyaddr: 0
rom:
- type: nor
  block: 64K
  partitions:
    - name: boot
      size: 0x40000
      sha1: 3be1edb5
    - name: env
      size: 0x10000
      sha1: a96b408d
      contains:
        - name: uboot-env
          offset: 0x0
    - name: kernel
      size: 0x200000
      sha1: 4e1407c1
    - name: rootfs
      size: 0x500000
      path: /,squashfs
      sha1: 7a9d7c58
    - name: rootfs_data
      size: 0xb0000
      path: /overlay,jffs2,rw
  size: 8M
  addr-mode: 3-byte
ram:
  total: 64M
  media: 32M
firmware:
  u-boot: "2010.06 (Nov 14 2022 - 19:23:46)"
  kernel: "4.9.37 (Mon Jul 31 01:09:25 UTC 2023)"
  toolchain: buildroot-gcc-12.2.0
  sdk: "Hi3518EV200_MPP_V1.0.5.0 B060 Release (Mar 21 2018, 08:56:43)"
sensors:
- vendor: SmartSens
  model: SC2235
  control:
    bus: 0
    type: i2c
    addr: 0x60
  data:
    type: DC
  clock: 24MHz
```

# ipctool reginfo
```txt
muxctrl_reg0 0x200f0000 0x1 GPIO0_4 [SENSOR_CLK]
muxctrl_reg1 0x200f0004 0 [SENSOR_RSTN] GPIO0_5
muxctrl_reg2 0x200f0008 0x1 GPIO0_6 [FLASH_TRIG] SFC_EMMC_BOOT_MODE SPI1_CSN1 VI_VS_BT1120
muxctrl_reg3 0x200f000c 0 [GPIO0_7] SHUTTER_TRIG SFC_DEVICE_MODE VI_HS_BT1120
muxctrl_reg4 0x200f0010 0x1 GPIO2_0 [RMII_CLK] VO_CLK SDIO1_CCLK_OUT
muxctrl_reg5 0x200f0014 0x1 GPIO2_1 [RMII_TX_EN] VO_VS SDIO1_CARD_DETECT
muxctrl_reg6 0x200f0018 0x1 GPIO2_2 [RMII_TXD0] VO_DATA5 SDIO1_CWPR
muxctrl_reg7 0x200f001c 0x1 GPIO2_3 [RMII_TXD1] VO_DE SDIO1_CDATA1
muxctrl_reg8 0x200f0020 0x1 GPIO2_4 [RMII_RX_DV] VO_DATA7 SDIO1_CDATA0
muxctrl_reg9 0x200f0024 0x1 GPIO2_5 [RMII_RXD0] VO_DATA2 SDIO1_CDATA3
muxctrl_reg10 0x200f0028 0x1 GPIO2_6 [RMII_RXD1] VO_DATA3 SDIO1_CCMD
muxctrl_reg11 0x200f002c 0x1 GPIO2_7 [EPHY_RST] BOOT_SEL VO_HS SDIO1_CARD_POWER_EN
muxctrl_reg12 0x200f0030 0 [GPIO0_3] SPI1_CSN1 VO_DATA0
muxctrl_reg13 0x200f0034 0x1 GPIO3_0 [EPHY_CLK] VO_DATA1 SDIO1_CDATA2
muxctrl_reg14 0x200f0038 0x1 GPIO3_1 [MDCK] BOOTROM_SEL VO_DATA6
muxctrl_reg15 0x200f003c 0x1 GPIO3_2 [MDIO] VO_DATA4
muxctrl_reg16 0x200f0040 0x2 GPIO3_3 SPI0_SCLK [I2C0_SCL]
muxctrl_reg17 0x200f0044 0x2 GPIO3_4 SPI0_SDO [I2C0_SDA]
muxctrl_reg18 0x200f0048 0 [GPIO3_5] SPI0_SDI
muxctrl_reg19 0x200f004c 0 [GPIO3_6] SPI0_CSN
muxctrl_reg20 0x200f0050 0 [GPIO3_7] SPI1_SCLK I2C1_SCL
muxctrl_reg21 0x200f0054 0 [GPIO4_0] SPI1_SDO I2C1_SDA
muxctrl_reg22 0x200f0058 0 [GPIO4_1] SPI1_SDI
muxctrl_reg23 0x200f005c 0 [GPIO4_2] SPI1_CSN0
muxctrl_reg24 0x200f0060 0 [GPIO4_3] I2C2_SDA
muxctrl_reg25 0x200f0064 0 [GPIO4_4] I2C2_SCL
muxctrl_reg26 0x200f0068 0x1 GPIO4_5 [USB_OVRCUR]
muxctrl_reg27 0x200f006c 0x1 GPIO4_6 [USB_PWREN]
muxctrl_reg28 0x200f0070 0 [GPIO0_0] IR_IN TEMPER_DQ
muxctrl_reg29 0x200f0074 0 [GPIO0_1] TEMPER_DQ
muxctrl_reg30 0x200f0078 0 [GPIO0_2] TEMPER_DQ
muxctrl_reg31 0x200f007c 0x1 GPIO1_0 [VI_DATA13] I2S_BCLK_TX PWM0
muxctrl_reg32 0x200f0080 0x1 GPIO1_1 [VI_DATA10] I2S_SD_TX UART1_TXD
muxctrl_reg33 0x200f0084 0x1 GPIO1_2 [VI_DATA12] I2S_MCLK UART1_CTSN
muxctrl_reg34 0x200f0088 0x1 GPIO1_3 [VI_DATA11] I2S_WS_TX UART2_RXD
muxctrl_reg35 0x200f008c 0x2 GPIO1_4 VI_DATA15 [VI_VS_SEN] I2S_WS_RX UART1_RXD
muxctrl_reg36 0x200f0090 0x2 GPIO1_5 VI_DATA14 [VI_HS_SEN] I2S_BCLK_RX UART1_RTSN
muxctrl_reg37 0x200f0094 0x1 GPIO1_6 [VI_DATA9] I2S_SD_RX UART2_TXD
muxctrl_reg38 0x200f0098 0x1 GPIO1_7 [SDIO0_CARD_POWER_EN]
muxctrl_reg39 0x200f009c 0x1 GPIO4_7 [SDIO0_CARD_DETECT]
muxctrl_reg40 0x200f00a0 0x1 GPIO5_0 [SDIO0_CWPR]
muxctrl_reg41 0x200f00a4 0x1 GPIO5_1 [SDIO0_CCLK_OUT]
muxctrl_reg42 0x200f00a8 0x1 GPIO5_2 [SDIO0_CCMD]
muxctrl_reg43 0x200f00ac 0x1 GPIO5_3 [SDIO0_CDATA0]
muxctrl_reg44 0x200f00b0 0x1 TEST_CLK [SDIO0_CDATA1] GPIO5_4
muxctrl_reg45 0x200f00b4 0x1 GPIO5_5 [SDIO0_CDATA2]
muxctrl_reg46 0x200f00b8 0x1 GPIO5_6 [SDIO0_CDATA3]
muxctrl_reg47 0x200f00bc 0 [GPIO5_7] EMMC_DATA6 I2S_SD_TX UART1_RTSN
muxctrl_reg48 0x200f00c0 0 [GPIO6_0] EMMC_DATA5 I2S_WS_TX UART1_RXD
muxctrl_reg49 0x200f00c4 0 [GPIO6_1] EMMC_DATA7 I2S_MCLK UART1_CTSN
muxctrl_reg50 0x200f00c8 0 [GPIO6_2] EMMC_DS I2S_SD_RX UART1_TXD
muxctrl_reg51 0x200f00cc 0 [GPIO6_3] EMMC_DATA1 UART2_RXD
muxctrl_reg52 0x200f00d0 0 [GPIO6_4] EMMC_DATA2 I2S_BCLK_TX UART2_TXD
muxctrl_reg53 0x200f00d4 0 [GPIO6_5] JTAG_TRSTN SPI1_CSN1 I2S_MCLK
muxctrl_reg54 0x200f00d8 0 [GPIO6_6] JTAG_TCK SPI1_SCLK I2S_WS_TX I2C1_SCL
muxctrl_reg55 0x200f00dc 0 [GPIO6_7] JTAG_TMS SPI1_CSN0 I2S_SD_TX
muxctrl_reg56 0x200f00e0 0 [GPIO7_0] JTAG_TDO SPI1_SDO I2S_SD_RX I2C1_SDA
muxctrl_reg57 0x200f00e4 0 [GPIO7_1] JTAG_TDI SPI1_SDI I2S_BCLK_TX
muxctrl_reg58 0x200f00e8 0 [PMC_PWM] GPIO7_2 PWM0
muxctrl_reg59 0x200f00ec 0 [PWM1] GPIO7_3
muxctrl_reg60 0x200f00f0 0 [PWM2] GPIO7_4 SPI1_CSN1
muxctrl_reg61 0x200f00f4 0 [PWM3] GPIO7_5
muxctrl_reg62 0x200f00f8 0 [SAR_ADC_CH0] GPIO7_6
muxctrl_reg63 0x200f00fc 0 [SAR_ADC_CH1] GPIO7_7
muxctrl_reg64 0x200f0100 0 [SAR_ADC_CH2] GPIO8_0
muxctrl_reg65 0x200f0104 0 [SAR_ADC_CH3] GPIO8_1
```

# dmesg
It's pretty long so I attached it instead of inlining it.
[dmesg.txt](https://github.com/OpenIPC/firmware/files/12240933/dmesg.txt)

# logread
Also pretty long.
[logread.txt](https://github.com/OpenIPC/firmware/files/12241266/logread.txt)

alandpearson, 2024-12-18:
Hi - I found your thread and I have the same  himci_control_cclk, himci_wait_cmd and himci_set_cclk issues

I then found this thread which may help with boot speed : https://github.com/OpenIPC/firmware/issues/868

weedplant, 2025-04-21:
You need to change the ini file of the sensor.. I have the same board (but other brand). It cost me a while to figure things out. I had the same bad video quality. I also grabbed the *.so file of the orignal firmware and replaced the openIPC driver with that one. 
The ini file I changed the 30fps to 25fps (this board doesn't support 30fps, that's why it ffs up your image).

Haven't tempered with the WiFi yet. But that's my next issue

Flynt56, 2025-04-30:
> You need to change the ini file of the sensor.. I have the same board (but other brand). It cost me a while to figure things out. I had the same bad video quality. I also grabbed the *.so file of [...]

Thanks for the heads up! I'll give that a shot once I get back to fiddling with the cameras.

> Haven't tempered with the WiFi yet. But that's my next thing on the to-do-list. Or have you managed to get it to work?

Nope, not yet, sorry.