OpenIPC/firmware #934: Wi-Fi mt7601u error found in Ingenic T31x lite

IPC-Probe, 2023-08-04:
I would like to test the Wifi-feature with T31x lite firmware that with both Ethernet and wireless connection.

first i connected MT7601U module D+/D-/GND to the Ingenic Snipe board micro usb port and providing 3,3V DC. 
with the putty i set the command - 
"**fw_setenv wlandev mt7601-t31-camhi**" 
and assigned wlanssid and wlanpass.

however after reboot, the system does not get any IP address in wlan0. and i dont find mt7601u with my mobile phone WiFi.
i type lsusb and read "Bus 001 Device 006: ID 148f:7601". 
so the mt7601u is detected by the system, right?

from the dmesg log.
it shows - < **ERROR!!! MT7601 E2PROM: WRONG VERSION 0xc, should be 9** >

and i found below from the file system
**/lib/firmware/mediatek/mt7601u.bin
/lib/firmware/mt7601u.bin**

please advise the correct procedures to add the WiFi mt7601u in T31x firmware.
any suggestion or extra information is helpful.

many thanks.

themactep, 2023-08-04:
please provide more of dmesg log, not just the error line.

IPC-Probe, 2023-08-04:
Hi Paul,

Thanks for your quick reply.
I probably send the whole log next Monday.

IPC-Probe, 2023-08-08:
hi Paul.
below is the full bootup log. it seems that the IP cam can get the IP address from router now. (10.10.100.84)
**however it is not able to enter the web page or stream the video.** 

[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 3.10.14__isvp_swan_1.0__ (builder@citadel) (buildroot-gcc-12.2.0) #2 PREEMPT Mon Jul 24 22:29:42 MSK 2023
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU0 RESET ERROR PC:080C101D
[    0.000000] CPU0 revision is: 00d00100 (Ingenic Xburst)
[    0.000000] FPU revision is: 00b70000
[    0.000000] cgu_get_rate, parent = 1392000000, rate = 0, m = 129, n = 255, reg val = 0x081000ff
[    0.000000] cgu_get_rate, parent = 1392000000, rate = 0, m = 129, n = 255, reg val = 0x081000ff
[    0.000000] CCLK:1392MHz L2CLK:696Mhz H0CLK:200MHz H2CLK:200Mhz PCLK:100Mhz
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 0053a000 @ 00010000 (usable)
[    0.000000]  memory: 00036000 @ 0054a000 (usable after init)
[    0.000000] User-defined physical RAM map:
[    0.000000]  memory: 04000000 @ 00000000 (usable)
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x00000000-0x03ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x00000000-0x03ffffff]
[    0.000000] On node 0 totalpages: 16384
[    0.000000] free_area_init_node: node 0, pgdat 80548030, node_mem_map 81000000
[    0.000000]   Normal zone: 128 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 16384 pages, LIFO batch:3
[    0.000000] Primary instruction cache 32kB, 8-way, VIPT, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 8-way, VIPT, no aliases, linesize 32 bytes
[    0.000000] pls check processor_id[0x00d00100],sc_jz not support!
[    0.000000] MIPS secondary cache 128kB, 8-way, linesize 32 bytes.
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping off.  Total pages: 16256
[    0.000000] Kernel command line: mem=64M rmem=64M@0x4000000 console=ttyS1,115200n8 panic=20 root=/dev/mtdblock3 rootfstype=squashfs init=/init mtdparts=jz_sfc:256k(boot),64k(env),3072k(kernel),10240k(rootfs),-(rootfs_data)
[    0.000000] PID hash table entries: 256 (order: -2, 1024 bytes)
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Memory: 58896k/65536k available (4006k kernel code, 6640k reserved, 1342k data, 216k init, 0k highmem)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] Preemptible hierarchical RCU implementation.
[    0.000000] NR_IRQS:358
[    0.000000] clockevents_config_and_register success.
[    0.000014] Calibrating delay loop... 1391.00 BogoMIPS (lpj=6955008)
[    0.087831] pid_max: default: 32768 minimum: 301
[    0.092688] Mount-cache hash table entries: 512
[    0.097593] Initializing cgroup subsys debug
[    0.101849] Initializing cgroup subsys freezer
[    0.107258] devtmpfs: initialized
[    0.111998] regulator-dummy: no parameters
[    0.116368] NET: Registered protocol family 16
[    0.123843] Skipping MSC1_PB GPIO setup
[    0.127722] Skipping WYZE GPIO setup
[    0.131269] Skipping WYZE GPIO setup
[    0.135480] set gpio strength: 32-2
[    0.135489] set gpio strength: 33-2set gpio strength: 34-2
[    0.135498] set gpio strength: 35-2set gpio strength: 36-2
[    0.135506] set gpio strength: 37-2set gpio pull: 59-90
[    0.153316] bio: create slab <bio-0> at 0
[    0.163126] jz-dma jz-dma: JZ SoC DMA initialized
[    0.169389] usbcore: registered new interface driver usbfs
[    0.175366] usbcore: registered new interface driver hub
[    0.180879] usbcore: registered new device driver usb
[    0.186426]  (null): set:249  hold:250 dev=100000000 h=500 l=500
[    0.194111] Switching to clocksource jz_clocksource
[    0.201326] dwc2 otg probe start
[    0.201353] jz-dwc2 jz-dwc2: cgu clk gate get error
[... 179 lines cut ...]
[   12.228365] Rcv Wcid(1) AddBAReq
[   12.228381] Start Seq = 00000000
root@convoy-t31:~# 
root@convoy-t31:~# 
root@convoy-t31:~# ifconfig
lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:32 errors:0 dropped:0 overruns:0 frame:0
          TX packets:32 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:2496 (2.4 KiB)  TX bytes:2496 (2.4 KiB)

wlan0     Link encap:Ethernet  HWaddr 2C:C3:E6:0A:75:95  
          inet addr:10.10.100.84  Bcast:10.10.100.255  Mask:255.255.255.0
          inet6 addr: fe80::2ec3:e6ff:fe0a:7595/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:2118 errors:0 dropped:20 overruns:0 frame:0
          TX packets:40 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:470989 (459.9 KiB)  TX bytes:7376 (7.2 KiB)
root@convoy-t31:~# lsusb
Bus 001 Device 002: ID 148f:7601
Bus 001 Device 001: ID 1d6b:0002

themactep, 2023-08-09:
Seems you have the driver loaded, you have the interface, even some networking. Is the IP address something you would expect from your dhcp? Disable autoload of wifi interface and run ifup -v wlan0 manually in console. Check for errors.

closed 2023-11-19 by viktorxda