OpenIPC/firmware #935: Adding Hi3881 SDI wifi module

jppw, 2023-08-05:
Hello,

im trying to add a Hi3881 SDI wifi module which is in my wifi camera with a T31zx.

Extracted all needed files from stock firmware.
[wlan-config.zip](https://github.com/OpenIPC/firmware/files/12268324/wlan-config.zip)

cp /mnt/mmcblk0p1/hi3881.ko /lib/modules/3.10.14__isvp_swan_1.0__/extra/

sh /mnt/mmcblk0p1/wifi.sh

```
cd /sys/class/gpio/
echo 6 > export
echo out > gpio6/direction
echo 1 > gpio6/value
sleep 1
echo 0 > gpio6/value 
sleep 1
echo 1 > gpio6/value
```

modprobe cfg80211

insmod /lib/modules/3.10.14__isvp_swan_1.0__/extra/hi3881.ko

dmesg:

[  185.366752] cfg80211: Calling CRDA to update world regulatory domain
[  188.679495] [VERSION]:Hi3881V100R001C00SPC021 2020-10-14 20:25:00
[  188.682138] oam_main_init SUCCESSFULLY!
[  188.682153] wal_customize_init SUCCESSFULLY!
[  188.682183] oal_main_init SUCCESSFULLY
[  188.684666] frw_main_init SUCCESSFULLY!
[  188.684678] hi_wifi_plat_init SUCCESSFULLY
[  188.800884] mmc1: host doesn't support card's voltages
[  188.800898] mmc1: error -22 whilst initialising SDIO card
[  188.819606] mmc1: host doesn't support card's voltages
[  188.819617] mmc1: error -22 whilst initialising MMC card
[  188.941668] mmc1: host doesn't support card's voltages
[  188.941680] mmc1: error -22 whilst initialising SDIO card
[  188.960117] mmc1: host doesn't support card's voltages
[  188.960129] mmc1: error -22 whilst initialising MMC card
[  189.081672] mmc1: host doesn't support card's voltages
[  189.081684] mmc1: error -22 whilst initialising SDIO card
[  189.100099] mmc1: host doesn't support card's voltages
[  189.100110] mmc1: error -22 whilst initialising MMC card
[  198.679081] [ERROR]:oal_sdio_host.c:oal_sdio_func_probe:2257:sdio enum timeout, reason[probe timeout]
[  198.679081]
[  198.679355] [ERROR]:1646:hcc_host_init:: sdio_probe failed!
[  198.679367] ------------[ cut here ]------------
[  198.679382] WARNING: at fs/sysfs/file.c:497 sysfs_attr_ns+0x34/0xc4()
[  198.679387] sysfs: kobject  without dirent
[  198.679393] Modules linked in: hi3881(O+) cfg80211 tx_isp_t31(O) avpu(O) sinfo(O) vfat fat f2fs
[  198.679421] CPU: 0 PID: 781 Comm: insmod Tainted: G           O 3.10.14__isvp_swan_1.0__ #2
[  198.679427] Stack : 00000000 00000000 00000000 00000000 8058316a 0000004f 10001c01 80580000
[  198.679427]    8049e04c 8205aa58 80508bc7 80582904 0000030d c0a60000 00000001 00000001
[  198.679427]    8007dc08 803efd9c 80511398 800371fc 804a3cb8 00000003 8049f9bc 82721b8c
[  198.679427]    00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  198.679427]    00000000 00000000 00000000 00000000 00000000 00000000 00000000 82721b18
[  198.679427]    ...
[  198.679505] Call Trace:
[  198.679516] [<8001fdf4>] show_stack+0x58/0x70
[  198.679531] [<803efd08>] warn_slowpath_common+0x78/0xa8
[  198.679543] [<80037350>] warn_slowpath_fmt+0x2c/0x38
[  198.679552] [<801326d8>] sysfs_attr_ns+0x34/0xc4
[  198.679561] [<80132b80>] sysfs_remove_file+0x20/0x54
[  198.679572] [<801f5f74>] bus_remove_driver+0x38/0xd4
[  198.679740] [<c09b097c>] oal_sdio_func_remove+0x1c/0x34 [hi3881]
[  198.680057] [<c0a34624>] hcc_host_init+0x6d8/0x7b0 [hi3881]
[... 88 lines cut ...]
[  198.685065] Modules linked in: hi3881(O+) cfg80211 tx_isp_t31(O) avpu(O) sinfo(O) vfat fat f2fs
[  198.685088] Process insmod (pid: 781, threadinfo=82720000, task=8205a880, tls=772bdd78)
[  198.685093] Stack : 804a9bb4 82721c3c 00000000 801326d8 00000000 82bc1834 00000000 803ec8c4
[  198.685093]    82bc1800 80132b80 00000000 00000000 80548020 80548020 82bc1834 8205a880
[  198.685093]    00000000 c0a653f8 c0a653f8 801f5fb0 00000000 00000000 00000000 00000000
[  198.685093]    82a70400 c0a361a8 803efd64 c09b097c 803efd64 82a7043c 00000000 c0a60000
[  198.685093]    00000001 c0a34624 00000002 00000000 80510000 80039a38 00000000 00000000
[  198.685093]    ...
[  198.685171] Call Trace:
[  198.685179] [<803ec764>] klist_put+0x1c/0xc0
[  198.685189] [<803ec8c4>] klist_remove+0x7c/0xbc
[  198.685199] [<801f5fb0>] bus_remove_driver+0x74/0xd4
[  198.685363] [<c09b097c>] oal_sdio_func_remove+0x1c/0x34 [hi3881]
[  198.685679] [<c0a34624>] hcc_host_init+0x6d8/0x7b0 [hi3881]
[  198.685997] [<c0a361a8>] hcc_hmac_init+0x24/0x14c [hi3881]
[  198.686315] [<c0a2c6a4>] hi_wifi_host_init+0x28/0x374 [hi3881]
[  198.686633] [<c0a21610>] init_module+0x1c8/0x308 [hi3881]
[  198.686797] [<800105a4>] do_one_initcall+0x1a4/0x1b0
[  198.686806] [<80080630>] load_module+0x1a6c/0x1d10
[  198.686815] [<80080a2c>] SyS_finit_module+0x6c/0x84
[  198.686826] [<80022adc>] stack_done+0x20/0x44
[  198.686831]
[  198.686835]
[  198.686835] Code: 8c820000  2403fffe  00431024 <8c51000c> 8f820014  00808025  24420001  10a0000b  af820014
[  198.686861] ---[ end trace 11c9cb8fddf3f62d ]---

jppw, 2023-08-05:
I added modprobe gpio but with the same result.

Maybe here is the problem

[  176.309961] [VERSION]:Hi3881V100R001C00SPC021 2020-10-14 20:25:00
[  176.312638] oam_main_init SUCCESSFULLY!
[  176.312654] wal_customize_init SUCCESSFULLY!
[  176.312682] oal_main_init SUCCESSFULLY
[  176.315182] frw_main_init SUCCESSFULLY!
[  176.315194] hi_wifi_plat_init SUCCESSFULLY
[  176.430892] mmc1: host doesn't support card's voltages
[  176.430906] mmc1: error -22 whilst initialising SDIO card
[  176.449612] mmc1: host doesn't support card's voltages
[  176.449624] mmc1: error -22 whilst initialising MMC card
[  176.571676] mmc1: host doesn't support card's voltages
[  176.571688] mmc1: error -22 whilst initialising SDIO card
[  176.590106] mmc1: host doesn't support card's voltages
[  176.590118] mmc1: error -22 whilst initialising MMC card
[  176.711674] mmc1: host doesn't support card's voltages
[  176.711685] mmc1: error -22 whilst initialising SDIO card
[  176.730104] mmc1: host doesn't support card's voltages
[  176.730115] mmc1: error -22 whilst initialising MMC card
[  186.309095] [ERROR]:oal_sdio_host.c:oal_sdio_func_probe:2257:sdio enum timeout, reason[probe timeout]
[  186.309095]
[  186.309405] [ERROR]:1646:hcc_host_init:: sdio_probe failed!

jppw, 2023-08-05:
root@openipc-t31:/mnt/mmcblk0p1# lsmod
Module                  Size  Used by    Tainted: G
hi3881               1041553  1 [permanent]
cfg80211              260210  1 hi3881
gpio                    1475  0
tx_isp_t31            986934  0
avpu                   16998  0
sinfo                  17696  0
vfat                    9665  1
fat                    56394  1 vfat
f2fs                  132628  0

daoudeddy, 2024-11-06:
did you find the cause of the problem?