OpenIPC/firmware #941: mt7601u kernel module missing in hi3518ev100-lite firmware

databus23, 2023-08-09:
Hi,

I just managed to flash the `openipc-hi3518ev100-lite-16mb.bin` firmware onto a Hikam S5 Camera but have no networking.
The initial problem seem to be that the firmware has no `mt7601u` that can be loaded to support the usb wireless adapter.

```
root@openipc-hi3518ev100:/lib/modules# lsusb
Bus 001 Device 001: ID 1d6b:0002
Bus 002 Device 001: ID 1d6b:0001
Bus 001 Device 002: ID 148f:7601
root@openipc-hi3518ev100:/lib/modules# modprobe mt7601u
modprobe: module mt7601u not found in modules.dep
```

The firmware for the wireless adapter is there though:
```
root@openipc-hi3518ev100:/lib/firmware# find .
.
./mediatek
./mediatek/mt7601u.bin
./mt7601u.bin
./rtlwifi
./rtlwifi/rtl8188eufw.bin
```

Any ideas how I can fix that?

kakou-fr, 2023-10-03:
Plz, I need it too for IPCAM camera

zerog2k, 2023-11-08:
suspicion here that the hi3518ev100 kernel version (3.0.8) is too old for the wifi drivers, like in #665 
this seems to indicate MT7601U and RTL8188EU support are not added until 3.10 kernel
https://www.kernelconfig.io/config_mt7601u?q=&kernelversion=3.0.8&arch=arm

so i guess the question is how to bump hi3518ev100 support to later kernel, like at least 3.18, which some of the other hisilicon families have support for in openipc
Not sure how much effort this will be, or if it's possible

closed 2026-10-16 by openipc-ai