OpenIPC/firmware #948: firmware hi3518ev300 2.3.08.12 no wireless connection

Goo15, 2023-08-13:
Since firmware hi3518ev300 2.3.08.12 it is no possible to connect to network wireless.
In the boot section from kernel "modul rt8188 not found FAIL" is displayed.
With firmware hi3518ev300 2.3.07.15 also no wireless problems.

viktorxda, 2023-08-14:
The driver was updated to a newer version and the module was renamed because of that:
`rtl8188fu` -> 8188fu`
Try to adjust the wlan0 interface file with the new name.

Goo15, 2023-08-14:
Now I change the the Name to 8188fu in the interface file, but it still not works. The boot section of kernel displayed following errors:
Starting network: Successfully initialized wpa_supplicant
rfkill: Cannot open RFKILL control device
udhcpc: started, v1.36.0
udhcpc: broadcasting discover
udhcpc: read error: Network is down, reopening socket
udhcpc: bind: No such device
FAIL

The interface file looks like that: (with my SSID and password)
auto wlan0
iface wlan0 inet dhcp
    pre-up modprobe 8188fu
    pre-up wpa_passphrase "SSID" "password" >/tmp/wpa_supplicant.conf
    pre-up sed -i '2i \\tscan_ssid=1' /tmp/wpa_supplicant.conf
    pre-up sleep 3
    pre-up wpa_supplicant -B -D nl80211 -i wlan0 -c/tmp/wpa_supplicant.conf
    post-down killall -q wpa_supplicant

Goo15, 2023-08-15:
Thank you for help. Now it works. It was my fault by the wlan0 configuration.

closed 2023-08-15 by viktorxda