OpenIPC/firmware #953: f2fs-tools are not present

southskies, 2023-08-16:
F2fs-tools is listed on the wiki [List of OpenIPC Buildroot packages](https://github.com/OpenIPC/wiki/blob/master/en/dev-buildroot-packages.md), but I cannot find the binaries. 

[image]

```
cat /etc/os-release
NAME=Buildroot
VERSION=-gcac1781
ID=buildroot
VERSION_ID=2023.02.2
PRETTY_NAME="Buildroot 2023.02.2"
OPENIPC_VERSION=2.3.08.02
GITHUB_VERSION="master+cac1781, 2023-08-02"
BUILD_OPTION=ultimate
TIME_STAMP=1690991432
```

Thanks for all the hard work!

widgetii, 2023-08-16:
I believe they simply don't fit into lite firmware due to its size, but nobody cares to add them into ultimate package yet.

As usual: feel free to make a PR

southskies, 2023-08-16:
@widgetii I appreciate the quick reply. Thanks!

closed 2023-08-16 by southskies