OpenIPC/firmware #954: Firmware hi3815ev300 Version 15.08.2023 no wlan connection

Goo15, 2023-08-16:
There is no possible to connect to the wlan network.
Follwing error displayed in the bootsection of kernel.

Starting network: Successfully initialized wpa_supplicant
Could not read interface wlan0 flags: No such device
nl80211: Driver does not support authentication/association or connect commands
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Could not read interface wlan0 flags: No such device
wlan0: Failed to initialize driver interface
wlan0: CTRL-EVENT-DSCP-POLICY clear_all
FAIL

The wlan0 configuration from the interfaces file lokks like that:

auto wlan0
iface wlan0 inet dhcp
    pre-up modprobe 8188fu
    pre-up wpa_passphrase "SSID" "Password" >/tmp/wpa_supplicant.conf
    pre-up sed -i '2i \\tscan_ssid=1' /tmp/wpa_supplicant.conf
    pre-up sleep 3
    pre-up wpa_supplicant -B -D nl80211 -i wlan0 -c/tmp/wpa_supplicant.conf
    post-down killall -q wpa_supplicant

What can be the reason and what can I do to eliminate the error?

doiiido, 2023-08-24:
Your device may need to set a GPIO to turn on your wifi module, my hi3518ev200 needed these lines to set the GPIO 9:
pre-up echo 9 > /sys/class/gpio/export
pre-up echo out > /sys/class/gpio/gpio9/direction
pre-up echo 0 > /sys/class/gpio/gpio9/value

Goo15, 2023-08-27:
With 2.3.08.25-ultimate no problems with wlan0 configuration.

themactep, 2023-11-05:
Please close the ticket if the issue is resolved.

closed 2026-09-04 by openipc-ai