OpenIPC/firmware #968: Increasing the popularity of OpenIPC

aha999, 2023-08-22:
I think that OpenIPC definitely deserves more recognition in the world of "shady" video surveillance systems.

**I opened this issue, so anybody can give ideas on how to increase OpenIPC popularity, get more funding, and to bring it to the masses.**

Currently you have to be half-hacker-developer. You have to buy a good enough supported camera, flashing hardware, tear apart the camera hardware, flash the firmware etc... Which is too much for most "users" imo. 

My inputs:

- OpenIPC should offer different tiers of pre-flashed cameras, for example 4 different models, until sold out,
- buy hardware from the factory, flash it and put a margin on it,
- A new connected web store could be started, where OpenIPC hardware, pre-flashed cameras, kits etc. could be sold.
- With more income, more could be targeted towards organization growth, support, recognizability and success of openIPC in general

Ultraexpert, 2023-08-25:
What about an affiliate link with a manufacturer?

wwebers, 2023-09-04:
What about a label on devices like "Ready for OpenIPC", which hints the buyer that the manufacturer actively supports reflashing the device (and maybe even provides the original firmware to be able to go back)?

Junnicutt, 2023-09-13:
Yeah you really don't need to sell anything. For now just highlight a few common cameras that have everything working. Like the recommended hardware but for finished product. I've been reading about openipc for a bit but I still haven't tried cause the only thing I know that will work is some cameras on Alibaba. And like most Americans I've never ordered anything from banggood or Alibaba. 
 If I had seen that the wyze cam v3 was fully supported I'd have ran to the store and picked one up the first day I read about openipc fpv. Even just highlighting a camera on amazon would be just as good. This is probably only useful for Americans probably but you could highlight easy to get cameras in different regions. Just mentioning those cams on the homepage would likely boost usage significantly.

closed 2026-09-04 by openipc-ai