OpenIPC/firmware #993: Other interesting  board GK72205v300 but need some workaround to flash

Nuts-Underline, 2023-09-14:
My board teken here https://aliexpress.com/item/1005005471371172.html
Looking good: GK7205V300+IMX335 +MicroSD slot + place for soldering WiFi (usb) + serial on pins, WiFi module is option
as fact 16MB and 2 UART (one prepared for RS485 but not ADM485 not soldered)
Problems: 
1) Non standart firmware p2p, it even loading XiongMai's OpenIPC frimware and showing like its flashing, but no any change at end
2) Unknown root password
3) Uboot non standart and having password ### Please input uboot password: ###

 [WORKAROUND, found here](https://libc0607.wordpress.com/2021/10/23/%E5%90%9B%E6%AD%A3-t31-%E6%91%84%E5%83%8F%E5%A4%B4%E6%A8%A1%E7%BB%84%E7%9A%84%E7%AE%80%E5%8D%95%E8%A7%82%E5%AF%9F%E8%AE%B0%E5%BD%95%EF%BC%88%E4%B8%80%EF%BC%89/)
(SPI reader + decompiling) also this human known here as @libc0607
1) geting connection  to onboard UART  (connector 1,25 or just wiring)
2) having _### Please input uboot password: ###_ - ubootpwd 
answer ->  **HI2105CHIP** <---
3) _sf lock 0_ for flashing not working, need **sf unlock 0x0 0x1000000** for it

Pimmetje, 2023-12-02:
Here a similar camera https://aliexpress.com/item/1005003340814126.html
Ill try if the sensor chip will work in a few days.

Pimmetje, 2023-12-05:
I tried that one i think. How did u get the password from the binary?

Nuts-Underline, 2023-12-06:
> I tried that one i think. How did u get the password from the binary?

all i know  - in my 1st post here

Pimmetje, 2024-01-14:
At the end i used burn

csvke, 2024-04-14:
Anyone here got this board working with a WiFi module? I was told by a seller in taobao to use this module https://item.taobao.com/item.htm?id=744788471484 which is a altobeam atbm6032ls

I have gone as far as building openipc firmware with the driver as .ko so I can `ismod` or `modprobe` but someone on the telegram channel mentioned the need of toggling a GPIO pin. Any info is appreciated

Nuts-Underline, 2024-04-16:
@csvke there can be a tons of problem
1) AltoBeam atbm603x need **special** driver, its included to only 2 configs of OpenIPC (_BR2_PACKAGE_ATBM603X=Y_) for other boad, not for our board
2)3.3V power to module can be measured with some kind of multimeter. If no  voltage - need to dig ito original firmware for GPIOs nubmers

taobao asking for login and not showing any picture. Better to add some pics directrly  here

csvke, 2024-04-16:
> @csvke there can be a tons of problem
>
> 1. AltoBeam atbm603x need **special** driver, its included to only 2 configs of OpenIPC (_BR2_PACKAGE_ATBM603X=Y_) for other boad, not for our board
>    2)3.3V [...]

[image]

I have this which is a *JFTech IVG-6GS* that has a Goke GK7205v300 SoC and the USB are exposed with a 12-pin flex cable connector. Taobao is selling a WiFi+SD card extension board which uses *Altobeam atbm6032ls* WiFi chip. I cannot get the usb working as it seems to need a trigger to pull the pin 7 of the flex cable connector to pass power through to the WiFi chip. I have observed that pin 7 at low until Openipc boots into loading the kernel. Like what you said, I have tried to pull all the GPIO from 1-79 and it still does not work. My educated guess is that the memory register that control the WIFI_PDN pin that wasn’t registered in Openipc but I carelessly didn’t backup the original firmware to investigate… if someone has the original firmware I maybe able to use ipctool to possibly investigate the GPIO registers.

One interesting thing is that when I put this WiFi board to a GK7205v210 board, ‘lsusb’ did recognise the usb device without the need to trigger any GPIO / memory register. But it only proves the WiFi board is functioning, as I cannot get the driver into a Openipc build, not too mention Openipc do not have a GK7205v210 ultimate firmware (despite the camera board comes with 16mb NOR) and the GK7205v210 has been identified as a GK7205v200 so I have to build a GK7205v200 firmware to load into a GK7205v210 (but still cannot get ATBM603X driver into the firmware)

And the second point is that no matter what I do I can’t build the BR2_PACKAGE_ATBM603X=Y into the actual firmware like you said. 

I guess it’s too much hurdles to get this WiFi board working, I may settle with buying a different WiFi chip such as a MT7601 and a RTL8188EU to test.

at least I tried.

Nuts-Underline, 2024-04-17:
> I carelessly didn’t backup the original firmware to investigate
Its most problem, i thinking it can be better to open special issue for (owners of) this board

funkypozzy, 2024-07-05:
> ![image](https://private-user-images.githubusercontent.com/4078574/322928572-dbdabdf0-d26f-4caa-b1b0-b5947ae5c5a5.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

I have the same camera GK7205V300+IMX335 +MicroSD slot and Wifi Module and I'm not able to enable the wifi. Here the link the original firmware backup: https://drive.google.com/file/d/1LBMsVoHrfXAH2sqlfecHysbKXadlIioG/view?usp=drive_link.
Hope this can help. Let me know...

widgetii, 2026-08-16:
Closing this one — the flashing question it was opened for is solved, and the thread has since become two unrelated conversations.

### The original question: flashing this locked GK7205v300

Solved by @Nuts-Underline in the first post, plus the password in the comments:

- **Bootloader password `HI2105CHIP`** works on this Xiongmai-derived camhipro build. Also try `#Ux6@9V&4_Rz`, which is confirmed across a wider range of Goke boards including gk7205v300 (OpenIPC/firmware#1494).
- @Pimmetje got there with `burn`.

For anyone arriving now: `burn` has been replaced by [defib](https://github.com/OpenIPC/defib), and gk7205v300 works in the browser with nothing to install — <https://openipc.github.io/defib/>. That bypasses the vendor bootloader entirely, so the password becomes optional, and the **Dump Flash** button backs up the stock firmware before you touch anything.

Two things from this thread that turned out to be general, both now documented:

- **"It shows like it's flashing, but no change at the end"** — exactly the symptom in your first post. The vendor bootloader arms the flash's status-register protection (`WPS=1`), and U-Boot's `sf lock 0` doesn't clear that kind of lock, so erase and write report success and do nothing. Root cause filed as OpenIPC/u-boot-gk7205v200#18; `defib agent flash` handles it correctly today.
- Both passwords are now on the [wiki bootloader page](https://github.com/OpenIPC/wiki/blob/master/en/help-uboot.md) instead of only in this thread.

### The WiFi conversation belongs in #1486

The AltoBeam ATBM6032LS / IF-SD-WIFI discussion between @csvke, @funkypozzy, @thedeadliestcatch and @Nuts-Underline has its own issue — **#1486, "IF-SD-WIFI (ex. IPC_38x38-WIFI-if v1.02) support and reverse engineering"** — which is still open and is the right place for it. @thedeadliestcatch's flash dump in particular deserves to be linked from there rather than buried at the end of a thread about bootloader passwords.

If the WiFi work stalls for lack of a driver, please continue in #1486 so it stays findable.

Thanks @Nuts-Underline for the original writeup — the `HI2105CHIP` password has helped a lot of people since.

closed 2026-08-16 by widgetii