OpenIPC/firmware #994: Cann't  set fps greater than 25 with GK7205V300 + IMX335

Lance0724, 2023-09-16:
GK7205V300 + IMX335 fps cannot be set to any value except 25 unexpectedly, 
otherwise the ground will not receive any packets

widgetii, 2023-09-16:
According to SoC's datasheet:

[image]

What resolution do you use?

Lance0724, 2023-09-16:
> According to SoC's datasheet:
> 
> [image]
> What resolution do you use?

I only tried 1920X1080@fps30 and 1280X720@fps30, and it didn't work

video0：
  enabled: true
  bitrate: 8192
  codec: h265
  size: 1920x1080
  # fps:30
  rcMode: cbr
  gopSize: 0.2

widgetii, 2023-09-16:
Make sure that your sensor also works on such fps (it could be figured out from the logs)

Lance0724, 2023-09-16:
Where can I see the log? Is it standout after putty connection?

Lance0724, 2023-09-20:
There are some new discoveries that when I enable the low latency mode, 
I can change the fps greater than 25 (e.g fps30, fps60), 
but there is no noticeable change in latency

widgetii, 2023-09-20:
There is no way to have low latency in GK7205V300 + IMX335, the only board recommended is described in wiki

closed 2023-09-20 by widgetii

Lance0724, 2023-09-20:
> There is no way to have low latency in GK7205V300 + IMX335, the only board recommended is described in wiki

Recommended board is IVG-G2S?  Right ?