OpenIPC/firmware #995: gk7205v210: usb error

lazyan, 2023-09-17:
Successfully installed the lite firmware for subject. Logs shown the normal starting up of xhci drivers, but finally the following messages are shown:
`xhci-hcd xhci-hcd.0.auto: xHCI host not responding to stop endpoint command.
xhci-hcd xhci-hcd.0.auto: Assuming host is dying, halting host.
xhci-hcd xhci-hcd.0.auto: HC died; cleaning up
usb usb1-port1: couldn't allocate usb_device`
What could cause the problem?

Any plans to make fpv drivers for this SOC?

lazyan, 2023-09-23:
Just to clarify the issue. Tried to connect different usb devices, to camera but they are not appear on lsusb.

lazyan, 2023-09-29:
lsusb:
`Bus 001 Device 001: ID 1d6b:0002
Bus 002 Device 001: ID 1d6b:0003`
dmesg:
`Booting Linux on physical CPU 0x0
Linux version 4.9.37 (runner@fv-az354-870) (buildroot-gcc-12.2.0) #1 Sat Sep 16 18:57:56 UTC 2023
CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c53c7d
CPU: div instructions available: patching division code
CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
OF: fdt:Machine model: Goke GK7205V200 DEMO Board
cma zone is not set!
Memory policy: Data cache writeback
On node 0 totalpages: 8192
free_area_init_node: node 0, pgdat c05132d8, node_mem_map c1fbb000
  Normal zone: 64 pages used for memmap
  Normal zone: 0 pages reserved
  Normal zone: 8192 pages, LIFO batch:0
CPU: All CPU(s) started in SVC mode.
pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
pcpu-alloc: [0] 0
Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 8128
Kernel command line: mem=32M console=ttyAMA0,115200 panic=20 rootfstype=squashfs root=/dev/mtdblock3 init=/init mtdparts=sfc:256k(boot),64k(env),2048k(kernel),5120k(rootfs),-(rootfs_data) mmz_allocator=gk
PID hash table entries: 128 (order: -3, 512 bytes)
Dentry cache hash table entries: 4096 (order: 2, 16384 bytes)
Inode-cache hash table entries: 2048 (order: 1, 8192 bytes)
Memory: 27004K/32768K available (3893K kernel code, 150K rwdata, 916K rodata, 200K init, 181K bss, 5764K reserved, 0K cma-reserved)
Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
    vmalloc : 0xc2800000 - 0xff800000   ( 976 MB)
    lowmem  : 0xc0000000 - 0xc2000000   (  32 MB)
    modules : 0xbf000000 - 0xc0000000   (  16 MB)
      .text : 0xc0008000 - 0xc03d56a8   (3894 kB)
      .init : 0xc04bc000 - 0xc04ee000   ( 200 kB)
      .data : 0xc04ee000 - 0xc0513ae0   ( 151 kB)
       .bss : 0xc0515000 - 0xc0542468   ( 182 kB)
SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
NR_IRQS:16 nr_irqs:16 16
Gic dist init...
arm_arch_timer: Architected cp15 timer(s) running at 50.00MHz (phys).
clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xb8812736b, max_idle_ns: 440795202655 ns
sched_clock: 56 bits at 50MHz, resolution 20ns, wraps every 4398046511100ns
Switching to timer-based delay loop, resolution 20ns
clocksource: arm,sp804: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 637086815595 ns
Console: colour dummy device 80x30
Calibrating delay loop (skipped), value calculated using timer frequency.. 100.00 BogoMIPS (lpj=500000)
pid_max: default: 32768 minimum: 301
Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
CPU: Testing write buffer coherency: ok
Setting up static identity map for 0x40008200 - 0x40008258
devtmpfs: initialized
VFP support v0.3: implementor 41 architecture 2 part 30 variant 7 rev 5
clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
futex hash table entries: 256 (order: -1, 3072 bytes)
pinctrl core: initialized pinctrl subsystem
NET: Registered protocol family 16
DMA: preallocated 256 KiB pool for atomic coherent allocations
Serial: AMBA PL011 UART driver
12040000.uart: ttyAMA0 at MMIO 0x12040000 (irq = 21, base_baud = 0) is a PL011 rev2
console [ttyAMA0] enabled
12041000.uart: ttyAMA1 at MMIO 0x12041000 (irq = 22, base_baud = 0) is a PL011 rev2
12042000.uart: ttyAMA2 at MMIO 0x12042000 (irq = 23, base_baud = 0) is a PL011 rev2
ssp-pl022 12070000.spi: ARM PL022 driver, device ID: 0x00041022
ssp-pl022 12070000.spi: mapped registers from 0x12070000 to c286f000
usbcore: registered new interface driver usbfs
usbcore: registered new interface driver hub
usbcore: registered new device driver usb
clocksource: Switched to clocksource arch_sys_counter
NET: Registered protocol family 2
[... 177 lines cut ...]
mtdpartsubi=setenv mtdparts nand:256k(boot),768k(wtf),3072k(kernel),-(ubi)
netmask=255.255.255.0
nfsroot=/srv/nfs/gk7205v200
osmem=32M
serverip=192.168.0.22
setargs=setenv bootargs mem=${osmem} console=ttyAMA0,115200 panic=20 root=/dev/mtdblock3 rootfstype=squashfs init=/init mtdparts=${mtdparts} ${extras}
setnand=run mtdpartsubi; setenv bootcmd ${bootcmdubi}; saveenv; reset
setnor16m=run mtdpartsnor16m; setenv bootcmd ${bootcmdnor}; saveenv; reset
setnor8m=run mtdpartsnor8m; setenv bootcmd ${bootcmdnor}; saveenv; reset
stderr=serial
stdin=serial
stdout=serial
totalmem=64M
uknand=mw.b ${baseaddr} ff 1000000; tftpboot ${baseaddr} uImage.${soc} && nand erase 0x100000 0x300000; nand write ${baseaddr} 0x100000 0x300000
uknor16m=mw.b ${baseaddr} ff 1000000; tftpboot ${baseaddr} uImage.${soc} && sf probe 0; sf erase 0x50000 0x300000; sf write ${baseaddr} 0x50000 ${filesize}
uknor8m=mw.b ${baseaddr} ff 1000000; tftpboot ${baseaddr} uImage.${soc} && sf probe 0; sf erase 0x50000 0x200000; sf write ${baseaddr} 0x50000 ${filesize}
urnand=mw.b ${baseaddr} ff 1000000; tftpboot ${baseaddr} rootfs.ubi.${soc} && nand erase 0x400000 0x7c00000; nand write ${baseaddr} 0x400000 ${filesize}
urnor16m=mw.b ${baseaddr} ff 1000000; tftpboot ${baseaddr} rootfs.squashfs.${soc} && sf probe 0; sf erase 0x350000 0xa00000; sf write ${baseaddr} 0x350000 ${filesize}
urnor8m=mw.b ${baseaddr} ff 1000000; tftpboot ${baseaddr} rootfs.squashfs.${soc} && sf probe 0; sf erase 0x250000 0x500000; sf write ${baseaddr} 0x250000 ${filesize}
vendor=goke
verify=n
bootargs=mem=32M console=ttyAMA0,115200 panic=20 rootfstype=squashfs root=/dev/mtdblock3 init=/init mtdparts=sfc:256k(boot),64k(env),2048k(kernel),5120k(rootfs),-(rootfs_data) mmz_allocator=gk
sensor=imx307
bootdelay=0
soc=gk7205v210`

themactep, 2023-09-29:
please use proper markdown markup

piaosn, 2023-12-23:
I encountered the same problem. Then I moved to Hi3516ev300.

RoboSchmied, 2024-01-19:
USB is working on gk7205v210. I flashed [gk7205v210-fpv-8mb.bin](https://openipc.org/cameras/vendors/goke/socs/gk7205v210/download_full_image?flash_size=8&flash_type=nor&fw_release=fpv).

Please check your wiring!

RoboSchmied, 2024-01-20:
Usb is working on gk7205v210.
```
root@openipc-gk7205v210:/etc/sensors# dmesg | grep -i usb
usbcore: registered new interface driver usbfs
usbcore: registered new interface driver hub
usbcore: registered new device driver usb
xhci-hcd xhci-hcd.0.auto: new USB bus registered, assigned bus number 1
hub 1-0:1.0: USB hub found
xhci-hcd xhci-hcd.0.auto: new USB bus registered, assigned bus number 2
usb usb2: We don't know the algorithms for LPM for this host, disabling LPM.
hub 2-0:1.0: USB hub found
hub 2-0:1.0: hub can't support USB3.0
usb 1-1: new high-speed USB device number 2 using xhci-hcd
usbcore: registered new interface driver rtl88xxau
usb 1-1: USB disconnect, device number 2
usb 1-1: new high-speed USB device number 3 using xhci-hcd
usb 1-1: USB disconnect, device number 3
usb 1-1: new high-speed USB device number 4 using xhci-hcd
usb 1-1: USB disconnect, device number 4
usb 1-1: new high-speed USB device number 5 using xhci-hcd
root@openipc-gk7205v210:/etc/sensors# lsusb 
Bus 001 Device 001: ID 1d6b:0002
Bus 001 Device 005: ID 0bda:8812
Bus 002 Device 001: ID 1d6b:0003
```

Please try flashing a recent image.

Rothammel, 2025-02-21:
> I encountered the same problem. Then I moved to Hi3516ev300.

what is your wiring diagram, I am looking for USB pins

closed 2026-10-26 by openipc-ai