OpenIPC/firmware #997: Can't find a valid spi flash chip after flashing

CezWAT, 2023-09-19:
I just flashed Hi3518ErbcV100 with full OpenIPC Firmware image via TFTP. Just after writing to memory it is said to reset and "run setnor8m". I can't run it, the response is:

> OpenIPC # run setnor8m
> setenv - set environment variables
> 
> Usage:
> setenv name value ...
>     - set environment variable 'name' to 'value ...'
> setenv name
>     - delete environment variable 'name'
> Saving Environment to SPI Flash...
> Check spi flash controller v350... Found
> Spi(cs1) ID: 0xF8 0x32 0x17 0xF8 0x32 0x17
> Spi(cs1): find unrecognized spi flash.
> Can't find a valid spi flash chip.
> Spi(cs1) ID: 0xF8 0x32 0x17 0xF8 0x32 0x17
> Spi(cs1): find unrecognized spi flash.
> Can't find a valid spi flash chip.
> Environment SPI flash not initialized
> resetting ...
> 
> U-Boot 2010.06 (Nov 14 2022 - 19:25:39)
> 
> Check spi flash controller v350... Found
> Spi(cs1) ID: 0xF8 0x32 0x17 0xF8 0x32 0x17
> Spi(cs1): find unrecognized spi flash.
> Can't find a valid spi flash chip.
> Spi(cs1) ID: 0xF8 0x32 0x17 0xF8 0x32 0x17
> Spi(cs1): find unrecognized spi flash.
> Can't find a valid spi flash chip.
> Check spi flash controller v350... Found
> Spi(cs1) ID: 0xF8 0x32 0x17 0xF8 0x32 0x17
> Spi(cs1): find unrecognized spi flash.
> Can't find a valid spi flash chip.
> Spi(cs1) ID: 0xF8 0x32 0x17 0xF8 0x32 0x17
> Spi(cs1): find unrecognized spi flash.
> Can't find a valid spi flash chip.
> *** Warning - bad CRC, using default environment
> 
> In:    serial
> Out:   serial
> Err:   serial
> RAM size: 64MB
> Hit any key to stop autoboot:  0

When I try to use TFTP it says:
> OpenIPC # tftp  
Hisilicon ETH net controler
MAC:   00-00-23-34-45-66
PHY not link.

What can I do about it?

themactep, 2023-09-19:
Is your flash memory chip from Fedelix? Can you post a photo? I think there is no support for this chip in our u-boot yet. It needs to be patched/added and recompilef, then flashed to your module, possibly with a programmer.

CezWAT, 2023-09-19:
No, it looks like I have DOSILICON FM25Q64A-1AI - SPI Quad, 64 Mb, 1st gen flash memory.

themactep, 2023-09-19:
I think these are two faces of the same Janus :)  Both use the same manufacturer ID F8h.

[FM25Q64A-Dosilicon.pdf](https://github.com/OpenIPC/firmware/files/12664795/FM25Q64A-Dosilicon.pdf)
[FM25Q64-Fidelix.pdf](https://github.com/OpenIPC/firmware/files/12664833/FM25Q64-Fidelix.pdf)

themactep, 2023-09-19:
Hi3518EV100 [uses](https://github.com/OpenIPC/firmware/blob/master/br-ext-chip-hisilicon/board/hi3516cv100/kernel/hi3518ev100.generic.config) U-Boot from [Hi3516CV100 family](https://github.com/OpenIPC/u-boot-hi3516cv100/).  You need to add parameters for your flash chip to U-Boot, somewhere [here](https://github.com/OpenIPC/u-boot-hi3516cv100/tree/master/drivers/mtd/spi).

Or just replace the chip with a supported one.

themactep, 2023-09-19:
`grep '.name =' *` in  u-boot-hi3516cv100/drivers/mtd/spi/ should give you a list of supported models.

CezWAT, 2023-10-04:
I just received S25FL064L which is the only memory that fit. After programming with CH341, I soldered it in place and found out it's not booting. The error is basically the same.
Things to do:
- check if IO3 / RESET# in S25FL064L is used properly as in FM25Q64A it has a different function (/HOLD(IO3)), if not then pull it from the board and keep the memory out of reset state,
- rewrite the memory and check it again.