OpenIPC/firmware #1001: Unable to use WI-FI in T30X

duchengyao, 2023-09-21:
How to use wifi in my device?

My wifi chip is:
```
SV6256P
TAC2309
1RA96B
```

Can't get wlan0 in `ifconfig -a`:
```
eth0      Link encap:Ethernet  HWaddr 1A:19:24:50:9A:DE  
          inet addr:192.168.2.10  Bcast:0.0.0.0  Mask:255.255.255.0
          inet6 addr: 240e:390:e07:6780:1819:24ff:fe50:9ade/64 Scope:Global
          inet6 addr: fe80::1819:24ff:fe50:9ade/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:1971 errors:0 dropped:0 overruns:0 frame:0
          TX packets:841 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:181465 (177.2 KiB)  TX bytes:241071 (235.4 KiB)

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:228 errors:0 dropped:0 overruns:0 frame:0
          TX packets:228 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:17816 (17.3 KiB)  TX bytes:17816 (17.3 KiB)

tunl0     Link encap:UNSPEC  HWaddr 00-00-00-00-31-39-00-32-00-00-00-00-00-00-00-00  
          NOARP  MTU:0  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
```

lsusb:
```
Bus 001 Device 001: ID 1d6b:0002
```

fw_printenv:
```
baseaddr=0x80600000
baudrate=115200
bootargs=mem=${osmem} rmem=${rmem} console=ttyS1,115200n8 panic=20 root=/dev/mtdblock3 rootfstype=squashfs init=/init mtdparts=${mtdparts} ${extras}
bootargsnfs=mem=${osmem} console=ttyS1,115200n8 panic=20 root=/dev/nfs rootfstype=nfs ip=${ipaddr}:::255.255.255.0::eth0 nfsroot=${serverip}:${nfsroot},v3,nolock rw ${extras}
bootargsubi=mem=${osmem} console=ttyS1,115200n8 panic=20 init=/init root=ubi0:rootfs rootfstype=ubifs ubi.mtd=3,2048 mtdparts=${mtdparts} ${extras}
bootcmd=setenv setargs setenv bootargs ${bootargs}; run setargs; sf probe 0; sf read ${baseaddr} 0x50000 0x300000; bootm ${baseaddr}
bootcmdnand=setenv setargs setenv bootargs ${bootargs}; run setargs; nand read ${baseaddr} 0x100000 0x300000; bootm ${baseaddr}
bootcmdnor=setenv setargs setenv bootargs ${bootargs}; run setargs; sf probe 0; sf read ${baseaddr} 0x50000 0x300000; bootm ${baseaddr}
bootcmdubi=setenv setargs setenv bootargs ${bootargsubi}; run setargs; nand read ${baseaddr} 0x100000 0x300000; bootm ${baseaddr}
bootdelay=1
bootnfs=setenv setargs setenv bootargs ${bootargsnfs}; run setargs; tftpboot ${baseaddr} uImage.${soc}; bootm ${baseaddr}
ethact=Jz4775-9161
gatewayip=192.168.1.1
ipaddr=192.168.1.10
loads_echo=1
mtdparts=jz_sfc:256k(boot),64k(env),2048k(kernel),5120k(rootfs),-(rootfs_data)
mtdpartsnand=setenv mtdparts jz_sfcnand:256k(boot),768k(wtf),3072k(kernel),10240k(rootfs),-(rootfs_data)
mtdpartsnor16m=setenv mtdparts jz_sfc:256k(boot),64k(env),3072k(kernel),10240k(rootfs),-(rootfs_data)
mtdpartsnor8m=setenv mtdparts jz_sfc:256k(boot),64k(env),2048k(kernel),5120k(rootfs),-(rootfs_data)
mtdpartsubi=setenv mtdparts jz_sfcnand:256k(boot),768k(wtf),3072k(kernel),-(ubi)
netmask=255.255.255.0
nfsroot=/srv/nfs/t31
osmem=64M
restore=n
rmem=64M@0x4000000
[... 168 lines cut ...]
[    0.653729] Freeing unused kernel memory: 216K (8054a000 - 80580000)
[    1.079082] jzmmc_v1.2 jzmmc_v1.2.0: card inserted, state=0
[    1.095525] jffs2: notice: (452) jffs2_build_xattr_subsystem: complete building xattr subsystem, 3 of xdatum (0 unchecked, 3 orphan) and 3 of xref (0 dead, 3 orphan) found.
[    1.507116] [resetd] GPIO undefined in /usr/sbin/resetd
[    1.739376] Bus Mode Reg after reset: 0x00020101, cnt=0
[    2.115049] name : i2c0 nr : 0
[    2.115067] wait stable.[289][cgu_cim]
[    2.199223] i2c i2c-0: i2c_jz_irq 441, I2C transfer error, ABORT interrupt
[    2.199237] i2c i2c-0: --I2C txabrt:
[    2.199246] i2c i2c-0: --I2C TXABRT[0]=I2C_TXABRT_ABRT_7B_ADDR_NOACK
[    2.199265] error: sensor_read,229 ret = -5
[    2.199271]  sensor_read: addr=0xa value = 0x0
[    2.199276] err sensor read addr = 0xa, value = 0x0
[    2.289222] i2c i2c-0: i2c_jz_irq 441, I2C transfer error, ABORT interrupt
[    2.289238] i2c i2c-0: --I2C txabrt:
[    2.289247] i2c i2c-0: --I2C TXABRT[0]=I2C_TXABRT_ABRT_7B_ADDR_NOACK
[    2.289265] error: sensor_read,229 ret = -5
[    2.289271]  sensor_read: addr=0x300a value = 0x0
[    2.289277] err sensor read addr = 0x300a, value = 0x0
[    2.319071] wait stable.[289][cgu_msc0]
[    2.339147] mmc0: error -1 whilst initialising SD card
[    2.379227] i2c i2c-0: i2c_jz_irq 441, I2C transfer error, ABORT interrupt
[    2.379239] i2c i2c-0: --I2C txabrt:
[    2.379247] i2c i2c-0: --I2C TXABRT[0]=I2C_TXABRT_ABRT_7B_ADDR_NOACK
```

themactep, 2023-09-21:
You will need to compile a driver for your WiFi module if it is not there.

gitgayhub, 2023-09-21:
openipc doesn't have driver for your wifi chip, you may try loading the wifi driver from your stock firmware, there's a chance that it will work

duchengyao, 2023-09-21:
Thanks. ^_^

themactep, 2023-09-21:
I have the driver source if you want to try to compile it.

duchengyao, 2023-09-21:
> I have the driver source if you want to try to compile it.

I really need the driver source, Thanks

themactep, 2023-09-21:
If you are on Telegram, ping me at @themactep there. I'll be near my desktop in anout an hour and send it to you.

duchengyao, 2023-09-26:
@gitgayhub 

Unable to load module from stock firmware.  Could you please tell me if I am doing it correctly?

```
root@openipc-t31:~# insmod ssv6x5x.ko 
insmod: can't insert 'ssv6x5x.ko': unknown symbol in module, or unknown parameter
```

themactep, 2023-09-26:
try
modprobe cfg80211
then the insmod

duchengyao, 2023-09-26:
not  work , the same error.

themactep, 2023-09-26:
Invalid module format. Did you cross compile it?

themactep, 2023-09-26:
Seems incompatible. Try to compile native mac80211 if it is not compiled, and if you are sure these functions are from that module.

sanyajorj, 2024-01-18:
Good afternoon, if possible for 8188fu driver T31L.

closed 2026-10-26 by openipc-ai