OpenIPC/firmware #1010: majstic crash when alarm activated

darkfuncat, 2023-09-28:
Hello,
I already posted this issue in majectic but I think this is most due to firmware part.
(see [https://github.com/OpenIPC/majestic/issues/133](url))
With : 
Camera board id: t31n (t31 family) Xiaomi MJSXJ03HL
OpenIPC version: 2.3.09.28-lite
installation method: UART

As soon as I activate the majestic/alarm "Enable motion detection" and reboot, majestic crash on startup :
Logs :

```
root@openipc-t31:~# majestic
21:42:05  <       majestic> [    main] main@123                      Majestic Lite for Ingenic, version master+fb5bea8, built on 2023-09-28
21:42:05  <       majestic> [app_conf] load_config@107               Using /etc/majestic.yaml as main configuration
21:42:05  <       majestic> [watchdog] watchdog_start@101            Detected jz Watchdog
21:42:05  <       majestic> [watchdog] watchdog_start@121            Watchdog timeout set to 10 seconds
21:42:05  <       majestic> [sns_conf] try_sensor_config@13          SENSOR=jxq03p
21:42:05  <       majestic> [sns_conf] load_sensor_config@44         Using /etc/sensor/jxq03p.yaml as sensor config
21:42:05  <       majestic> [sns_conf] parse_sensor_config@125       VI: jxq03p @ 0x40 [2304x1296]
21:42:05  <       majestic> [     sdk] start_sdk@1247                CPU T31-N, SDK SYSUTILS-1.1.6, model Swan, devID 5624000c
---- FPGA board is ready ----
  Board UID : 30AB6E51
  Board HW ID : 72000460
  Board rev.  : 5DE5A975
  Board date  : 20190326
-----------------------------
21:42:05  <       majestic> [     sdk] imp_system_init@438           ImpSystemInit success
21:42:05  <       majestic> [     sdk] start_sdk@1390                [0] H.265 2304x1296 25fps 4096Kbit [VB:1]
21:42:05  <       majestic> [     osd] init_osd@147                  OSD initialized
21:42:05  <       majestic> [     log] printf@268                    VBMCreatePool[335] poolid = 2 IMP_ALLOC failed size = 706560
21:42:05  <       majestic> [     sdk] imp_framesource_streamon@828  IMP_FrameSource_EnableChn(-1) error: 2
21:42:05  <       majestic> [     sdk] start_sdk@1474                ImpStreamOn failed
21:42:05  <       majestic> [    main] main@209                      Shutdown main thread
```

Any suggestion please ?
Thanks

widgetii, 2023-09-29:
It's not a crash, but an ordinary error when it cannot allocate memory for video buffers (due to lack of memory)

darkfuncat, 2023-10-02:
Thanks for your response.
I did some additionnal tests, and it seems not due to lack of memory but a problem with resolution.
Even if the camera has 2304x1296 capabilities, if I set 1920x1080 as video stram resolution majestic actually runs perfect.
And it has some free memory left.
For any reason that I dont know, majestic does not like 2304x1296 resolution.
This workaround degrades image quality significantly but now it is working and I can continue my tests....
Other people with kind of resolution can start majestic and alarm without problem ?

impala3525, 2023-12-31:
I have the same issue with a Goke7502v210. Advertised as 4MP but only able to get 2MP Resolution which is 1920x1080.