OpenIPC/firmware #1039: SD card not available on T31N

Slusko88, 2023-10-11:
Installed openIPC on my camera([Xiaomi MJSXJ03HL) but I am unable to access SD card.

I have tried gpio clear 54 procedure as well as rebooting with and without SD card inserted.

In dmesg output I have noticed that there is some problem with mmc initialization.

`[    1.440648] [resetd] GPIO undefined in /usr/sbin/resetd
[    2.104820] mmc0: host doesn't support card's voltages
[    2.104835] mmc0: error -22 whilst initialising SDIO card
[    2.123575] mmc0: host doesn't support card's voltages
[    2.123591] mmc0: error -22 whilst initialising MMC card
[    2.245693] mmc0: host doesn't support card's voltages
[    2.245709] mmc0: error -22 whilst initialising SDIO card
[    2.264091] mmc0: host doesn't support card's voltages
[    2.264109] mmc0: error -22 whilst initialising MMC card
[    2.385695] mmc0: host doesn't support card's voltages
[    2.385713] mmc0: error -22 whilst initialising SDIO card
[    2.404029] mmc0: host doesn't support card's voltages
[    2.404045] mmc0: error -22 whilst initialising MMC card
`
Here is complete dmesg output.
[dmesg_output.txt](https://github.com/OpenIPC/firmware/files/12871231/dmesg_output.txt)

themeadery, 2023-10-16:
I have similar errors on Wyze cam V3 which is T31X I believe.

[image]

themeadery, 2023-10-16:
It seems getting into uboot by pressing ctrl-c quickly after power up, then issuing `gpio set 57; gpio clear 57` then `boot` will fix the sd card reader (and also fixed my wlan).

darkfuncat, 2023-10-18:
You can also refer to this issue : https://github.com/OpenIPC/device-mjsxj03hl/issues/15#issuecomment-1762963710
The workaround to manualy do gpio commands at boot works but now it will be great to have this done automaticaly.
To be honest I dont know who could handle this and how (modifiy bootloader ? is it possible ?)
I tried to download and compile firmware (not bootloader) to test but it is very complicated for me.

themeadery, 2023-10-18:
@gtxaspec Can you commit this to our uboot and trigger a compile/release?

gtxaspec, 2023-10-18:
@themeadery there are several issues regarding GPIOs on the Ingenic platform in OpenIPC as a whole, involving both u-boot and the linux kernel.  I'm taking a look at them with some other devs, it will take some time to make some proper changes.

themeadery, 2023-10-18:
Awesome. I kinda figured you were working on a cleaner fix. Thx

Slusko88, 2023-10-19:
I managed to get sd card to work by running these commands:
```
gpio clear 54

echo "INSERT" > /sys/devices/platform/jzmmc_v1.2.0/present
```