OpenIPC/firmware #1041: Сборка прошивки

formatizator77, 2023-10-12:
Прошу помочь собрать прошивку для GK7205v210 для FPV с драйвером RTL8188EUS

widgetii, 2023-10-12:
Опишите конкретную проблему (текст ошибки, логи, т.п.)

formatizator77, 2023-10-12:
Устройство USB обнаруживается 

root@openipc-gk7205v210:~# lsusb
Bus 001 Device 001: ID 1d6b:0002
Bus 001 Device 002: ID 2357:010c
Bus 002 Device 001: ID 1d6b:0003

Но адаптер не появляется

root@openipc-gk7205v210:~# iwconfig
lo        no wireless extensions.

eth0      no wireless extensions.

root@openipc-gk7205v210:~# wifibroadcast start
Loading modules and wifi card driver...
Detecting wifi card vendor...
No usb wifi card detected. Check wifi stick connection, usb power or possible bad soldering.

widgetii, 2023-10-12:
В текущем конфиге компиляция данного драйвера не предусмотрена https://github.com/OpenIPC/firmware/blob/master/br-ext-chip-goke/configs/gk7205v210_fpv_defconfig#L99

Поддерживаются только RTL8812AU

closed 2023-10-12 by widgetii

Nuts-Underline, 2023-10-12:
если уж совсем точно то   wifibroadcast вообще поддерживает ограниченное число адаптеров, а официальных - еще меньше  https://github.com/svpcom/wfb-ng/wiki/WiFi-hardware

formatizator77, 2023-10-12:
Предлагаю попробовать собрать, по крайней мере этот проект https://github.com/jeanlemotan/esp32-cam-fpv на этих адаптерах работает. RTL8812AU сейчас нереально купить.

widgetii, 2023-10-12:
Вы можете это сделать самостоятельно в собственном профиле