OpenIPC/firmware #1043: Installed OpeIpc on a boward with hi3516ev100 but no cammera stream

likt0r, 2023-10-12:
Hi,

first THX for this project. I have here some Cameras I wanted to use, but the web interface needs Adobe Flash. So I looked out and found your wonderful project. I installed it on this board.
[image]
The Web interface is working, but I got no image. It seems it could not find the camera. 
I added the diagnostic messages. Any clue what is wrong and what i can try?
Thx for help, would be very nice to safe these cams from the dumpster.

Greetings 

`
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Linux version 3.18.20 (runner@fv-az163-423) (buildroot-gcc-12.2.0) #2 Thu Oct 12 10:28:24 UTC 2023
[    0.000000] CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=0005317f
[    0.000000] CPU: VIVT data cache, VIVT instruction cache
[    0.000000] Machine model: Hisilicon Hi3516CV300 DEMO Board
[    0.000000] Memory policy: Data cache writeback
[    0.000000] On node 0 totalpages: 8192
[    0.000000] free_area_init_node: node 0, pgdat c049e764, node_mem_map c1fba000
[    0.000000]   Normal zone: 64 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 8192 pages, LIFO batch:0
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 8128
[    0.000000] Kernel command line: mem=32M console=ttyAMA0,115200 panic=20 root=/dev/mtdblock3 rootfstype=squashfs init=/init mtdparts=hi_sfc:256k(boot),64k(env),2048k(kernel),5120k(rootfs),-(rootfs_data)
[    0.000000] PID hash table entries: 128 (order: -3, 512 bytes)
[    0.000000] Dentry cache hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Inode-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000000] Memory: 27392K/32768K available (3348K kernel code, 172K rwdata, 1020K rodata, 156K init, 242K bss, 5376K reserved, 0K highmem)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xffe00000   (2048 kB)
[    0.000000]     vmalloc : 0xc2800000 - 0xff000000   ( 968 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xc2000000   (  32 MB)
[    0.000000]     pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
[    0.000000]     modules : 0xbf000000 - 0xbfe00000   (  14 MB)
[    0.000000]       .text : 0xc0008000 - 0xc044c63c   (4370 kB)
[    0.000000]       .init : 0xc044d000 - 0xc0474000   ( 156 kB)
[    0.000000]       .data : 0xc0474000 - 0xc049f200   ( 173 kB)
[    0.000000]        .bss : 0xc049f200 - 0xc04dbda0   ( 243 kB)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000000] VIC @c2800000: id 0x00641190, vendor 0x41
[    0.000035] sched_clock: 32 bits at 3000kHz, resolution 333ns, wraps every 1431655765682ns
[    0.000810] Console: colour dummy device 80x30
[    0.000868] Calibrating delay loop... 398.13 BogoMIPS (lpj=1990656)
[    0.050024] pid_max: default: 32768 minimum: 301
[    0.050197] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.050223] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.051446] Initializing cgroup subsys devices
[    0.051492] Initializing cgroup subsys freezer
[    0.051518] Initializing cgroup subsys blkio
[    0.051573] CPU: Testing write buffer coherency: ok
[    0.052275] Setting up static identity map for 0x80330690 - 0x803306e8
[    0.054402] devtmpfs: initialized
[    0.056123] VFP support v0.3: not present
[    0.057007] pinctrl core: initialized pinctrl subsystem
[    0.065308] NET: Registered protocol family 16
[    0.066753] DMA: preallocated 256 KiB pool for atomic coherent allocations
[    0.082967] Serial: AMBA PL011 UART driver
[    0.083333] 12100000.uart: ttyAMA0 at MMIO 0x12100000 (irq = 21, base_baud = 0) is a PL011 rev2
[    0.356575] console [ttyAMA0] enabled
[    0.387579] SCSI subsystem initialized
[    0.391580] ssp-pl022 12120000.spi: ARM PL022 driver, device ID: 0x00800022
[    0.398650] ssp-pl022 12120000.spi: mapped registers from 0x12120000 to c286c000
[    0.430849] usbcore: registered new interface driver usbfs
[    0.436524] usbcore: registered new interface driver hub
[    0.442035] usbcore: registered new device driver usb
[... 102 lines cut ...]
[    2.249362] usb 1-1: new high-speed USB device number 2 using ehci-platform
[    2.399513] himci: mmc host probe
[    2.569933] hidraw: raw HID events driver (C) Jiri Kosina
[    2.576896] usbcore: registered new interface driver usbhid
[    2.582548] usbhid: USB HID core driver
[    2.587733] ip_tables: (C) 2000-2006 Netfilter Core Team
[    2.593294] TCP: cubic registered
[    2.596640] NET: Registered protocol family 17
[    2.601305] bridge: automatic filtering via arp/ip/ip6tables has been deprecated. Update your scripts to load br_netfilter if you need this.
[    2.635930] VFS: Mounted root (squashfs filesystem) readonly on device 31:3.
[    2.650069] devtmpfs: mounted
[    2.655264] Freeing unused kernel memory: 156K (c044d000 - c0474000)
[    3.536631] devpts: called with bogus options
[    4.188808] [resetd] GPIO undefined in /usr/sbin/resetd
[    4.204519] random: seedrng urandom read with 6 bits of entropy available
[    8.299664] hisi-femac 10050000.ethernet eth0: Link is Up - 100Mbps/Full - flow control rx/tx
[   18.562441] hibvt-i2c 12110000.i2c: wait idle abort!, RIS: 0x611
[   18.563632] hibvt-i2c 12110000.i2c: wait idle abort!, RIS: 0x611
[   18.563838] hibvt-i2c 12110000.i2c: wait idle abort!, RIS: 0x611
[   18.566143] hibvt-i2c 12110000.i2c: wait idle abort!, RIS: 0x611
[   18.566339] hibvt-i2c 12110000.i2c: wait idle abort!, RIS: 0x611
[   18.566534] hibvt-i2c 12110000.i2c: wait idle abort!, RIS: 0x611
[   18.566730] hibvt-i2c 12110000.i2c: wait idle abort!, RIS: 0x611
[   18.566929] hibvt-i2c 12110000.i2c: wait idle abort!, RIS: 0x611
`

Nuts-Underline, 2023-10-13:
Can be firmware with wrong sensor type or wrong settings. What file was flashed (link), what sensor on board?

likt0r, 2023-10-15:
Hi, thx for replying, I used these [instructions](https://openipc.org/cameras/vendors/hisilicon/socs/hi3516cv100) with this [File](https://openipc.org/cameras/vendors/hisilicon/socs/hi3516cv100/download_full_image?flash_size=8&flash_type=nor&fw_release=lite)
The Sensor, I am not sure on the camera, is mark: IR716180401
[image]
[image]

Greetings

likt0r, 2023-10-15:
Ah ok thx 
vendor: SmartSens
model: SC2235

```bash
root@openipc-hi3516cv300:~# ipctool
The ipctool installed as remote GitHub plugin
---
chip:
  vendor: HiSilicon
  model: 3516EV100
board:
  vendor: OpenIPC
  version: 2.3.10.12
ethernet:
  mac: "9a:81:a8:e4:b2:d1"
  u-mdio-phyaddr: 1
  phy-id: 0x001cc816
  d-mdio-phyaddr: 0
rom:
- type: nor
  block: 64K
  partitions:
    - name: boot
      size: 0x40000
      sha1: e210b643
    - name: env
      size: 0x10000
      sha1: a5bbb8a1
      contains:
        - name: uboot-env
          offset: 0x0
    - name: kernel
      size: 0x200000
      sha1: 97c4f275
    - name: rootfs
      size: 0x500000
      path: /,squashfs
      sha1: b4d8f3e9
    - name: rootfs_data
      size: 0xb0000
      path: /overlay,jffs2,rw
  size: 8M
  addr-mode: 3-byte
ram:
  total: 32M
firmware:
  u-boot: "2010.06 (Nov 14 2022 - 19:27:26)"
  kernel: "3.18.20 (Thu Oct 12 10:28:24 UTC 2023)"
  toolchain: buildroot-gcc-12.2.0
sensors:
- vendor: SmartSens
  model: SC2235
  control:
    bus: 0
    type: i2c
    addr: 0x60
  vicap-state: down
``

Nuts-Underline, 2023-10-16:
Well SC2235 here
so SC2235 must be in settings
_hibvt-i2c_ is someking of software layer for I2C interface, it sends some datas for sensors to
 _wait idle abort!_ software cannot read someting via I2C it can be with srong settings

likt0r, 2023-10-16:
Anybody an idea, what i can try or should do ?

themactep, 2023-10-17:
come to the [telegram group](https://t.me/OpenIPC) for help. issues are not the place for one-on-one tech support.

catulacnpoly, 2024-04-27:
> ```shell
> ipctool
> ```

seems I got the same problem：

root@openipc-hi3516ev100:~# ipctool
The ipctool installed as remote GitHub plugin
---
chip:
  vendor: HiSilicon
  model: 3516EV100
board:
  vendor: OpenIPC
  version: 2.4.04.25
ethernet:
  mac: "32:10:81:4b:6a:cb"
  u-mdio-phyaddr: 1
  phy-id: 0x02430c54
  d-mdio-phyaddr: 0
rom:
- type: nor
  block: 64K
  partitions:
    - name: boot
      size: 0x40000
      sha1: e210b643
    - name: env
      size: 0x10000
      sha1: 667429e1
      contains:
        - name: uboot-env
          offset: 0x0
    - name: kernel
      size: 0x300000
      sha1: 3b905d0c
    - name: rootfs
      size: 0xa00000
      path: /,squashfs
      sha1: c1244305
    - name: rootfs_data
      size: 0x2b0000
      path: /overlay,jffs2,rw
  size: 16M
  addr-mode: 3-byte
ram:
  total: 32M
firmware:
  u-boot: "2010.06 (Nov 14 2022 - 19:27:26)"
  kernel: "3.18.20 (Thu Apr 25 02:10:11 UTC 2024)"
  toolchain: buildroot-gcc-13.2.0
sensors:
- vendor: SmartSens
  model: SC2235
  control:
    bus: 0
    type: i2c
    addr: 0x60
  vicap-state: down