OpenIPC/firmware #1052: range feature for httpd

darkfuncat, 2023-10-15:
Hello,
I try to download video files from openipc camera, but only some part of file. (range feature for http download)
The purpose is to send video file of alarm few seconds before trigger.
The feature from HTTP1.1 seems not enabled by default in openipc.
Is it possible to activate it ?
Exemple, when doing :
`curl -r -5000 http://user:pwd@IP:85/cgi-bin/dl.cgi?file=/mnt/mmcblk0p1/2023/10/14/XX.mp4`
Curl should download only last 5000 bytes from file, but instead it download whole file (which is very long and useless)
Or perhaps there is another way to do this ?
Thanks

themactep, 2023-10-15:
Enable this https://github.com/OpenIPC/firmware/blob/001687b3b736a715303f92621bff01da0d20c389/general/package/busybox/busybox.config#L887, recompile and update fw, check that is works, send a pr.

darkfuncat, 2023-10-17:
Thanks for your advice.
OK so I have followed instructions to recompile firmware.
I changed the file you mentioned with :
`CONFIG_FEATURE_HTTPD_RANGES=y`
(I guess this is the good syntax regarding other lines)
After compile I resintall rootfs.squashfs.t31 and uImage.t31, reboot and still not able to download range of file with above curl command.
Any tip or something I forgot ?
(I think that I am reaching my limits in term of development...)
Thanks

themactep, 2023-10-19:
Did you try to download a piece of a real file to check if the range function works? The dl.cgi is a script that proxies content of the requested file from anyplace in the system to the http client. This might be a problem for determining the range. If direct dowload works with the enabled option, we'll start from there.

darkfuncat, 2023-10-24:
Hello @themactep 
The purpose here is to trigger alarm and send a sample of video few seconds before and after trigger (basic thing from all other camera I own)
To be honest, I am facing many difficulties to achieve this with OpenIPC :
- SD Card is very unstable with my camera model so I cannot record continuously : https://github.com/OpenIPC/device-mjsxj03hl/issues/15#issuecomment-1762963710 and https://github.com/OpenIPC/firmware/issues/1039#issuecomment-1768980836 > manual fix OK
- I cannot activate alarm trigger too without some problems : https://github.com/OpenIPC/firmware/issues/1010#issuecomment-1743194539 > workaround with lower resolution OK
- here in this issue I cannot download via HTTP/curl last piece of video > KO
- And even if I send (manualy) the end of recording file, I cannot repair video stream via ffmpeg (on any other camera it is good) to have a valid video file > KO, I dont understand what is the video format and how it is recorded (perhaps linked to the sdcard problem)

This is a lot of work and many problems and it takes time...