OpenIPC/firmware #1091: 8188eu driver compile error for gk7205v200_ultimate

whoim2, 2023-10-28:
enable BR2_PACKAGE_RTL8188EU in gk7205v200_ultimate_defconfig
 ./building.sh gk7205v200_ultimate
 compile error:
 https://pastebin.com/73NLYX3h

zerog2k, 2023-11-15:
I think that's an older driver, and that board using kernel 4.9.x

can you instead try:
```
BR2_PACKAGE_LINUX_FIRMWARE_OPENIPC_RTL_8188EU=y
BR2_PACKAGE_RTL8188EUS_OPENIPC=y
```

sciuridae0603, 2024-10-06:
Hi @zerog2k 

I'm facing the same problem when building hi3518ev100 firmware.
I tried the option BR2_PACKAGE_LINUX_FIRMWARE_OPENIPC_RTL_8188EU and BR2_PACKAGE_RTL8188EUS_OPENIPC, but it still gives compile errors.

Compile log: https://gist.github.com/sciuridae0603/287b8c71ae1be85ee64b5bfdd5b1e261

closed 2026-10-26 by openipc-ai