OpenIPC/firmware #1092: Realtek / rts3903n support

DSchndr, 2023-10-29:
Hi, this platform might be interesting to add support for since it is not listed

It is used in cameras like dlink dcs-6100 / aoni e96kr7f (odm), some tuya cams,... which cost ~15€ and are available from amazon
Its a mipsel cpu with 64MB ram running a "fairly recent" kernel (4.9.51) with h264/mjpeg encoder, gbit mac, usb, audio,...
The "sdk" and open source (incomplete) streamers as well as examples are available
PS: The aoni streamer from the first dlink update package comes with debug symbols

According to the sdk documentation there are even more realtek chips (which it supports)

I might get some more to do tests on if needed

widgetii, 2023-10-29:
Please provide a link for SDK and examples you mentioned above

DSchndr, 2023-10-29:
SDK: https://dlink-gpl.s3.amazonaws.com/GPL2000202/DCS-6100LH_v1.x_GPLCode.tar.gz
Under `DCS-6100LH_v1.x_GPLCode.tar.gz\DCS-6100LH_v1.x_GPLCode.tar\DCS-6100LH_v1.x_GPLCode\rts3903_sdk_v3.1.1\platform\source\rtscore\rts_sample\rtstream\` are the realtek examples

(Incomplete) Streamer: https://github.com/cjj25/Yi-RTS3903N-RTSPServer

widgetii, 2023-10-29:
Looks than everything is needed for the start, but main issue should be resolved first: Dlink uses Buildroot 2018.02, we're using Buildroot 2023.02 and a lot of efforts should be done to bump codetree to mainline version

zry98, 2024-03-03:
RTS3903 is also used in TP-Link Tapo C200, would be awesome to be supported!

gortizt, 2024-04-09:
> RTS3903 is also used in TP-Link Tapo C200, would be awesome to be supported!

+1

sperglord8008s, 2024-04-23:
> Many Tools Are Already Fairly Developed and Documented.
> 
> SDK --> https://github.com/cjj25/RTS3903N-rsdk-4.8.5-5281
> 
> Useful binaries for RTS3903N --> https://github.com/cjj25/RTS3903N-Tools
> 
> Dumped Firmware + hard-coded credentials for console --> https://drmnsamoliu.github.io/firmware.html