OpenIPC/firmware #1096: XM530: image.jpg does not work

carbofos, 2023-10-30:
xm530 (xm530 family), sc2315e, 8 MB nor flash, 2.3.10.28-lite, openipc-xm530, 00:12:41:20:52:22

500 Internal Server Error
Internal error

```
GET /image.jpg HTTP/1.0

HTTP/1.1 500 Internal error
Content-Type: text/html
Connection: close
Date: Mon, 30 Oct 2023 19:24:04 GMT
Content-Length: 127

<html><head><title>500 Internal Server Error</title></head><body><h1>500 Internal Server Error</h1>Internal error</body></html>Connection closed by foreign host
```

Majestic settings
.jpeg.enabled: true
.jpeg.size: 
.jpeg.qfactor: 
.jpeg.toProgressive: 

log
```
19:25:12  <       majestic> [    main] main@123                      Majestic Lite for Xiongmai, version master+71d6c43, built on 2023-10-26
19:25:12  <       majestic> [app_conf] load_config@108               Using /etc/majestic.yaml as main configuration
19:25:12  <       majestic> [watchdog] watchdog_start@121            Watchdog timeout set to 10 seconds
DEBUG: [ISP_SDK]BUILD TIME:Sep 21 2021 11:58:04
DEBUG: OSC:12M
19:25:12  <       majestic> [     log] printf@268                    SYS/VI set in      [ Sofia ]
19:25:12  <       majestic> [     log] printf@268                    Vstd_init  [ PAL ]
19:25:12  <       majestic> [     log] printf@268                    Rslt       [ 0 ]
19:25:12  <       majestic> [     log] printf@268                    IPC_Venc:  [ H.264 ]
19:25:12  <       majestic> [     log] printf@268                    Sns_IF:    [ By src ]

DEBUG: SC307E!
DEBUG: sensor_get_chip:
19:25:13  <       majestic> [     log] printf@268                    0x30 
 19:25:13  <       majestic> [     log] printf@268                    0x2 
 19:25:13  <       majestic> [     log] printf@268                    0x1 
 19:25:13  <       majestic> [     log] printf@268                    0x45 
19:25:13  <       majestic> [    puts]                               
ERR: open /mnt/mtd/Config/SensorType.bat failed
DEBUG: Venc(264):495
DEBUG: DspChip:XM530
DEBUG: SnsIF:MIPI
DEBUG: Vstd:PAL[25fps]
DEBUG: ------------- SC307E (@20190110_mipi) ----------------
DEBUG: sensor_init Ok!
DEBUG: Init over ~~~~~~~~
DEBUG: ProductType: 0x0
DEBUG: libfvideo: SC307E!
DEBUG: isp_sample end!
DEBUG: imageStyle XM!
19:25:13  <       majestic> [     sdk] ViChnConfig@164               ViChn = 0, enSize = 14
19:25:13  <       majestic> [     sdk] COMM_VENC_Start@249           VencChn = 0, enType = 96, enSize = 14 (1920x1080), enRcMode = 1
19:25:13  <       majestic> [     sdk] COMM_VENC_Start@249           VencChn = 2, enType = 26, enSize = 0 (704x576), enRcMode = 1
19:25:13  <move_detect_pro> [    puts]                               XM_MPI_ISP_MemInit OK...
19:25:13  <       majestic> [   httpd] new_http_server@373           HTTP server started on 0.0.0.0:80
DEBUG: black and white 
DEBUG: EShutter 1/1
DEBUG: color 
19:25:15  <       majestic> [     sdk] sdk_take_jpeg@866             take jpeg venc_chn(2)   0x0  q:50
19:25:15  <       majestic> [     sdk] sdk_take_jpeg@858             Stuck in progress
19:25:15  <       majestic> [  on_jpg] on_jpeg_data_cb@71            Empty data has been returned for JPEG request
19:25:15  <       majestic> [     sdk] sdk_take_jpeg@858             Stuck in progress
19:25:15  <       majestic> [  on_jpg] on_jpeg_data_cb@71            Empty data has been returned for JPEG request
DEBUG: Delay Over !!!

```

ZigFisher, 2023-10-30:
Нужно, это имеется ввиду что не работает h264 ?

carbofos, 2023-10-31:
Я имел ввиду что она работает не перегружаясь если переключить видео в h264.
А JPEG таки хотелось бы иметь чтобы скриншоты на стену отправлялись и все такое. Очень нужная функция.

widgetii, 2023-10-31:
Когда-то JPEG работало и есть смысл отыскать релиз когда сломалось (возможно, это случилось при переходе на новый Buildroot)

carbofos, 2023-11-01:
> Когда-то JPEG работало и есть смысл отыскать релиз когда сломалось (возможно, это случилось при переходе на новый Buildroot)

Дайте знать когда точно был переход на новый BR и я найду в телеге прошивки до и после перехода и сравню функционал.

carbofos, 2023-11-18:
> Когда-то JPEG работало и есть смысл отыскать релиз когда сломалось (возможно, это случилось при переходе на новый Buildroot)

Самая старая сборка из телеги за 25 июля уже не рабочая.
Но я нашел у себя сборку за 5 ноября 2022 и там JPEG работает.

closed 2026-10-14 by widgetii

widgetii, 2026-10-14:
Исправление влито в majestic и должно попасть в ближайший nightly: majestic версии `master+862c458` или новее.

**Причина.** На XM530 JPEG-кодер (VENC 2) получает кадры только пока работает кодер **VENC 1**, то есть второй поток. Поэтому `/image.jpg` и работал только с `video1.enabled: true`, как @carbofos и нашёл. Без него запрос ждал кадр, который не приходил. После этого все следующие снимки отвечали 500 "Stuck in progress", пока majestic не перезапускали.

**Что изменилось:**
- VENC 1 теперь запускается всегда, когда включён `jpeg`. Если `video1` выключен, его поток никуда не публикуется.
- Снимки встали в общую очередь ожидания, как на других SoC. Одновременные запросы получают один кадр. Если кадра нет 10 с, запрос получает отказ (503 с причиной) и не зависает навсегда.
- Сохранение настроек (WebUI, `cli`, SIGHUP) раньше оставляло камеру без видео и потом могло уронить majestic: отсюда SIGILL в отчётах о сбоях в WebUI. На XM530 SDK не умеет пересобрать конвейер внутри процесса, поэтому majestic теперь перезапускает себя с тем же PID.

Проверено на XM550 + IMX335:
- `/image.jpg` с настройками по умолчанию: около 0,1 с на снимок.
- 120 из 120 снимков при 10 одновременных запросах.

Последняя правка перезапуска не проверена на железе: тестовая плата вышла из строя. Если у кого-то есть XM530, проверьте, пожалуйста, на nightly:
- снимки при `video1` выключенном;
- несколько сохранений настроек в WebUI.

Если что-то не так, пришлите `logread | grep majestic`, `/proc/umap/venc` и файлы из `/etc/crash`.

Размер JPEG по-прежнему фиксирован (704x576).

carbofos, 2026-10-16:
@widgetii 
Обновил до master+1d9c834 от 2026-10-15 и проверил:
```
root@openipc-xm530:~# yaml-cli -g .video1.enabled 
false
```

JPEG отдает и в логах отчитывается:
```
19:03:25  <       majestic> [  on_jpg] send_regular_jpeg@22          JPEG was sent to client
19:03:25  <       majestic> [  on_jpg] send_regular_jpeg@22          JPEG was sent to client
19:03:25  <       majestic> [  on_jpg] send_regular_jpeg@22          JPEG was sent to client

```