OpenIPC/firmware #1105: SSC333 Compatibility (Tapo C110)

cjdell, 2023-11-04:
Hello. I notice there is no guide for SSC333 however there are references to it in the source code. Does this mean that it could still work (unofficially) if I build the firmware myself? Is there any way I can help with the testing so support for this chip can be added?

Thank you

viktorxda, 2023-11-04:
Hello, theoretically it should work due to the general compatibility of the SDK.
So far it hasn't been tested, if you feel adventurous you can try the SSC335 version and report back if it's working.

cjdell, 2023-11-04:
Thanks for the quick reply. Also noticed this file added 2 weeks ago:

https://github.com/OpenIPC/firmware/blob/master/br-ext-chip-sigmastar/configs/ssc333_lite_defconfig

Does this mean someone has tried already to make this work?

Here are the photos:

[image]
[image]

viktorxda, 2023-11-04:
Thanks for the photos. Yes, Sigmastar gained a lot of attention recently and we are trying to expand the device list.
I have ordered the device to check it as well, this could potentially become a promising device for openipc.

cjdell, 2023-11-04:
Here's the first boot log. The pad near "TP501" is the TX pin:
https://gist.github.com/cjdell/c48f95d9910f2e23fb0ee7dfa6a41d34

FlyRouter, 2023-11-04:
Please give information what version your C110 device is, v1.0, v2.0 or other. 
Thank you.

cjdell, 2023-11-04:
I think 2.0 according to the sticker on the PCB.

viktorxda, 2023-11-06:
Do you have a programmer and soic clip available for direct write access?

cjdell, 2023-11-06:
@viktorxda No but I will buy one. Can you recommend a good SOIC clip / programmer? I am willing to get my hands dirty with these devices.

viktorxda, 2023-11-06:
The CH341A, it is not particularly good or powerful, but it is decent enough and easy to obtain.

viktorxda, 2023-11-07:
I forgot to mention, for Sigmastar you can also use [mstar-snander](https://github.com/OpenIPC/wiki/blob/master/en/sigmastar-unbrick.md) to read and write directly to the flash chip over the uart connection.

cjdell, 2023-11-09:
@viktorxda I got the stock firmware extracted with the SOIC clip. The `binwalk` yielded a lot of interesting secrets within the binary. It's built on OpenWRT and most of the startup and initial config is written as plain text shell scripts. I'm willing to share the binary with you if you're interested.

I found the SSC333 build of OpenIPC and decided to try writing it directly with the SOIC clip but it bricked the device. Appears it doesn't like being programmed this way. Fortunately `snander` worked perfectly with the CH341A so thank you for telling me this otherwise I would have been stuck.

So, end result is OpenIPC is working on the C110! I had to extract `ssw101b_wifi_usb.ko` from the original firmware which was within the `squashfs`. Copied to the camera via SD card so it's now on the overlay filesystem. Manually loaded the module and configured `wpa_supplicant` and the RTSP stream came to life.

Very impressed with the project. To round this off as a success I still need to get the IR/night vision configured which means trial and error searching for the GPIO pins. Also need to make the kernel module for WiFi auto load. Tomorrow's job...

Thanks for all your help so far. Going to buy more C110's as the price/performance for these cameras is now extremely appealing.

viktorxda, 2023-11-09:
Thank you for the feedback, I'm glad that it worked out in the end.

An initial wiki entry for the C110 with the gpios has been created here: https://github.com/OpenIPC/wiki/blob/master/en/device-tapo-c110.md
Btw, feel free to add any information to the wiki that you feel is helpful.

For the wlan driver we have a profile system to set the driver: https://github.com/OpenIPC/firmware/blob/master/general/overlay/etc/wireless/usb#L94

Please try the following:
```
fw_setenv wlandev ssw101b-ssc333-tapo-c110
fw_setenv wlanssid SSID
fw_setenv wlanpass Password

```

cjdell, 2023-11-09:
@viktorxda Thanks for that. All working now. Just one more thing. I noticed that only 32MB of RAM is available to Linux and memory is a bit tight. I know this is a 64MB SoC. Is this normal?

viktorxda, 2023-11-09:
Yes, some of the ram is assigned to the vendor kernel driver and encoder.

nikdoof, 2023-11-12:
I can also report another successful device. Direct flashing via a SOP clip didn't work, but using snander worked perfectly. Connected to the UART and configured the `wlandev` and the SSID, reboot and everything came up fine.

I didn't try and read out the flash directly after the snander flash to see if it had any differences from the bin, but that might be for the next camera.

On a side note I did direct flash a Tapo C310 with the SSC335 version and it worked out the box with ethernet without any other changes, but I'll commit to the wiki and such for that device.

nikdoof, 2023-12-11:
From my and Chris' experimentation with these devices, we've found that they don't like direct flashing (for some reason we still can't get to the bottom of) so its a little bit of a pain to get running.

1. De-case the camera, and remove the board from the case
2. Either make a [clamp](https://www.printables.com/model/139112-pogo-pin-clamp-block-for-use-with-flash-adapter-wi) or solder on wires onto the 4 pads on the back of the board starting with 'TP501'
3. Hook up a USB to Serial/TTL device to the wires, you'll have TX/RX pins, and a 5v/GND pins, hook up the ground but not the 5v.
4. Open a serial console to the serial port, plug in the device and check if you get the boot up log.
5. Grab mstar-snander and the latest C110 firmware from the OpenIPC website.
6. Use mstar-snander to [flash the firmware image](https://github.com/OpenIPC/wiki/blob/master/en/sigmastar-unbrick.md)
7. Reboot the device, login via the console (root/12345) and set the [WiFi config](https://github.com/OpenIPC/firmware/issues/1105#issuecomment-1803202836)
8. Reboot again, and hopefully it'll be on your WiFi and you can tweak it from there on.

Like I said, it'd be great if a direct flash of the firmware worked using a clamp as it saves quite a bit of hassle, but for some reason we just never got it to work. Flashing the exact firmware via mstar-snander works perfectly every time.

fabianmuehlberger, 2023-12-11:
Hey @nikdoof Thank you for the guide, I appreciate it!

Luckily, I have invested in some PCBite gooseneck probes, so no soldering needed ;) (cant praise them enough, they are so handy) . But the pogo pin clamp block looks pretty handy too.

Hopefully I will be able to report a success tomorrow 

Best

viktorxda, 2023-12-12:
Theoretically it should be possible to create a complete binary, the flash layout should be:
```
- 0x000000: u-boot-ssc333-nor.bin
- 0x050000: uImage.ssc333
- 0x250000: rootfs.squashfs.ssc333
```

Current firmware for the C110: [OpenIPC/builder](https://github.com/OpenIPC/builder/releases/tag/latest)
And the bootloader from here: [OpenIPC/distributor](https://github.com/OpenIPC/distributor/releases/tag/latest)

nikdoof, 2023-12-12:
We're going to have another C110v2 to setup soon, so we'll experiment more with direct flashing the NOR. Our internal build [has that layout](https://github.com/leigh-hackspace/openipc-composer/blob/main/workplace/scripts/firmware_builder.py), and the same bin works when wrote via `mstar-snander`.

viktorxda, 2023-12-12:
If the offsets are correct it might be a problem with the flasher/software.
Are you using mstar-snander for this as well?

fabianmuehlberger, 2023-12-13:
just bought a crappy CH341a [device](https://www.amazon.de/dp/B08HP4X326?psc=1&ref=ppx_yo2ov_dt_b_product_details) from amazon, got a chip labeled with CH341B. RX,TX are on 5V. Is this expected? I usually don't trust those dodgy devices.. 
I got a Lattice, ATMEL and other smaller programmers on hand, or do I need the snander for this flash?

themactep, 2023-12-13:
Ch341a is a decent versatile device that needs some TLC. [Fix the voltage bug](https://github.com/openipc/wiki/blob/master/en/hardware-programmer-ch341a-voltage-fix.md) and switch jumper from serial mode to programmer mode to use it as a programmer not as a serial adapter.