OpenIPC/firmware #1112: HI3535dv100 cannot connect to the USB shared network

suiaiyun, 2023-11-06:
Hello. I am currently testing the NVR board, with the main controller being HI3536DV100. I am now able to display images normally through HDMI. However, when I tried to share the network on my phone's USB to display images in "fpv_vr", QOpenHD, and QGC, I was unsuccessful. When using the ifconfig command to view the network card device, no USB0 network card was found, and the same was true for a different phone. I have added the configuration of USB0 in the/etc/network/interfaces file according to the instructions in WiFi. When testing the Android systems of vivo and Oppo, is it because Openipc does not support these two phones? Or do I lack any key configurations? How should I solve this problem?

Nuts-Underline, 2023-11-16:
looks like you are using so named "usb tethering" mode also known as RNDIS (gives USB0 network device)
that means yours phones must be configured as USB modem 
and there need RNDIS device driver for OpenIPC: "CONFIG_USB_NET_RNDIS_WLAN=y" its not added to any base configs so all it not working at all
so i thing there need to rebuild firmware with improving "CONFIG_USB_NET_RNDIS_WLAN=y" in config at least

widgetii, 2026-10-21:
`rndis_wlan` is the wrong driver: it is for RNDIS WiFi dongles. Phone tethering uses `rndis_host`, which the hi3536dv100 image has built as a module since #709, together with `cdc_ether` and `cdc_ncm`. I checked on a hi3536dv100 running a current build: the modules are on the image and load cleanly. The firmware does not autoload USB drivers, though, so nothing loads `rndis_host` when the phone is plugged in. That is why `usb0` never appears.

To have the driver loaded at boot and `usb0` brought up with DHCP:

```
fw_setenv wlandev rndis-generic
reboot
```

- Plug the phone in and switch on USB tethering **before** the NVR boots. Android turns tethering off whenever the cable is unplugged. If you switch it on after boot, run `ifup usb0` yourself.
- If your phone tethers over NCM rather than RNDIS, run `modprobe cdc_ncm` and then look for `usb0` or `eth1` in `ip link`.

Could you try this on a current image and post `dmesg | tail -30` and `ip addr` with the phone attached?

closed 2026-10-21 by widgetii