OpenIPC/firmware #1114: Wyze Devices Issues with current OpenIPC

peterHz83, 2023-11-06:
Greetings,

I hope this message finds you and your team well. In light of the persistent concerns regarding the security and stability of the original firmware, I have transitioned my Wyze devices to the OpenIPC firmware.

I have explored the documentation and wiki and despite extensive research, I continue encountering certain technical difficulties and would like to inquire if these are common among users or specific to my setup:

1. Wyze V3 Camera - Night Mode Issue: I observe significant blurring with any movement during Night Mode. This issue occurs consistently across multiple devices and various settings, a problem not evident with the original Wyze firmware.

2. Wyze V3 Camera - Frame Rate Issue: Activating the On-Screen Display (OSD) feature causes the frame rate to decrease to below 20 fps, even though it is configured for 30 fps. Deactivating OSD resolves this issue.

3. Wyze V3 Camera - OSD Flickering: With the OSD feature enabled, the display text intermittently flickers without apparent pattern.

4. Wyze V3 Camera - SD Card Recording: Similar to the original firmware settings, I have programmed the camera to record and save footage to an SD card. However, the saved videos are malfunctioning, showing extended playtimes for one-minute recordings and not playing back correctly.

5. Wyze Pan Version 2 Camera - All issues are the same as with Wyze V3 Camera above.

6. Wyze V2 Camera - Sensor Recognition: The device fails to recognize the image sensor.

Given these issues, I would like to understand if they are widespread or unique to my experience. This inquiry is particularly pertinent as many, many of my colleagues with Wyze devices are also exploring alternative firmware options due to recent security concerns.

I remain optimistic about the potential of OpenIPC firmware to serve as a robust alternative to the stock firmware. Should it be helpful, I am ready to provide further logs or diagnostic information.

Warm regards,
peterh

gtxaspec, 2023-12-02:
@peterHz83 just saw this issue, heres some advice

1.  disable the automatic nightmode feature in majestic, and use `imp-control.sh` to enable night mode, this should stop the blurring during nightvision.
6.  this is fixed on the latest openipc builds.

i'll double check the other stuff and see if it still happens.

csev1755, 2024-03-21:
Just chiming in as I have OpenIPC fully functional on 2 Wyze V2 cameras. Maybe there's been a recent update that fixed your issue?

peterHz83, 2024-03-22:
@csev1755, The last time I ran OpenIPC on my V2, the sensor was recognized, but other listed issues still occurred. In my attempts to research and diagnose fixes, I found the developers unhelpful and perceived them as hostile towards suggestions for correcting the problems. I have since transitioned from OpenIPC to another alternative firmware for my devices, which is functioning well.  Closing this issue as it no longer pertains to my usage.

regards,
peterh

closed 2024-03-22 by peterHz83