OpenIPC/firmware #1126: unable to access sdcard / mmc on hi3518ev100 with openipc uboot

zerog2k, 2023-11-08:
after flashing my ipcam (jovision h210?), unit boots fine, but I'm unable to do anything with mmc (sdcard) via fatload/fatinfo in uboot
`fatinfo mmc` gives:
```
** Invalid boot device **
```
(mmc/sdcard works fine in linux)

serial console log:
```
U-Boot 2010.06 (Nov 14 2022 - 19:25:39)

Check spi flash controller v350... Found
Spi(cs1) ID: 0xEF 0x40 0x17 0x00 0x00 0x00
Spi(cs1): Block:64KB Chip:8MB Name:"W25Q64FV"
In:    serial
Out:   serial
Err:   serial
RAM size: 64MB
Hit any key to stop autoboot:  0 
OpenIPC # 
OpenIPC # fatinfo mmc

** Invalid boot device **
OpenIPC # version

U-Boot 2010.06 (Nov 14 2022 - 19:25:39)
OpenIPC # getinfo
getinfo - print hardware information

Usage:
getinfo bootmode - get start memeory type e.g. nand/spi etc
getinfo nand - get nand flash information
getinfo spi - get spi flash information
getinfo version - get system version

OpenIPC # getinfo version
version: 3.0.3
OpenIPC # 
```

ipctool output from original image:
```
# ./ipctool 
Error: unexpected value for SOI == 0x9711
---
chip:
  vendor: HiSilicon
  model: 3518EV100
board:
  possible-IR-cut-GPIO: 6,18,20,42,43,44,73,76,79
ethernet:
  mac: "3c:33:00:c2:7d:19"
  u-mdio-phyaddr: 1
  phy-id: 0x0000ffff
  d-mdio-phyaddr: 0
  phy-mode: mii
rom:
- type: nor
  block: 64K
  partitions:
    - name: boot
      size: 0x40000
      sha1: 867356df
      contains:
        - name: uboot-env
          offset: 0x30000
    - name: appfs
      size: 0x480000
      path: /home,squashfs
      sha1: 36e885c9
    - name: ker
      size: 0x1b0000
      sha1: 2502100d
    - name: rootfs
      size: 0x100000
      sha1: f61bac71
    - name: hwconfig
      size: 0x10000
      sha1: 3becb626
    - name: config
      size: 0x80000
      path: /etc/ppp,jffs2,rw
    - name: flash
      size: 0x800000
      sha1: 37a108d6
  size: 16M
  addr-mode: 3-byte
ram:
  total: 64M
  media: 21M
firmware:
  u-boot: "2010.06 (Jul 31 2015 - 17:02:35)"
  kernel: "3.0.8 (Mon Jun 29 11:56:43 CST 2015)"
  toolchain: gcc version 4.4.1 (Hisilicon_v100(gcc4.4-290+uclibc_0.9.32.1+eabi+linuxpthread)) 
  libc: uClibc 0.9.32.1
  sdk: "Hi3518_MPP_V1.0.A.1  (Aug 24 2015, 22:43:01)"
  main-app: /tmp/sctrl
sensors:
- vendor: OmniVision
  model: OV9712
  control:
    bus: 0
    type: i2c
    addr: 0x60
  data:
    type: DC
  clock: 24MHz
# 
```

zerog2k, 2023-11-15:
I'm actually wondering if it's compiled without mmc support. There is not even `mmc` command available.
https://github.com/OpenIPC/u-boot-hi3516cv100/blob/master/common/cmd_mmc.c#L36-L102
I'm going to try to figure out how to build it if i can.

zerog2k, 2023-11-15:
I'm guessing it's disabled here 
https://github.com/OpenIPC/u-boot-hi3516cv100/blob/master/include/configs/hi3518e.h#L243-L265
but I'm not sure how to build this particular uboot

I have tried
```
make hi3518e_config
make
```

but fails at
```
board.c: In function ‘start_armboot’:
board.c:390:9: warning: implicit declaration of function ‘set_default_env’ [-Wimplicit-function-declaration]
  390 |         set_default_env();
      |         ^~~~~~~~~~~~~~~
board.c: At top level:
board.c:108:6: error: ‘coloured_LED_init’ aliased to external symbol ‘__coloured_LED_init’
  108 | void coloured_LED_init (void) __attribute__((weak, alias("__coloured_LED_init")));
      |      ^~~~~~~~~~~~~~~~~
board.c:124:6: error: ‘blue_LED_off’ aliased to external symbol ‘__blue_LED_off’
  124 | void blue_LED_off(void) __attribute__((weak, alias("__blue_LED_off")));
      |      ^~~~~~~~~~~~
board.c:122:6: error: ‘blue_LED_on’ aliased to external symbol ‘__blue_LED_on’
  122 | void blue_LED_on(void) __attribute__((weak, alias("__blue_LED_on")));
      |      ^~~~~~~~~~~
board.c:120:6: error: ‘yellow_LED_off’ aliased to external symbol ‘__yellow_LED_off’
  120 | void yellow_LED_off(void) __attribute__((weak, alias("__yellow_LED_off")));
      |      ^~~~~~~~~~~~~~
board.c:118:6: error: ‘yellow_LED_on’ aliased to external symbol ‘__yellow_LED_on’
  118 | void yellow_LED_on(void) __attribute__((weak, alias("__yellow_LED_on")));
      |      ^~~~~~~~~~~~~
board.c:116:6: error: ‘green_LED_off’ aliased to external symbol ‘__green_LED_off’
  116 | void green_LED_off(void) __attribute__((weak, alias("__green_LED_off")));
      |      ^~~~~~~~~~~~~
board.c:114:6: error: ‘green_LED_on’ aliased to external symbol ‘__green_LED_on’
  114 | void green_LED_on(void) __attribute__((weak, alias("__green_LED_on")));
      |      ^~~~~~~~~~~~
board.c:112:6: error: ‘red_LED_off’ aliased to external symbol ‘__red_LED_off’
  112 | void red_LED_off(void) __attribute__((weak, alias("__red_LED_off")));
      |      ^~~~~~~~~~~
board.c:110:6: error: ‘red_LED_on’ aliased to external symbol ‘__red_LED_on’
  110 | void red_LED_on (void) __attribute__((weak, alias("__red_LED_on")));
      |      ^~~~~~~~~~
/home/jens/build/u-boot-hi3516cv100/config.mk:252: recipe for target 'board.o' failed
make[1]: *** [board.o] Error 1
make[1]: Leaving directory '/home/jens/build/u-boot-hi3516cv100/arch/arm/lib'
Makefile:420: recipe for target 'arch/arm/lib/libarm.a' failed
make: *** [arch/arm/lib/libarm.a] Error 2
```
I'm missing something important ;)

zerog2k, 2023-11-16:
ok through some hacking, I was able to get it to build, with either `hi3518e_config` or `hi3516c_config`
however because of some outdated function calls (e.g. `wait_ms` that need to port to `udelay`) due to ancient uboot base,
I did have to disable USB functions but I dont think that should have mattered. I did enable mmc stuff.
I was able to side-load from existing uboot (loady -> go) and execute the uboot without issue. However, something still isnt fully right (possibly with cpu/register/etc init code), as once I flashed the uboot, it would not boot from it.

So I'm back to point of wondering, where did this uboot come from and how was it built?
https://github.com/OpenIPC/firmware/releases/download/latest/u-boot-hi3518ev100-universal.bin

viktorxda, 2023-11-16:
We added a repo to automate the building process for uboot.
Please take a look at: https://github.com/OpenIPC/distributor

closed 2023-11-18 by zerog2k