OpenIPC/firmware #1129: gk7205v300+imx335_venc_firmware

srtbit, 2023-11-10:
gk7205v300+imx335_venc,nvr3536 ground station can be connected, but there is no video available.Through inspection, it was found that
~#ps | grep venc
887 root     venc -h 127.0.0.1 -p 5600 -v 200_imx307B -s 720p -f 60 -c 264cbr -g 10 -m compact -d frame -r 4096 -n 1400 --low-delay
 1331 root     grep venc
 
It's 307, not 335

GK7205v300 / IMX335
        300_imx335F4 - v300, IMX335, 4-lane MIPI | 2592x1520  | 25  fps only
        300_imx335F5 - v300, IMX335, 4-lane MIPI | 2592x1944  | 25  fps only
        300_imx335B  - v300, IMX335, 4-lane MIPI | 1292x972   | 30 / 60  fps onl   y

venc -h 127.0.0.1 -p 5600 -v 300_imx335F4 -s 720p -f 25 -c 264cbr -g 10 -m compact -d frame -r 4096 -n 1400 --low-delay

Unable to enable even after changes，The camera will automatically restart···········

srtbit, 2023-11-10:
Loading of kernel modules...
mmz_start: 0x44000000, mmz_size: 64M
goke: Get data from U-Boot ENV and set SENSOR as imx335
insert audio
Starting rc.local
Starting FPV datalink...
Starting wifibroadcast service...
Loading modules and wifi card driver...
Detecting wifi card vendor...
Detected: realtek
Awaiting interface wlan0 in system...
Preparing interface wlan...
drone key exist...
Starting Wifibroadcast service...
Done.
Loading MAVLink telemetry service...
Done.
Starting venc encoder service...

Welcome to OpenIPC
openipc-gk7205v300 login: Using data frames
                                           Listen on 5600 for wlan0
                                                                   Using data frames
 Listen on 14550 for wlan0

cronyx, 2023-11-10:
By default, the presets in venc firmware are made for im307, if you have imx335, specify this in the /etc/venc.conf config. In the future, automatic tuning will detect the sensor and configure.

closed 2023-11-10 by cronyx