OpenIPC/firmware #1138: Hi3516EV300 flashed with OpenIPC

impala3525, 2023-11-14:
For the last couple of months I have been playing around with a ZOSI 2615-D which has the Hisilicon3516EV300 SOC. I have managed to get U-boot OpenIPC flashed to it then I tried running the setnor8 or setnor16 command and the camera reboots and I keep getting the following message

Wrong Image Format for Bootm command
Error: Cannot get kernel.

The current Flash Chip im using is a GMQ25128. U-boot detects the SPI Table

Another issue I am having is the environment variables are not saying after I run the saveenv command...after I reboot the variables reset to default.

themactep, 2023-11-14:
Did you flash just u-boot? Flash full image instead. Flash 8mb lite version and see if it works.

impala3525, 2023-11-14:
The full image.

When I attempt to use hi3516ev300-lite-8mb.bin following the installation instruction guide it applys successfully but when I reset the camera I get a SPI Nor Table not found over and over.

When this happens Im not able to get a prompt from the bootloader. And I have to use the burn method to reload OpenIPC U-boot.

After I restore the OpenIPC U-boot I then tried the Alternatively, flash OpenIPC Firmware by its parts it seems to work fine but when I get to the run setnor8m command the camera reboots..but nothing changes.  I was told this is what's causing the wrong image format for bootm command

Is it possible for someone to add flash chip support to the 8mb lite version?

impala3525, 2023-11-18:
Used SF Probe 0; SF Lock 1 now I am getting:

Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    1870941 Bytes = 1.8 MiB
   Load Address: 40008000
   Entry Point:  40008000
   Loading Kernel Image ... OK

Starting kernel ...

Booting Linux on physical CPU 0x0
Linux version 4.9.37 (runner@fv-az1198-959) (buildroot-gcc-12.2.0) #1 Tue Nov 14 23:46:11 UTC 2023
CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c53c7d
CPU: div instructions available: patching division code
CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
OF: fdt:Machine model: Hisilicon HI3516EV300 DEMO Board
cmz zone is not set!
Memory policy: Data cache writeback
CPU: All CPU(s) started in SVC mode.
Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 8128
Kernel command line: mem=32M console=ttyAMA0,115200 panic=20 root=/dev/mtdblock3 rootfstype=squashfs init=/init mtdparts=hi_sfc:256k(boot),64k(env),2048k(kernel),5120k(rootfs),-(rootfs_data)
PID hash table entries: 128 (order: -3, 512 bytes)
Dentry cache hash table entries: 4096 (order: 2, 16384 bytes)
Inode-cache hash table entries: 2048 (order: 1, 8192 bytes)
Memory: 27004K/32768K available (3894K kernel code, 148K rwdata, 916K rodata, 200K init, 181K bss, 5764K reserved, 0K cma-reserved)
Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
    vmalloc : 0xc2800000 - 0xff800000   ( 976 MB)
    lowmem  : 0xc0000000 - 0xc2000000   (  32 MB)
    modules : 0xbf000000 - 0xc0000000   (  16 MB)
      .text : 0xc0008000 - 0xc03d5ba8   (3895 kB)
      .init : 0xc04bc000 - 0xc04ee000   ( 200 kB)
      .data : 0xc04ee000 - 0xc0513360   ( 149 kB)
       .bss : 0xc0515000 - 0xc05424a0   ( 182 kB)
SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
NR_IRQS:16 nr_irqs:16 16
Gic dist init...
arm_arch_timer: Architected cp15 timer(s) running at 50.00MHz (phys).
clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xb8812736b, max_idle_ns: 440795202655 ns
sched_clock: 56 bits at 50MHz, resolution 20ns, wraps every 4398046511100ns
Switching to timer-based delay loop, resolution 20ns
clocksource: arm,sp804: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 637086815595 ns
Console: colour dummy device 80x30
Calibrating delay loop (skipped), value ca

System startup

Uncompress Ok!

U-Boot 2016.11-g6463078-dirty (Oct 28 2023 - 00:17:57 +0000)hi3516ev300

Relocation Offset is: 0774b000
Relocating to 47f4b000, new gd at 47b3aef0, sp at 47b3aed0
SPI Nor:  hifmc_ip_ver_check(44): Check Flash Memory Controller v100 ...hifmc_ip_ver_check(50):  Found
hifmc_spi_nor_probe(1844): SPI Nor ID Table Version 1.0
hifmc_spi_nor_probe(1869): SPI Nor(cs 0) ID: 0x1c 0x40 0x18
hifmc_spi_nor_probe(1934): Block:64KB hifmc_spi_nor_probe(1935): Chip:16MB hifmc_spi_nor_probe(1936): Name:"GM25Q128A"
Spi is locked. lock address[0 => 0xff0000]
hifmc100_spi_nor_probe(146): SPI Nor total size: 16MB
NAND:  0 MiB
MMC:   Card did not respond to voltage select!
No SD device found !
hisi-sdhci: 0
*** Warning - bad CRC, using default environment

Erasing SPI flash...
ERROR: The erase area was locked.
In:    serial
Out:   serial
[... 164 lines cut ...]
010f           65536 ram15  (driver?)
Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)
CPU: 0 PID: 1 Comm: swapper Not tainted 4.9.37 #1
Hardware name: Generic DT based system
Backtrace:
[<c0013e2c>] (dump_backtrace) from [<c001401c>] (show_stack+0x18/0x1c)
 r7:00008001 r6:c044e826 r5:c05152e8 r4:00000000
[<c0014004>] (show_stack) from [<c01ae944>] (dump_stack+0x20/0x28)
[<c01ae924>] (dump_stack) from [<c00658e0>] (panic+0xcc/0x238)
[<c0065818>] (panic) from [<c04bd3a4>] (mount_block_root+0x20c/0x2a8)
 r3:00000000 r2:00000000 r1:c1839ea4 r0:c044e826
 r7:00008001
[<c04bd198>] (mount_block_root) from [<c04bd648>] (mount_root+0xf0/0x134)
 r10:c04bc4ec r9:c04ec698 r8:c04e2854 r7:c04e2834 r6:c0515024 r5:c04f394c
 r4:00000000
[<c04bd558>] (mount_root) from [<c04bd80c>] (prepare_namespace+0x180/0x1c0)
 r9:c04ec698 r8:c04e2854 r7:c04e2834 r6:00000008 r5:c0515024 r4:c04e2858
[<c04bd68c>] (prepare_namespace) from [<c04bcea4>] (kernel_init_freeable+0x158/0x1b8)
 r6:00000008 r5:c0515000 r4:00000061
[<c04bcd4c>] (kernel_init_freeable) from [<c03d02a4>] (kernel_init+0x14/0xf4)
 r10:00000000 r9:00000000 r8:00000000 r7:00000000 r6:00000000 r5:c03d0290
 r4:c0515000
[<c03d0290>] (kernel_init) from [<c000f068>] (ret_from_fork+0x14/0x2c)
 r5:c03d0290 r4:00000000
Rebooting in 20 seconds..