OpenIPC/firmware #1152: Speaker volume not working on Ingenic t31x / t31zx / wyze cam v3

aarnaud, 2023-11-20:
Hi,

Thanks for this great project. Very appreciate. 

I tried to control the speaker volume, but nothing works, with the default GPIO51 or the one present in the wyze documentation GPIO63. 
I tried rebooting too. 

I saw commits from @gtxaspec that add libimp_contro, can that help me to adjust the volume, I saw the option. 

there is missing execution flag on `/usr/sbin/imp-control.sh`  and nothing listen on port 4000 yet.

Thanks,

gtxaspec, 2023-11-20:
Hi, the feature has not completely been enabled yet, as there is still work to be done on the webui side.

If you want to enable it in the mean time, read the PR instructions on editing the majestic init script, and set the execution flag on the control applet script.

More PRs fully enabling will be coming soon

aarnaud, 2023-11-20:
Nice, it's working, thanks

Ref: https://github.com/OpenIPC/firmware/pull/1148

closed 2023-11-20 by aarnaud

aarnaud, 2023-11-20:
Badly the update broke 2 of my cameras:
```
16:19:30  <       majestic> [    main] main@123                      Majestic Lite for Ingenic, version master+b5b5dfe, built on 2023-11-17
16:19:30  <       majestic> [app_conf] load_config@108               Using /etc/majestic.yaml as main configuration
16:19:31  <       majestic> [watchdog] watchdog_start@101            Detected jz Watchdog
16:19:31  <       majestic> [watchdog] watchdog_start@121            Watchdog timeout set to 10 seconds
16:19:31  <       majestic> [sns_conf] try_sensor_config@13          SENSOR=gc2053
16:19:31  <       majestic> [sns_conf] load_sensor_config@44         Using /etc/sensor/gc2053.yaml as sensor config
16:19:31  <       majestic> [sns_conf] parse_sensor_config@125       VI: gc2053 @ 0x37 [1920x1080]
16:19:31  <       majestic> [     sdk] start_sdk@1254                CPU T31-ZX, SDK SYSUTILS-1.1.6, model Swan, devID 16180006
16:19:31  <       majestic> [     sdk] imp_system_init@326           failed to open ISP
16:19:31  <       majestic> [     sdk] start_sdk@1363                IMP_System_Init() failed
16:19:31  <       majestic> [    main] main@207                      Shutdown main thread
```

Both have `T31-ZX` instead `T31-X`

gtxaspec, 2023-11-20:
Can you share output of `logcat` ?  Did you enable ld_preload?

aarnaud, 2023-11-20:
I tried with LD_PRELOAD and without:
Is there a way to find  and download an old  openipc.t31-nor-lite.tgz  version from 2nd november ?  

logcat
```
Nov 19 16:07:07 backstreet-2 syslog.info syslogd started: BusyBox v1.36.1
Nov 19 16:07:07 backstreet-2 kern.notice kernel: klogd started: BusyBox v1.36.1 (2023-11-19 20:57:04 UTC)
Nov 19 16:07:07 backstreet-2 kern.info kernel: [    0.000000] Initializing cgroup subsys cpu
Nov 19 16:07:07 backstreet-2 kern.info kernel: [    0.000000] Initializing cgroup subsys cpuacct
Nov 19 16:07:07 backstreet-2 kern.notice kernel: [    0.000000] Linux version 3.10.14__isvp_swan_1.0__ (runner@fv-az1040-956) (buildroot-gcc-12.2.0) #2 PREEMPT Sun Nov 19 21:03:41 UTC 2023
Nov 19 16:07:07 backstreet-2 kern.info kernel: [    0.000000] bootconsole [early0] enabled
Nov 19 16:07:07 backstreet-2 kern.err kernel: [    0.000000] CPU0 RESET ERROR PC:801BE684
Nov 19 16:07:07 backstreet-2 kern.err kernel: [    0.000000] [<801be684>] __delay+0x4/0x10
Nov 19 16:07:07 backstreet-2 kern.info kernel: [    0.000000] CPU0 revision is: 00d00100 (Ingenic Xburst)
Nov 19 16:07:07 backstreet-2 kern.info kernel: [    0.000000] FPU revision is: 00b70000
Nov 19 16:07:07 backstreet-2 kern.debug kernel: [    0.000000] cgu_get_rate, parent = 1392000000, rate = 0, m = 129, n = 255, reg val = 0x081000ff
Nov 19 16:07:07 backstreet-2 kern.debug kernel: [    0.000000] cgu_get_rate, parent = 1392000000, rate = 0, m = 129, n = 255, reg val = 0x081000ff
Nov 19 16:07:07 backstreet-2 kern.err kernel: [    0.000000] CCLK:1392MHz L2CLK:696Mhz H0CLK:200MHz H2CLK:200Mhz PCLK:100Mhz
Nov 19 16:07:07 backstreet-2 kern.info kernel: [    0.000000] Determined physical RAM map:
Nov 19 16:07:07 backstreet-2 kern.info kernel: [    0.000000]  memory: 00554000 @ 00010000 (usable)
Nov 19 16:07:07 backstreet-2 kern.info kernel: [    0.000000]  memory: 0003c000 @ 00564000 (usable after init)
Nov 19 16:07:07 backstreet-2 kern.info kernel: [    0.000000] User-defined physical RAM map:
Nov 19 16:07:07 backstreet-2 kern.info kernel: [    0.000000]  memory: 04000000 @ 00000000 (usable)
Nov 19 16:07:07 backstreet-2 kern.info kernel: [    0.000000] Initrd not found or empty - disabling initrd
Nov 19 16:07:07 backstreet-2 kern.err kernel: [    0.000000] Zone ranges:
Nov 19 16:07:07 backstreet-2 kern.err kernel: [    0.000000]   Normal   [mem 0x00000000-0x03ffffff]
Nov 19 16:07:07 backstreet-2 kern.err kernel: [    0.000000] Movable zone start for each node
Nov 19 16:07:07 backstreet-2 kern.err kernel: [    0.000000] Early memory node ranges
Nov 19 16:07:07 backstreet-2 kern.err kernel: [    0.000000]   node   0: [mem 0x00000000-0x03ffffff]
Nov 19 16:07:07 backstreet-2 kern.debug kernel: [    0.000000] On node 0 totalpages: 16384
Nov 19 16:07:07 backstreet-2 kern.debug kernel: [    0.000000] free_area_init_node: node 0, pgdat 8055e050, node_mem_map 81000000
Nov 19 16:07:07 backstreet-2 kern.debug kernel: [    0.000000]   Normal zone: 128 pages used for memmap
Nov 19 16:07:07 backstreet-2 kern.debug kernel: [    0.000000]   Normal zone: 0 pages reserved
Nov 19 16:07:07 backstreet-2 kern.debug kernel: [    0.000000]   Normal zone: 16384 pages, LIFO batch:3
Nov 19 16:07:07 backstreet-2 kern.err kernel: [    0.000000] Primary instruction cache 32kB, 8-way, VIPT, linesize 32 bytes.
Nov 19 16:07:07 backstreet-2 kern.err kernel: [    0.000000] Primary data cache 32kB, 8-way, VIPT, no aliases, linesize 32 bytes
Nov 19 16:07:07 backstreet-2 kern.err kernel: [    0.000000] pls check processor_id[0x00d00100],sc_jz not support!
Nov 19 16:07:07 backstreet-2 kern.err kernel: [    0.000000] MIPS secondary cache 128kB, 8-way, linesize 32 bytes.
Nov 19 16:07:07 backstreet-2 kern.debug kernel: [    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
Nov 19 16:07:07 backstreet-2 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 
Nov 19 16:07:07 backstreet-2 kern.err kernel: [    0.000000] Built 1 zonelists in Zone order, mobility grouping off.  Total pages: 16256
Nov 19 16:07:07 backstreet-2 kern.notice kernel: [    0.000000] Kernel command line: mem=64M rmem=64M@0x4000000 console=ttyS1,115200n8 panic=20 root=/dev/mtdblock3 rootfstype=squashfs init=/init mtdparts=jz_sfc:256k(boot),64k(env),2048k(kernel),5120k(rootfs),-(rootfs_data) nog
Nov 19 16:07:07 backstreet-2 kern.info kernel: [    0.000000] PID hash table entries: 256 (order: -2, 1024 bytes)
Nov 19 16:07:07 backstreet-2 kern.info kernel: [    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
Nov 19 16:07:07 backstreet-2 kern.info kernel: [    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
Nov 19 16:07:07 backstreet-2 kern.info kernel: [    0.000000] Memory: 58768k/65536k available (4049k kernel code, 6768k reserved, 1402k data, 240k init, 0k highmem)
Nov 19 16:07:07 backstreet-2 kern.info kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
Nov 19 16:07:07 backstreet-2 kern.info kernel: [    0.000000] Preemptible hierarchical RCU implementation.
Nov 19 16:07:07 backstreet-2 kern.info kernel: [    0.000000] NR_IRQS:358
Nov 19 16:07:07 backstreet-2 kern.err kernel: [    0.000000] clockevents_config_and_register success.
Nov 19 16:07:07 backstreet-2 kern.info kernel: [    0.000014] Calibrating delay loop... 1391.00 BogoMIPS (lpj=6955008)
Nov 19 16:07:07 backstreet-2 kern.info kernel: [    0.087831] pid_max: default: 32768 minimum: 301
[... cut ...]
Nov 19 16:07:17 backstreet-2 kern.err kernel: [   12.308441] [atbm_log]:wlan0: authenticated
Nov 19 16:07:17 backstreet-2 kern.err kernel: [   12.308877] [atbm_log]:wlan0:free authen bss ++
Nov 19 16:07:17 backstreet-2 kern.err kernel: [   12.333019] [atbm_log]:wlan0:free authen bss --
Nov 19 16:07:17 backstreet-2 kern.err kernel: [   12.344663] [atbm_log]:wlan0: associated
Nov 19 16:07:17 backstreet-2 kern.info kernel: [   12.348608] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready

```

gtxaspec, 2023-11-20:
That's `logread` ... I need `logcat` for diagnostics.  Old builds are available here https://t.me/openipc_dev

gtxaspec, 2023-11-20:
Ah I see you are on lite, not included there currently.
[logcat.zip](https://github.com/OpenIPC/firmware/files/13418791/logcat.zip)
You can transfer this to the device after extraction and run it

aarnaud, 2023-11-20:
Update: I did a firmware-reset on one of them it's fix the issue

aarnaud, 2023-11-20:
unfortunately after transfer:
```
~# logcat 
Error loading shared library libstdc++.so.6: No such file or directory (needed by /sbin/logcat)
Error relocating /sbin/logcat: _ZdlPvj: symbol not found
Error relocating /sbin/logcat: _Znwj: symbol not found
Error relocating /sbin/logcat: __gxx_personality_v0: symbol not found
Error relocating /sbin/logcat: __gxx_personality_v0: symbol not found
```

aarnaud, 2023-11-20:
So I tried to downgrade, but during the process `verification mismatch`:
```
SoC OK
Skip version checking
Erasing block: 160/160 (100%) 
Writing kb: 5116/5116 (100%) 
Verifying kb: 1312/5116 (25%) flashcp: verification mismatch at 0x148000
```
I tried to reflash with UART and sdcard by following https://github.com/themactep/device-wyze-v3/blob/master/installation.md
but now the boot stop here :

```
DDRC_REMAP5         0x13121110
DDRC_AUTOSR_EN      0x00000000
sdram init finished
SDRAM init ok
board_init_r
image entry point: 0x80100000
```
Is it bad ?  What can I do ?

themactep, 2023-11-20:
Did you select the proper letter for the first card?

themactep, 2023-11-20:
Ah, it was during the upgrade. I see. Connect camera with a trusted usb cable to your pc and reflash it using cloner.

themactep, 2023-11-20:
https://openipc.org/utilities
Top link on the page.

aarnaud, 2023-11-21:
Thanks a lot for all the share here,  

I have question to confirm:
- what is a trusted usb cable  ? USB cable with data is good ?
- where can I found the correct partition value ? 
- Do I need to put the camera in a special mode an how ? because I don't see any usb/tty device match the camera. 

I also order a  SOIC8 SOP8 Flash Chip IC in case of to flash directly the ship, If I have no choice.

aarnaud, 2023-11-21:
So I success to have u-boot working and reflash the camera but It's booting loop with :
```
the manufacturer 1c
SF: Detected EN25QH128A

--->probe spend 4 ms
SF: 3145728 bytes @ 0x50000 Read: OK
--->read spend 1010 ms
## Booting kernel from Legacy Image at 80600000 ...
   Image Name:   Linux-3.10.14__isvp_swan_1.0__-t
   Image Type:   MIPS Linux Kernel Image (lzma compressed)
   Data Size:    1867683 Bytes = 1.8 MiB
   Load Address: 80010000
   Entry Point:  803f7770
   Verifying Checksum ... Bad Data CRC
ERROR: can't get kernel image!
reset in 4ms

```

aarnaud, 2023-11-21:
Reflash it, and it's fix. 
Is the memory dying ?

aarnaud, 2023-11-21:
Trying the sysupgrade and `verification mismatch` at the end

```
> sysupgrade -k -r 
OpenIPC System Updater v1.0.21

Vendor	ingenic
SoC	t31
Kernel	21:55:28 2023-11-02
RootFS	master+e1ba943, 2023-11-02

Synchronizing time
ntpd: setting time to 2023-11-21 06:55:43.578673 (offset +1587613.742212s)
Tue Nov 21 06:55:43 GMT 2023

Checking for sysupgrade update...
Same version. No update required.

Stop services, sync files, free up memory
Killing majestic ............................ OK
Killing crond ......... OK
Killing klogd .......... OK
Killing ntpd .......... OK
Killing rngd .......... OK
Killing syslogd ......... OK

Uptime:
 06:55:52 up 2 min,  load average: 1.56, 0.67, 0.25

Memory:
              total        used        free      shared  buff/cache   available
Mem:          59008        8284       47608           0        3116           0
-/+ buffers/cache:         8284       50724
Swap:             0           0           0

Processes:
PID   USER     COMMAND
    1 root     init
  574 root     wpa_supplicant -B -i wlan0 -D nl80211,wext -c /tmp/wpa_supplicant.conf
  594 root     udhcpc -x hostname:openipc-t31 -A 0 -T 1 -t 1 -R -b -O search -p /var/run/udhcpc.wlan0.pid -i wlan0
  651 root     /usr/sbin/dropbear -R -B -k -p 22 -K 300
  657 root     /usr/sbin/httpd httpd -p 85 -f -c /etc/httpd.conf -r Authentication
  866 root     -sh
  962 root     {sysupgrade} /bin/sh /usr/sbin/sysupgrade -k -r
 1146 root     ps

Unmounting SD card
/mnt/mmcblk0p1
- checking for /mnt/mmcblk0p1/autoupdate-kernel.img
- checking for /mnt/mmcblk0p1/autoupdate-rootfs.img
- checking for /mnt/mmcblk0p1/autoupdate-uboot.img

Firmware
Download from https://github.com/OpenIPC/firmware/releases/download/latest/openipc.t31-nor-lite.tgz
Received and unpacked

Kernel
Update kernel from /tmp/uImage.t31
New version, going to update
Erasing block: 58/58 (100%) 
Writing kb: 1824/1824 (100%) 
Verifying kb: 1824/1824 (100%) 
Kernel updated to 21:03:44 2023-11-19

RootFS
Update rootfs from /tmp/rootfs.squashfs.t31
SoC OK
New version, going to update
Erasing block: 160/160 (100%) 
Writing kb: 5116/5116 (100%) 
Verifying kb: 1312/5116 (25%) flashcp: verification mismatch at 0x148000
```

aarnaud, 2023-11-21:
@gtxaspec   about the issue for Majestic

I thinks it's because the overlayfs, first an update with wipe don't have issue, 
second:
```
/overlay/lib:
total 0
drwxr-xr-x    3 root     root             0 Nov 15 10:16 modules

/overlay/lib/modules:
total 0
drwxr-xr-x    3 root     root             0 Nov 15 10:20 3.10.14__isvp_swan_1.0__

/overlay/lib/modules/3.10.14__isvp_swan_1.0__:
total 3
drwxr-xr-x    2 root     root             0 Nov 15 10:18 extra
-rw-r--r--    1 root     root          2343 Nov 15 10:18 modules.dep

/overlay/lib/modules/3.10.14__isvp_swan_1.0__/extra:
total 2656
-rwxr-xr-x    1 root     root       1566900 Nov 15 10:18 8189fs.ko
-rwxr-xr-x    1 root     root         22368 Nov 15 10:18 sensor_gc2053_t31.ko
-rwxr-xr-x    1 root     root       1129876 Nov 15 10:18 tx-isp-t31.ko
```

So I dirty copy past the module and it's working

themactep, 2023-11-21:
> Writing kb: 5116/5116 (100%) 
> Verifying kb: 1312/5116 (25%) flashcp: verification mismatch at 0x148000

That sounds like a faulty flash chip or bad soldering. You can touch up legs of the flash chip with a soldering iron to reflow solder and make sure all contacts are good.

aarnaud, 2023-11-21:
I found the issue that break Majestic after an update, because wifi driver modified  `modules.dep` so the content doesn't match anymore file path from the update:
[image]

Axon07, 2023-11-27:
I am having the same issue with my V3. Will there be an update that will fix the Majestic issue?

aarnaud, 2023-11-27:
Meanwhile with a recent update you can adjust the volume in cli by following the config in the PR

Ref: https://github.com/OpenIPC/firmware/pull/1148

```
sh /usr/sbin/imp-control.sh  aovol 80
```

Axon07, 2023-11-27:
> @gtxaspec I found the issue that break Majestic after an update, because wifi driver modified `modules.dep` so the content doesn't match anymore file path from the update: ![image](https://private-user-images.githubusercontent.com/4802105/284629580-abe1338e-c2c5-4a95-8037-7afe76a35ee0.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDExMjAyNDksIm5iZiI6MTcwMTExOTk0OSwicGF0aCI6Ii80ODAyMTA1LzI4NDYyOTU4MC1hYmUxMzM4ZS1jMmM1LTRhOTUtODAzNy03YWZlNzZhMzVlZTAucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQUlXTkpZQVg0Q1NWRUg1M0ElMkYyMDIzMTEyNyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyMzExMjdUMjExOTA5WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9OWU4MmQ2NjZjMGM4OGFhMGI1NzdhMmQ0MGE3ZGJjODU3OGYyYzdkNjkyN2VmODdjYjg4NjA5ODhmZWQ3MmIwOCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.U3WesR87l5gDTBgvRMBPyd17LpeFFaZPdSGehEmby8M)

I had the same issue with an update breaking Majestic.

aarnaud, 2023-11-27:
Okay I see, I success to handle update by using `--no_reboot` option and manually edit `modules.dep` before rebooting. but yeah not ideal.
I don't know really how we can be handle with sysupgrade. 
It's still better solution for me than Cloud service offer by camera manufacturers.