OpenIPC/firmware #1157: Milk-V Duo, a new opensource hardware may be ported to OpenIPC

Ncerzzk, 2023-11-23:
### Here are some main feature:
- 1GHz and 700MHz RISC-V C906 processors
- Integrated CVITEK TPU for smart detection.
- Supports H.264/H.265 video encoding, up to 2880x1620@20fps.
- Compatible with high-definition CMOS sensors.

### Other advantages:
- enough cheap with $5 board and $3.5 sensor(gc2083)
- enough small to DIY
- totally opensource hardware
- package of the chip cv1800b(QFN) is easy to solder(if you want to make it smaller or use other camera)

### Get SDK:
https://github.com/milkv-duo/duo-buildroot-sdk

### Current process:
for now I just successfully boot and run its camera test program. 
and because it's risc-v arch,  the majestic streamer  may be need to build a risc-v target?

https://milkv.io/assets/images/duo-v1.2-9bf1d36ef7632ffba032796978cda903.png

FlyRouter, 2023-11-23:
Good afternoon
Thanks for the interesting note.
We can accept PR if you send us a verified organizational structure compiled in OpenIPC Buildroot
As for the streamer, you first need to launch and publish examples from the SDK so that at least snapshots and other solutions work.

widgetii, 2023-11-24:
Streamer CI uses OpenIPC based toolchain as well, so the first and necessary step is to integrate vendor BSP to OpenIPC firmware repo

Ncerzzk, 2023-11-25:
Thank you @widgetii  @FlyRouter. I would start the first step recently.

Ncerzzk, 2023-12-03:
issue close. I will update the status in pull request.

closed 2023-12-03 by Ncerzzk