OpenIPC/firmware #1181: RFQ: gk7205v300 - ICSEE Dome camera

hvegh, 2023-12-04:
Recently I bought this [Icsee 5mp Poe Ip Ptz Camera 4x optical Zoom](https://nl.aliexpress.com/item/1005006157233218.html) IP cam.

The camera controller is connected via the UART to the PTZ controller board.

- Is there a way that we can run OpenIPC on this device?
- How can I help?

I have managed to extract the firmware from the device.

Here is a bootlog:
```
System startup

Uncompress Ok!

U-Boot 2016.11-g4bd9c94-dirty (Apr 22 2022 - 21:25:43 +0800)gk7205v300

Relocation Offset is: 0771f000
Relocating to 47f1f000, new gd at 47edeef0, sp at 47edeed0
SPI Nor:  Check Flash Memory Controller v100 ... Found
SPI Nor ID Table Version 1.0
@hifmc_spi_nor_probe(), SPI Nor(cs 0) ID: 0xa1 0x40 0x18 <Read>
@hifmc_spi_nor_probe(), SPI Nor(cs 0) ID: 0xa1 0x40 0x18 <Found>
SPI Nor(cs 0) ID: 0xa1 0x40 0x18
eFlashType: 30.
@XmSpiNor_initSr3() need No SR3!
@XmSpiNor_printWps(), WPS Enabled!
Flash Name: XM_FM25Q128A{0xA14018), 0x1000000.
@hifmc_spi_nor_probe(), XmSpiNor_ProtMgr_probe(): OK.
@XmSpiNor_enableQuadMode(), Quad was Disabled, SRx: [2, 0x8].
Block:64KB Chip:16MB Name:"XM_FM25Q128A"
CONFIG_CLOSE_SPI_8PIN_4IO = y.
read->iftype[0: STD, 1: DUAL, 2: DIO, 3: QUAD, 4: QIO]: 1.
lk[7 => 0x1000000], Total Lock Blks: 286 <WPS=1>
SRx val: {[1, 0x20], [1, 0x8], [1, 0x0], [0, 0x0]}, SrVal: 0x700000000000820.
SPI Nor total size: 16MB
In:    serial
Out:   serial
Err:   serial
Net:   eth0
autoboot:  0 
@do_spi_flash_probe() flash->erase_size: 65536, flash->sector_size: 0
device 0 offset 0x40000, size 0x540000

SF: 5505024 bytes @ 0x40000 Read: OK
srcAddr: 0x43000000, dstAddr: 0x42000000, filename: boot/uImage.
created_inode 0x47edfc18
find_squashfs_file: name bin, start_block 0, offset 1824, type 1
find_squashfs_file: name boot, start_block 0, offset 1996, type 1
read inode: name boot, sb 0, of 1996, type 1
find_squashfs_file: name uImage, start_block 0, offset 1856, type 2
read inode: name uImage, sb 0, of 1856, type 2
write_file: regular file, blocks 27
len 1723581
### get_squashfs_file OK: loade 1723581 bytes to 0x42000000
## Booting kernel from Legacy Image at 42000000 ...
   Image Name:   Linux-4.9.37
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    1723517 Bytes = 1.6 MiB
   Load Address: 40008000
   Entry Point:  40008000
   Loading Kernel Image ... OK
using: ATAGS
at setup_xminfo_tag() g_nXmBootSysIndex: 0, g_nXmRomfsIndex: 0.

Starting kernel ...
```

stipe-jurkovic, 2023-12-04:
I have the same one, with the dirty. Managed to install openipc bit now i am having no video.

hvegh, 2023-12-04:
I have 5MP, also Sc5239P I suspect.

Tried to access the bootloader by shorting the clk/data pin, but is password protected. 
So I just de-soldered the flash chip and read the contents:

```
./40000.squashfs
./40000/bin
./40000/bin/EnvPartCheck
./40000/bin/XmDvrBox
./40000/bin/ash
./40000/bin/awk
./40000/bin/btools
./40000/bin/busybox
./40000/bin/cat
./40000/bin/chmod
./40000/bin/cp
./40000/bin/cut
./40000/bin/date
./40000/bin/dmesg
./40000/bin/dumpleases
./40000/bin/echo
./40000/bin/free
./40000/bin/grep
./40000/bin/gunzip
./40000/bin/gzip
./40000/bin/head
./40000/bin/hiddrs
./40000/bin/himc
./40000/bin/himd
./40000/bin/himd.l
./40000/bin/himm
./40000/bin/i2c_read
./40000/bin/i2c_write
./40000/bin/kill
./40000/bin/killall
./40000/bin/ln
./40000/bin/login
./40000/bin/ls
./40000/bin/mkdir
./40000/bin/mkfifo
./40000/bin/mknod
./40000/bin/mount
./40000/bin/mv
./40000/bin/netinit
./40000/bin/netstat
./40000/bin/ping
./40000/bin/ping6
./40000/bin/ps
./40000/bin/pwd
./40000/bin/rm
./40000/bin/rmdir
./40000/bin/sed
./40000/bin/sh
./40000/bin/sleep
./40000/bin/ssp_read
./40000/bin/ssp_write
./40000/bin/sync
./40000/bin/tar
./40000/bin/top
./40000/bin/touch
./40000/bin/tty
./40000/bin/umount
./40000/bin/unlink
./40000/bin/upgrader
./40000/bin/xmmm
./40000/boot
./40000/boot/uImage
./40000/dev
./40000/etc
[... 319 lines cut ...]
./E40000/data/Strings/French
./E40000/data/Strings/German
./E40000/data/Strings/Greek
./E40000/data/Strings/Hebrew
./E40000/data/Strings/Hungarian
./E40000/data/Strings/Indonesian
./E40000/data/Strings/Italian
./E40000/data/Strings/Japanese
./E40000/data/Strings/Poland
./E40000/data/Strings/Portugal
./E40000/data/Strings/Romanian
./E40000/data/Strings/Russian
./E40000/data/Strings/SimpChinese
./E40000/data/Strings/Slovakia
./E40000/data/Strings/Spanish
./E40000/data/Strings/Swedish
./E40000/data/Strings/Thai
./E40000/data/Strings/TradChinese
./E40000/data/Strings/Turkey
./E40000/data/Strings/Vietnamese
./E40000/xm_version
./E40000/xm_version/buildtime
./flash.bin
./u-boot.bin
```

stipe-jurkovic, 2023-12-04:
I used the openipc burn utility, worked well

widgetii, 2026-08-16:
Closing this one — it was answered, and the answer is worth stating plainly for anyone who finds it by searching.

**The board works with OpenIPC.** @stipe-jurkovic got it running with the `burn` utility. @hvegh confirmed the hardware by desoldering the flash and reading it: `gk7205v300`, XM_FM25Q128A 16 MB NOR (`0xa1 0x40 0x18`), Xiongmai `XmDvrBox` firmware, SC5239P sensor.

### Current instructions

`burn` has been replaced by [defib](https://github.com/OpenIPC/defib). For this board you don't need to desolder anything:

- **Bootloader password.** `#Ux6@9V&4_Rz` is confirmed working on a lot of gk7205v300 boards (OpenIPC/firmware#1494). `HI2105CHIP` also turns up on Xiongmai-derived builds of this vintage (OpenIPC/firmware#993). Either gets you a console without hardware work.
- **If neither is accepted**, gk7205v300 is fully supported in the browser build — no install, no desoldering, and it bypasses the vendor bootloader entirely: <https://openipc.github.io/defib/>
- **Back up first.** The **Dump Flash** button saves the stock firmware to your computer. @hvegh had to desolder the chip to get that; it's one button now.

Full install guide: <https://openipc.org/cameras/vendors/goke/socs/gk7205v300>

### One thing to watch

Your boot log shows the vendor bootloader armed the flash's status-register protection. If a later `sf erase` reports success but nothing changes, that's why — see OpenIPC/u-boot-gk7205v200#18. `defib agent flash` clears those lock bits explicitly and verifies what it wrote.

@stipe-jurkovic — your "no video after install" is a sensor configuration problem, not a flashing one. SC5239P needs the right sensor profile selected; if it's still unresolved after all this time, please open a separate issue with your `/etc/majestic.yaml` and the output of `ipctool`.

Thanks @hvegh for the flash dump and the hardware detail — that's what made the board identifiable.

closed 2026-08-16 by widgetii