OpenIPC/firmware #1195: RFQ: supporting RISC-V architecture based Milk-V Duo

milkrisc-v, 2023-12-13:
Milk-V Duo is an ultra-compact embedded development platform based on the CV1800B chip with two RISC-V Cores, one C906 running Linux and one C906 running RTOS. 

- Integrated CVITEK TPU for smart detection.
- Supports H.264/H.265 video encoding, up to 2880x1620@20fps
- Features a 16-pin FPC interface for 2-lane MIPI camera input
- Up to 4M Pixel ISP support

milkrisc-v, 2023-12-13:
We(Milk-V) are very happy to donate Duos and Cameras to have OpenIPC support. If the team is interested, please contact me at: v@milkv.io .

FlyRouter, 2023-12-14:
Good afternoon
Thank you very much for your suggestion.
We as a team will think about it, read the documentation and I will contact you a little later.

piotr-defendeye, 2023-12-19:
Yes! Support for MilkV would be amazing! Really looking forward to it!

piotr-defendeye, 2024-08-01:
Hi! Any success with that? :D