OpenIPC/firmware #1209: How to install rtl8188fu & rtl8812au on Ingenic T21?

kangz543g, 2023-12-25:
please answer about my question.

no sudo command??? 

 sudo
-sh: sudo: not found

install rtl8188fu & rtl8812au???

kangz543g, 2023-12-26:
../rtl8812au-openipc/os_dep/linux/wifi_regd.c:339:99: error: expected statement before ‘)’ token
make[4]: *** [scripts/Makefile.build:308: /home/kkk/Downloads/firmware-master/output/build/rtl8812au-openipc/./os_dep/linux/wifi_regd.o] Error 1
make[3]: *** [Makefile:1228: _module_/home/kkk/Downloads/firmware-master/output/build/rtl8812au-openipc/.] Error 2
make[2]: *** [package/pkg-generic.mk:284: /home/kkk/Downloads/firmware-master/output/build/rtl8812au-openipc/.stamp_built] Error 2
make[1]: *** [Makefile:82: _all] Error 2
make[1]: Leaving directory '/home/kkk/Downloads/firmware-master/output/buildroot-2023.02.7'
make: *** [Makefile:40: br-rtl8812au-openipc-rebuild] Error 2

Please Error????

themactep, 2023-12-27:
Can you be any more su than root? :)

themactep, 2023-12-27:
Where did you get the module? Is it cross-compiled for mips architecture?

kangz543g, 2023-12-27:
ubuntu 20.04 is compiler
make br-rtl8812au-openipc-rebuild BOARD=t21
An error occurred? There is no solution.

kangz543g, 2023-12-27:
kkk@kkk-Lenovo-ideapad-100-15IBY:~/Downloads/firmware$ make br-rtl8812au-openipc-rebuild BOARD=t21
--- br-ext-chip-ingenic/configs/t21_lite_defconfig
make[1]: Entering directory '/home/kkk/Downloads/firmware/output/buildroot-2023.02.7'
  GEN     /home/kkk/Downloads/firmware/output/Makefile
#
# configuration written to /home/kkk/Downloads/firmware/output/.config
#
make[1]: Leaving directory '/home/kkk/Downloads/firmware/output/buildroot-2023.02.7'
make[1]: Entering directory '/home/kkk/Downloads/firmware/output/buildroot-2023.02.7'
rm -f /home/kkk/Downloads/firmware/output/build/rtl8812au-openipc/.stamp_installed
rm -f /home/kkk/Downloads/firmware/output/build/rtl8812au-openipc/.stamp_staging_installed
rm -f /home/kkk/Downloads/firmware/output/build/rtl8812au-openipc/.stamp_target_installed
rm -f /home/kkk/Downloads/firmware/output/build/rtl8812au-openipc/.stamp_images_installed
rm -f /home/kkk/Downloads/firmware/output/build/rtl8812au-openipc/.stamp_host_installed
rm -f /home/kkk/Downloads/firmware/output/build/rtl8812au-openipc/.stamp_built
/usr/bin/make -j1  O=/home/kkk/Downloads/firmware/output HOSTCC="/usr/lib/ccache/gcc" HOSTCXX="/usr/lib/ccache/g++" syncconfig
  GEN     /home/kkk/Downloads/firmware/output/Makefile
for f in /home/kkk/Downloads/firmware/general/../br-ext-chip-ingenic/board/t21/kernel/t21.generic.config ; do if [ ! -f "${f}" ]; then printf "Kconfig file or fragment '%s' for '%s' does not exist\n" "${f}" "linux"; exit 1; fi; done
mkdir -p /home/kkk/Downloads/firmware/output/per-package/linux/host
rsync -a --hard-links  --link-dest=/home/kkk/Downloads/firmware/output/per-package/toolchain/host/ /home/kkk/Downloads/firmware/output/per-package/toolchain/host/ /home/kkk/Downloads/firmware/output/per-package/linux/host
mkdir -p /home/kkk/Downloads/firmware/output/per-package/linux/target
rsync -a --hard-links  --link-dest=/home/kkk/Downloads/firmware/output/per-package/toolchain/target/ /home/kkk/Downloads/firmware/output/per-package/toolchain/target/ /home/kkk/Downloads/firmware/output/per-package/linux/target
Using /home/kkk/Downloads/firmware/output/build/linux-custom/.config as base
#
# merged configuration written to /home/kkk/Downloads/firmware/output/build/linux-custom/.config (needs make)
#
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/kconfig/conf.o
  HOSTCC  scripts/kconfig/zconf.tab.o
In file included from scripts/kconfig/zconf.tab.c:2500:
scripts/kconfig/confdata.c: In function ‘conf_write’:
scripts/kconfig/confdata.c:762:19: warning: ‘%s’ directive writing likely 7 or more bytes into a region of size between 1 and 4097 [-Wformat-overflow=]
  762 |  sprintf(newname, "%s%s", dirname, basename);
      |                   ^~~~~~
scripts/kconfig/confdata.c:762:19: note: assuming directive output of 7 bytes
In file included from /usr/include/stdio.h:867,
                 from scripts/kconfig/zconf.tab.c:85:
/usr/include/x86_64-linux-gnu/bits/stdio2.h:36:10: note: ‘__builtin___sprintf_chk’ output 1 or more bytes (assuming 4104) into a destination of size 4097
   36 |   return __builtin___sprintf_chk (__s, __USE_FORTIFY_LEVEL - 1,
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   37 |       __bos (__s), __fmt, __va_arg_pack ());
      |       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from scripts/kconfig/zconf.tab.c:2500:
scripts/kconfig/confdata.c:765:20: warning: ‘.tmpconfig.’ directive writing 11 bytes into a region of size between 1 and 4097 [-Wformat-overflow=]
  765 |   sprintf(tmpname, "%s.tmpconfig.%d", dirname, (int)getpid());
      |                    ^~~~~~~~~~~~~~~~~
In file included from /usr/include/stdio.h:867,
                 from scripts/kconfig/zconf.tab.c:85:
/usr/include/x86_64-linux-gnu/bits/stdio2.h:36:10: note: ‘__builtin___sprintf_chk’ output between 13 and 4119 bytes into a destination of size 4097
   36 |   return __builtin___sprintf_chk (__s, __USE_FORTIFY_LEVEL - 1,
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   37 |       __bos (__s), __fmt, __va_arg_pack ());
      |       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from scripts/kconfig/zconf.tab.c:2503:
scripts/kconfig/menu.c: In function ‘get_symbol_str’:
scripts/kconfig/menu.c:567:18: warning: ‘jump’ may be used uninitialized in this function [-Wmaybe-uninitialized]
  567 |     jump->offset = r->len - 1;
      |     ~~~~~~~~~~~~~^~~~~~~~~~~~
scripts/kconfig/menu.c:528:19: note: ‘jump’ was declared here
  528 |  struct jump_key *jump;
      |                   ^~~~
  HOSTLD  scripts/kconfig/conf
scripts/kconfig/conf --olddefconfig Kconfig
#
# configuration written to .config
#
if grep -q "^CONFIG_DEBUG_INFO_BTF=y" /home/kkk/Downloads/firmware/output/build/linux-custom//.config; then echo "To use CONFIG_DEBUG_INFO_BTF, enable host-pahole (BR2_LINUX_KERNEL_NEEDS_HOST_PAHOLE)" 1>&2; exit 1; fi
# As the kernel gets compiled before root filesystems are
# built, we create a fake cpio file. It'll be
# replaced later by the real cpio archive, and the kernel will be
[... 801 lines cut ...]
                 from ../rtl8812au-openipc/os_dep/linux/ioctl_cfg80211.c:17:
include/net/cfg80211.h:1225:12: note: ‘ssid’ declared here
 1225 |         u8 ssid[IEEE80211_MAX_SSID_LEN];
      |            ^~~~
../rtl8812au-openipc/os_dep/linux/ioctl_cfg80211.c: In function ‘cfg80211_rtw_get_channel’:
[... cut ...]
make[4]: *** [scripts/Makefile.build:308: /home/kkk/Downloads/firmware/output/build/rtl8812au-openipc/./os_dep/linux/ioctl_cfg80211.o] Error 1
make[3]: *** [Makefile:1228: _module_/home/kkk/Downloads/firmware/output/build/rtl8812au-openipc/.] Error 2
make[2]: *** [package/pkg-generic.mk:284: /home/kkk/Downloads/firmware/output/build/rtl8812au-openipc/.stamp_built] Error 2
make[1]: *** [Makefile:82: _all] Error 2
make[1]: Leaving directory '/home/kkk/Downloads/firmware/output/buildroot-2023.02.7'
make: *** [Makefile:40: br-rtl8812au-openipc-rebuild] Error 2
kkk@kkk-Lenovo-ideapad-100-15IBY:~/Downloads/firmware$

kangz543g, 2023-12-28:
mount -o remount,rw /rom
rm /rom/lib/modules/3.10.14__isvp_turkey_1.0__/ingenic/sensor_ov2735b_t21z.ko
rm: remove '/rom/lib/modules/3.10.14__isvp_turkey_1.0__/ingenic/sensor_ov2735b_t21z.ko'? yes
rm: can't remove '/rom/lib/modules/3.10.14__isvp_turkey_1.0__/ingenic/sensor_ov2735b_t21z.ko': Read-only file system

kangz543g, 2023-12-28:
root@openipc-t21:/dev# lsmod
Module                  Size  Used by    Tainted: G
audio                  73090  0
sensor_jxf37_t21        9722  0
tx_isp_t21            659258  1 sensor_jxf37_t21
sinfo                  16508  0
vfat                    9665  1
fat                    56394  1 vfat
f2fs                  132628  0

without /dev/video0????

themactep, 2023-12-28:
Sorry, this is not a Linux crash course. @kangz543g, your issues are irrelevant to the firmware. I am closing the ticket now. Pump your linux skills and feel free to open a new ticket if you face a real firmware issue.

closed 2023-12-28 by themactep

kangz543g, 2023-12-28:
themactep,thank you. Ingenic T21 firmware (factory) is being analyzed.

kangz543g, 2023-12-28:
error sensor_jxf37_t21l.ko  --> sensor_jxf37_t21n.ko (Please correct it).

kangz543g, 2023-12-29:
/init/app_init.sh
#!/bin/sh

#ifconfig eth0 192.168.10.189 netmask 255.255.255.0
#route add default gw 192.168.10.1
#udhcpc -i eth0

#ulimit -n 2048

#echo 1 > /proc/sys/vm/overcommit_memory
#echo 1 > /proc/sys/vm/drop_caches

#echo 2048 > /proc/sys/vm/min_free_kbytes
#echo 200 > /proc/sys/vm/vfs_cache_pressure

#echo 100 > /proc/sys/vm/swappiness

if [ -w /sys/block/zram0/disksize ]; then
  #echo 14300160 > /sys/block/zram0/disksize
  echo 15728640 > /sys/block/zram0/disksize
  mkswap /dev/zram0
  swapon /dev/zram0
fi

#/user/init/setir 0 1

/user/init/start.sh

/init/start.sh
#!/bin/sh

MODULES_PATH=/lib/modules
SINFO_KO_PATH=${MODULES_PATH}
SENSOR_DRV_PATH=${MODULES_PATH}
ISP_DRV_PATH=${MODULES_PATH}

# unit is kB
GIGABYTE_KB=1048576
let SD_CARD_SIZE_64GB_KB=GIGABYTE_KB*64

###  functions
check_reboot()
{
  if [ $? -ne 0 ] ;then
    echo "err: $1"
    echo "reboot"
    reboot
  fi
}

format_sd_card()
{
fdisk /dev/mmcblk0 << EOF
d
1
o
n
p
1

w
EOF

  if [ $# -eq 1 ]; then
    SD_LABEL=$1
  else
    SD_LABEL=PW_IPC
  fi
	
  mkfs.vfat -n ${SD_LABEL} /dev/mmcblk0p1
[... 257 lines cut ...]
  if [ $? -eq 0 ]; then
    UPDATE_FLAG=`sed -n "1p" /puwell/mtd/up_state.txt`
    if [ ${UPDATE_FLAG} -eq 1 ]; then
      [ -e /puwell/voice/UpgradeMy ] && /puwell/voice/UpgradeMy &
    else
      if [ -z "$TF" ]; then
        FORMATTING_FLAG_FILE="/puwell/mtd/formatting_flag.txt"
        if [ -f $FORMATTING_FLAG_FILE ]; then 
          FORMATTING_FLAG=`cat $FORMATTING_FLAG_FILE`
        else 
          FORMATTING_FLAG=0
        fi
		
        mount_sd_card $FORMATTING_FLAG
        test $? -eq 0 -a $FORMATTING_FLAG -eq 1 && echo 0 > $FORMATTING_FLAG_FILE
      fi
	  
      Alloca &
    fi
  else
    echo "error: wpa_supplicant failed!"
    echo "reboot"
    reboot
  fi
fi

kangz543g, 2023-12-29:
RTL8812au compiler success on Windows10/Ubuntu 18.04.6 LTS. It starts with the  Ingenic T21 test.

kangz543g, 2024-01-02:
use build openipc-t21-lite-8mb.bin???
make command???

themactep, 2024-01-02:
Look, this in not a proper place to seek answers to random questions. Read the wiki, or at least come to the telegram group and ask away.

kangz543g, 2024-01-03:
https://github.com/kangz543g/Ingenic-T10-40

themactep, 2024-01-03:
Ok. But why?!
```
while :;do gpio toggle 78; sleep 1; done
```

kangz543g, 2024-01-06:
U-Boot 2013.07-g05f0cca (Oct 10 2023 - 00:05:13)

Board: ISVP (Ingenic XBurst T21 SoC)
DRAM:  64 MiB
Top of RAM usable for U-Boot at: 84000000
Reserving 473k for U-Boot at: 83f88000
Reserving 32832k for malloc() at: 81f78000
Reserving 32 Bytes for Board Info at: 81f77fe0
Reserving 124 Bytes for Global Data at: 81f77f64
Reserving 256k for boot params() at: 81f37f64
Stack Pointer at: 81f37f48
Now running in RAM - U-Boot at: 83f88000
MMC:   msc: 0
the manufacturer ef
SF: Detected W25Q128

In:    serial
Out:   serial
Err:   serial
Net:   ====>phy 0:0x0-0x0 found
SPEED:0, DUPLEX:0
Jz4775-9161
Autoupdate...
Interface:  MMC
  Device 0: Vendor: Man 000003 Snr 8b3b8601 Rev: 13.10 Prod: SS08G0
            Type: Removable Hard Disk
            Capacity: 7386.0 MB = 7.2 GB (15126528 x 512)
Filesystem: FAT32 "NO NAME    "
the manufacturer ef
SF: Detected W25Q128

reading autoupdate-uboot.img
reading autoupdate-kernel.img
reading autoupdate-rootfs.img
reading autoupdate-full.img
Press Ctrl-c to abort autoboot... 1
the manufacturer ef
SF: Detected W25Q128

--->probe spend 4 ms
SF: 3145728 bytes @ 0x50000 Read: OK
--->read spend 954 ms
## Booting kernel from Legacy Image at 80600000 ...
   Image Name:   Linux-3.10.14__isvp_turkey_1.0__
   Image Type:   MIPS Linux Kernel Image (lzma compressed)
   Data Size:    1578118 Bytes = 1.5 MiB
   Load Address: 80010000
   Entry Point:  8035e820
   Verifying Checksum ... OK
   Uncompressing Kernel Image ... OK

Starting kernel ...

[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 3.10.14__isvp_turkey_1.0__ (runner@fv-az702-998) (buildroot-gcc-12.3.0) #2 PREEMPT Sat Jan 6 06:19:08 UTC 2024
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU0 RESET ERROR PC:801BDD00
[    0.000000] [<801bdd00>] __delay+0x0/0x10
[    0.000000] CPU0 revision is: 00d00100 (Ingenic Xburst)
[    0.000000] FPU revision is: 00b70000
[    0.000000] CCLK:864MHz L2CLK:432Mhz H0CLK:225MHz H2CLK:225Mhz PCLK:112Mhz
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 0047b000 @ 00010000 (usable)
[    0.000000]  memory: 00035000 @ 0048b000 (usable after init)
[    0.000000] User-defined physical RAM map:
[    0.000000]  memory: 02800000 @ 00000000 (usable)
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x00000000-0x027fffff]
[... 93 lines cut ...]
[    0.596872] =========>gmacdev->MacBase = 0xb34b0000 DmaBase = 0xb34b1000
[    0.604284] Bus Mode Reg after reset: 0x00020101, cnt=0
[    0.611878] jz_mac jz_mac.0: JZ on-chip Ethernet MAC driver, Version 1.0
[    0.619183] i2c /dev entries driver
[    0.625383] jzmmc_v1.2 jzmmc_v1.2.0: vmmc regulator missing
[    0.631868] jzmmc_v1.2 jzmmc_v1.2.0: register success!
[    0.638232] ipip: IPv4 over IPv4 tunneling driver
[    0.644360] TCP: cubic registered
[    0.650097] NET: Registered protocol family 10
[    0.656118] NET: Registered protocol family 17
[    0.662762] soc_vpu probe success,version:1.0.0-03203fd46d
[    0.669112] input: gpio-keys as /devices/platform/gpio-keys/input/input0
[    0.677348] drivers/rtc/hctosys.c: unable to open rtc device (rtc0)
[    0.687115] List of all partitions:
[    0.691005] 1f00             256 mtdblock0  (driver?)
[    0.696241] 1f01              64 mtdblock1  (driver?)
[    0.701563] 1f02            2048 mtdblock2  (driver?)
[    0.706790] 1f03            5120 mtdblock3  (driver?)
[    0.712152] 1f04            8896 mtdblock4  (driver?)
[    0.717373] No filesystem could mount root, tried:  squashfs
[    0.723312] Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(31,3)
[    0.731944] Rebooting in 20 seconds..

on a camera with Ingenic T21 SoC and NOR 16M SPI flash chip
error Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(31,3) ??????

gtxaspec, 2024-01-06:
did you issue the command `run setnor16m` in u-boot?

kangz543g, 2024-01-06:
use xgpro program.

kangz543g, 2024-01-06:
A reboot occurs in 10 seconds.
mmc sd cannot be read.

gtxaspec, 2024-01-06:
logs? can you join the telegram group? easier to chat.

kangz543g, 2024-01-06:
25q64  -- ok 
25q128 -- mmc sd cannot be read. 
A reboot occurs in 10 seconds.  --- chmod -x /usr/bin/majestic

kangz543g, 2024-01-06:
mkdir /mnt/mmcblk0p1 
mount /dev/mmcblk0p1 /mnt/mmcblk0p1
mount 
reboot

kangz543g, 2024-01-06:
vi /etc/network/interfaces.d/wlan0
auto wlan0
iface wlan0 inet dhcp
    pre-up  modprobe mac80211
    sleep 1
    pre-up  insmod /lib/modules/3.10.14__isvp_swan_1.0__/extra/8188fu.ko
    sleep 1
    pre-up wlan_addr=$(fw_printenv -n wlanaddr); if [ -n "$wlan_addr" ]; then ip link set dev wlan0 address $wlan_addr; fi
    post-up wpa_passphrase "$(fw_printenv -n wlanssid || echo OpenIPC)" "$(fw_printenv -n wlanpass || echo OpenIPC12345)" > /tmp/wpa_supplicant.conf
    post-up sed -i '2i \\tscan_ssid=1' /tmp/wpa_supplicant.conf
    post-up wpa_supplicant -B -i wlan0 -D nl80211,wext -c /tmp/wpa_supplicant.conf
    post-down killall -q wpa_supplicant

ifconfig does not run

kangz543g, 2024-01-06:
8818fu Installation failed. There is no solution. This is a fact-check.

kangz543g, 2024-01-06:
root@openipc-t21:/# ifconfig
lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:28 errors:0 dropped:0 overruns:0 frame:0
          TX packets:28 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:2184 (2.1 KiB)  TX bytes:2184 (2.1 KiB)

wlan0     Link encap:Ethernet  HWaddr DC:BD:7A:2B:9D:C9
          inet addr:192.168.219.115  Bcast:192.168.219.255  Mask:255.255.255.0
          inet6 addr: fe80::debd:7aff:fe2b:9dc9/64 Scope:Link
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:5 errors:0 dropped:1 overruns:0 frame:0
          TX packets:5 errors:0 dropped:9 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:665 (665.0 B)  TX bytes:775 (775.0 B)

8188fu Is it a driver and network problem? failure!!!!

kangz543g, 2024-01-06:
root@openipc-t21:/# iwconfig
lo        no wireless extensions.

tunl0     no wireless extensions.

wlan1     unassociated  Nickname:"RTL8188FU"
          Mode:Managed  Frequency=2.412 GHz  Access Point: Not-Associated
          Sensitivity:0/0
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          Link Quality:0  Signal level:0  Noise level:0
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

wlan0     unassociated  Nickname:"RTL8188FU"
          Mode:Managed  Frequency=2.427 GHz  Access Point: Not-Associated
          Sensitivity:0/0
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          Link Quality:0  Signal level:0  Noise level:0
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

eth0      no wireless extensions.

kangz543g, 2024-01-06:
rtl8812au testing...
root@openipc-t21:/# lsusb
Bus 001 Device 002: ID 0bda:881a
Bus 001 Device 001: ID 1d6b:0002

ifconfig 
root@openipc-t21:/# ifconfig
lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:128 errors:0 dropped:0 overruns:0 frame:0
          TX packets:128 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:9984 (9.7 KiB)  TX bytes:9984 (9.7 KiB)

wlan0     Link encap:Ethernet  HWaddr BC:EC:23:1E:89:0A
          inet addr:192.168.219.115  Bcast:192.168.219.255  Mask:255.255.255.0
          inet6 addr: fe80::beec:23ff:fe1e:890a/64 Scope:Link
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:16 errors:0 dropped:0 overruns:0 frame:0
          TX packets:16 errors:0 dropped:21 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:2128 (2.0 KiB)  TX bytes:2480 (2.4 KiB)

iwconfigroot@openipc-t21:/# iwconfig
lo        no wireless extensions.

tunl0     no wireless extensions.

wlan0     IEEE 802.11abgn  ESSID:off/any
          Mode:Managed  Access Point: Not-Associated   Tx-Power=18 dBm
          Retry  long limit:7   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off

eth0      no wireless extensions.

rtl8812au Is it a driver and network problem? failure!!!!

kangz543g, 2024-01-07:
* camera ip
wlan0     Link encap:Ethernet  HWaddr DC:BD:7A:2B:9D:C9
          inet addr:192.168.219.115  Bcast:192.168.219.255  Mask:255.255.255.0
          inet6 addr: fe80::debd:7aff:fe2b:9dc9/64 Scope:Link
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:7 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:532 (532.0 B)  TX bytes:620 (620.0 B)

root@openipc-t21:~# iwconfig
lo        no wireless extensions.

tunl0     no wireless extensions.

wlan0     unassociated  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency=2.442 GHz  Access Point: Not-Associated
          Sensitivity:0/0
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          Link Quality:0  Signal level:0  Noise level:0
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

eth0      no wireless extensions.

* pc ip test
C:\Users\kangz>nmap 192.168.219.115
Starting Nmap 7.94 ( https://nmap.org ) at 2024-01-07 10:41 대한민국 표준시
Note: Host seems down. If it is really up, but blocking our ping probes, try -Pn
Nmap done: 1 IP address (0 hosts up) scanned in 3.45 seconds

problem openipc fireware....

kangz543g, 2024-01-07:
Linux fixes completed on Ingenic T21. Congratulations on your success!!!!!!!!!

kangz543g, 2024-01-07:
youtube:
  enabled: true
  key: xxxx-xxxx-xxxx-xxxx-xxxx
rtmp://a.rtmp.youtube.com/live2

Do you know how to use it?

kangz543g, 2024-01-08:
[image]
[image]
[image]
[image]
[image]

kangz543g, 2024-01-08:
Thank you openipc team.

kangz543g, 2024-01-08:
[image]

kangz543g, 2024-01-08:
[image]

kangz543g, 2024-01-08:
No problem with Ingenic T21. Thank you openipc team.

kangz543g, 2024-01-09:
# rotate image by none, 90 or 270 degrees
  rotate: none  <-- not 90 or 270?????

kangz543g, 2024-01-09:
FPV installation, RTL8812AU installation is successful.

kangz543g, 2024-01-09:
OPENIPC -- EPS32 USB WIFI is being installed.

kangz543g, 2024-01-09:
This is a 3D printer printout. This is the production of the camera body.
[image]

kangz543g, 2024-01-10:
ready....
[image]
[image]
[image]

kangz543g, 2024-01-11:
can not openipc telegram group???
testing rtl8812au ok. How to use fpv???

Note: gk7205v300-fpv/etc/datalink.conf
/etc/mavlink.conf
/etc/telemetty.conf
/etc/wfb.conf
/etc/wireguard.conf
/etc/init.d/S98datalink
/usr/bin/mavfwd
/usr/bin/mavlink-routerd
/usr/bin/telemetry
/usr/wfb_keygen
/usr/wfb_rx
/usr/wfb_tx
/usr/wifibroadcast

please help...

kangz543g, 2024-01-13:
Supported Channels and Widths
In order to be able to set 80 MHz channels with iw, you need version  4.14 or greater. In my case, I have installed version 5.9:

iw --version
iw version 5.9
The syntax of settings channels has changed after version 4.14, and in order to get the correct syntax you can enter an incomplete command to get the help output as follows:

netbeez$ iw dev wlan0 set channel
Usage:  iw [options] dev <devname> set channel <channel> [NOHT|HT20|HT40+|HT40-|5MHz|10MHz|80MHz]
Options:
        --debug         enable netlink debugging

Once you [set the interface to monitor mode](https://netbeez.net/blog/linux-how-to-configure-monitoring-mode-wifi-interface/) the channel and channel width are left at the same configuration they were while the interface was connected to your WiFi network (if it was ever). In my case, it was on channel 149 at 80MHz:

netbeez$ iw wlan0 info
Interface wlan0
        ifindex 3
        wdev 0x1
        addr 20:0d:b0:47:57:79
        type monitor
        wiphy 0
        channel 149 (5745 MHz), width: 80 MHz, center1: 5775 MHz
        txpower 18.00 dBm
As always, when setting a channel and channel width we have to make sure that 

They have acceptable values (e.g. we can’t set channel 1 at 40- MHz)
The driver and hardware support the requested channel and width values
The first step it to determine what channels your setup supports with the following:

netbeez$ iw list
Wiphy phy0
        wiphy index: 0
        max # scan SSIDs: 9
        max scan IEs length: 2304 bytes
        max # sched scan SSIDs: 0
        max # match sets: 0
        max # scan plans: 1
        max scan plan interval: -1
. . . 
. . .
. . .
             Frequencies:
                        * 2412 MHz [1] (20.0 dBm)
                        * 2417 MHz [2] (20.0 dBm)
                        * 2422 MHz [3] (20.0 dBm)
                        * 2427 MHz [4] (20.0 dBm)
                        * 2432 MHz [5] (20.0 dBm)
                        * 2437 MHz [6] (20.0 dBm)
                        * 2442 MHz [7] (20.0 dBm)
                        * 2447 MHz [8] (20.0 dBm)
                        * 2452 MHz [9] (20.0 dBm)
                        * 2457 MHz [10] (20.0 dBm)
                        * 2462 MHz [11] (20.0 dBm)
                        * 2467 MHz [12] (20.0 dBm) (no IR)
                        * 2472 MHz [13] (20.0 dBm) (no IR)
                        * 2484 MHz [14] (20.0 dBm) (no IR)

. . .
. . .
. . .
            Frequencies:
                        * 5180 MHz [36] (20.0 dBm) (no IR)
                        * 5200 MHz [40] (20.0 dBm) (no IR)
                        * 5220 MHz [44] (20.0 dBm) (no IR)
                        * 5240 MHz [48] (20.0 dBm) (no IR)
                        * 5260 MHz [52] (20.0 dBm) (no IR, radar detection)
                        * 5280 MHz [56] (20.0 dBm) (no IR, radar detection)
                        * 5300 MHz [60] (20.0 dBm) (no IR, radar detection)
                        * 5320 MHz [64] (20.0 dBm) (no IR, radar detection)
                        * 5500 MHz [100] (20.0 dBm) (no IR, radar detection)
[... 168 lines cut ...]
    channels="1 2 3 4 5 6 7 8 9 36 40 44 48 52 56 60 100 104 108 112 116 120 124 128 132 136 140 153 157"
elif [[ "${1}" = "HT40-" ]]; then
    channels="5 6 7 8 9 10 11 12 13 40 44 48 52 56 60 64  104 108 112 116 120 124 128 132 136 140 144 153 157 161"
elif [[ "${1}" = "80MHz" ]]; then
    channels="${channels_50}"
fi

iw dev wlan0 set type monitor
ifconfig wlan0 up

for channel in ${channels}; do
    echo "Setting channel ${channel}, ${width}"
    iw dev wlan0 set channel "${channel}" "${width}"
    sleep 0.1
done
To run this script in parallel with the tcpdump capture you have either to use two different terminals (one for the channel hopping script and one for the tcpdump capture) or to run the channel hopping script as a background process and then launch the tcpdump capturing command as follows:

netbeez$ ./set_channels.sh HT20 &
[2] 15917
netbeez$ tcpdump -i wlan0 -w capture.pcap
tcpdump: listening on wlan0, link-type IEEE802_11_RADIO (802.11 plus radiotap header), capture size 262144 bytes
^C930 packets captured
930 packets received by filter
0 packets dropped by kernel
Once you stop tcpdump with “Ctrl+C” you can stop the channel hopping script with

kangz543g, 2024-01-16:
Installation of RTL8812BU in OPENIPC is successful.

[image]
[image]
[image]
[
[image]
[image]
](url)

kangz543g, 2024-01-16:
It looks like you're providing a set of commands and configurations related to building a kernel module 
(88xx2bu.ko) for a MIPS platform, specifically for a T21 device. 
Additionally, you mention that it successfully operates on an OpenIPC system.

Here's a breakdown of the provided commands:

Change directory to /Main/kernel.
Copy the t21.genertic.config file to .config.
Change directory to 88x2bu-20210702.
Open the Markfile using the nano text editor and modify its contents.
Set CONFIG_PLATFORM_MIPS_T21 to 'y'.
Add some extra compiler flags.
Set architecture (ARCH) to 'mips'.
Set the cross-compiler (CROSS_COMPILE) path.
Set kernel source (KSRC) path.
Save the changes made to the Markfile.
Run the make command to build the kernel module.
After successful compilation, the 88xx2bu.ko module can be used in an OpenIPC environment, and internet
 functionality is confirmed.
 
 제공된 명령어와 구성은 MIPS 플랫폼을 위한 커널 모듈 (88xx2bu.ko)을 빌드하는 것으로 보입니다. 
 특히 T21 장치를 위한 것이며, OpenIPC 시스템에서 성공적으로 작동한다고 언급하고 있습니다.

제공된 명령어를 살펴보겠습니다:

디렉터리를 /Main/kernel로 변경합니다.
t21.genertic.config 파일을 .config로 복사합니다.
디렉터리를 88x2bu-20210702로 변경합니다.
Markfile을 nano 텍스트 편집기로 열고 내용을 수정합니다.
CONFIG_PLATFORM_MIPS_T21을 'y'로 설정합니다.
몇 가지 추가 컴파일러 플래그를 추가합니다.
아키텍처 (ARCH)를 'mips'로 설정합니다.
크로스 컴파일러 (CROSS_COMPILE) 경로를 설정합니다.
커널 소스 (KSRC) 경로를 설정합니다.
Markfile에서 수행한 변경 사항을 저장합니다.
make 명령어를 사용하여 커널 모듈을 빌드합니다.
빌드가 성공하면 88xx2bu.ko 모듈을 OpenIPC 환경에서 사용할 수 있으며, 인터넷 기능이 확인됩니다.

kangz543g, 2024-01-16:
RTL8812BU - Recommended WiFi Router/ Access Point Settings
Note: These are general recommendations, some of which may not apply to your specific situation.

Security: Set WPA2-AES or WPA2/WPA3 mixed or WPA3. Do not set WPA2 mixed mode or WPA or TKIP.

Channel width for 2.4 GHz: Set 20 MHz fixed width. Do not use 40 MHz or 20/40 automatic.

Channels for 2.4 GHz: Set channel 1 or 6 or 11 depending on the congestion at your location. Do not set automatic channel selection. As time passes, if you notice poor performance, recheck congestion and set channel appropriately. The environment around you can and does change over time.

Mode for 2.4 GHz: For best performance, set "N only" if you no longer use B or G capable devices.

Network names: Do not set the 2.4 GHz Network and the 5 GHz Network to the same name. Note: Unfortunately many routers come with both networks set to the same name. You need to be able to control which network that is in use so changing the name of one of the networks is recommended. Since many IoT devices use the 2.4 GHz network, it may be better to change the name of the 5 GHz network.

Channels for 5 GHz: Not all devices are capable of using DFS channels (I'm looking at you Roku.) It may be necessary to set a fixed channel in the range of 36 to 48 or 149 to 165 in order for all of your devices to work on 5 GHz. (For US, other countries may vary.)

Best location for the WiFi router/access point: Near center of apartment or house, at least a couple of feet away from walls, in an elevated location. You may have to test to see what the best location is in your environment.

Check congestion: There are apps available for smart phones that allow you to get an idea of the congestion levels on WiFi channels. The apps generally go by the name of WiFi Analyzer or something similar.

After making and saving changes, reboot the router.

kangz543g, 2024-01-16:
RTL8812BU TEST OK!!!!

root@openipc-t21:~# iwgetid -f
wlan0     Frequency=2.442 GHz   (2.4GHZ~5.8GHZ)

root@openipc-t21:~# iwgetid -a
wlan0     Access Point/Cell: 18:C5:01:8F:2A:2A

root@openipc-t21:~# iwgetid -c
wlan0     Channel:7   (1~165)

root@openipc-t21:~#

kangz543g, 2024-01-16:
RTL8812BU TEST Current Wireless Channel and Frequency:
wlan0     32 channels in total; available frequencies :
          Channel 01 : 2.412 GHz
          Channel 02 : 2.417 GHz
          Channel 03 : 2.422 GHz
          Channel 04 : 2.427 GHz
          Channel 05 : 2.432 GHz
          Channel 06 : 2.437 GHz
          Channel 07 : 2.442 GHz
          Channel 08 : 2.447 GHz
          Channel 09 : 2.452 GHz
          Channel 10 : 2.457 GHz
          Channel 11 : 2.462 GHz
          Channel 12 : 2.467 GHz
          Channel 13 : 2.472 GHz
          Channel 36 : 5.18 GHz
          Channel 40 : 5.2 GHz
          Channel 44 : 5.22 GHz
          Channel 48 : 5.24 GHz
          Channel 52 : 5.26 GHz
          Channel 56 : 5.28 GHz
          Channel 60 : 5.3 GHz
          Channel 64 : 5.32 GHz
          Channel 100 : 5.5 GHz
          Channel 104 : 5.52 GHz
          Channel 108 : 5.54 GHz
          Channel 112 : 5.56 GHz
          Channel 116 : 5.58 GHz
          Channel 120 : 5.6 GHz
          Channel 124 : 5.62 GHz
          Channel 128 : 5.64 GHz
          Channel 132 : 5.66 GHz
          Channel 136 : 5.68 GHz
          Channel 140 : 5.7 GHz
          Current Frequency:2.442 GHz (Channel 7)

kangz543g, 2024-01-17:
I'm analyzing RTL8812BU tomorrow.  openipc ingenic t21 drone fpv  's successful.!!!!

kangz543g, 2024-01-17:
[image]
[image]
[image]

kangz543g, 2024-01-17:
openipc-gk7205v300-fpv-8mb.bin --> openipc-IngenicT21-fpv-16mb.bin Analysis complete!!!!!!

kangz543g, 2024-01-17:
[image]

kangz543g, 2024-01-18:
Automatic booting supports RTL8188EU, RTL8812AU, and RTL8812BU drivers.

kangz543g, 2024-01-18:
[image]
[image]

kangz543g, 2024-01-19:
THE END

kangz543g, 2024-01-19:
error br-8188eu-openipc compiler!!!!!!!

Installation of RTL8188eu in OPENIPC is successful.

make ARCH=mips CROSS_COMPILE=/home/kkk/mips-gcc472-glibc216-64bit/bin/mips-linux-gnu- -C /home/kkk/Main/kernel M=/home/kkk/rtl8188eus  modules
make[1]: 디렉터리 '/home/kkk/Main/kernel' 들어감
  CC [M]  /home/kkk/rtl8188eus/core/rtw_cmd.o
  CC [M]  /home/kkk/rtl8188eus/core/rtw_security.o
  CC [M]  /home/kkk/rtl8188eus/core/rtw_debug.o
  CC [M]  /home/kkk/rtl8188eus/core/rtw_io.o
  CC [M]  /home/kkk/rtl8188eus/core/rtw_ioctl_query.o
  CC [M]  /home/kkk/rtl8188eus/core/rtw_ioctl_set.o
  CC [M]  /home/kkk/rtl8188eus/core/rtw_ieee80211.o
  CC [M]  /home/kkk/rtl8188eus/core/rtw_mlme.o
  CC [M]  /home/kkk/rtl8188eus/core/rtw_mlme_ext.o
  CC [M]  /home/kkk/rtl8188eus/core/rtw_mi.o
  CC [M]  /home/kkk/rtl8188eus/core/rtw_wlan_util.o
  CC [M]  /home/kkk/rtl8188eus/core/rtw_vht.o
  CC [M]  /home/kkk/rtl8188eus/core/rtw_pwrctrl.o
  CC [M]  /home/kkk/rtl8188eus/core/rtw_rf.o
  CC [M]  /home/kkk/rtl8188eus/core/rtw_recv.o
  CC [M]  /home/kkk/rtl8188eus/core/rtw_sta_mgt.o
  CC [M]  /home/kkk/rtl8188eus/core/rtw_ap.o
  CC [M]  /home/kkk/rtl8188eus/core/mesh/rtw_mesh.o
  CC [M]  /home/kkk/rtl8188eus/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /home/kkk/rtl8188eus/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /home/kkk/rtl8188eus/core/rtw_xmit.o
  CC [M]  /home/kkk/rtl8188eus/core/rtw_p2p.o
  CC [M]  /home/kkk/rtl8188eus/core/rtw_rson.o
  CC [M]  /home/kkk/rtl8188eus/core/rtw_tdls.o
  CC [M]  /home/kkk/rtl8188eus/core/rtw_br_ext.o
  CC [M]  /home/kkk/rtl8188eus/core/rtw_iol.o
  CC [M]  /home/kkk/rtl8188eus/core/rtw_sreset.o
  CC [M]  /home/kkk/rtl8188eus/core/rtw_btcoex_wifionly.o
  CC [M]  /home/kkk/rtl8188eus/core/rtw_btcoex.o
  CC [M]  /home/kkk/rtl8188eus/core/rtw_beamforming.o
  CC [M]  /home/kkk/rtl8188eus/core/rtw_odm.o
  CC [M]  /home/kkk/rtl8188eus/core/rtw_rm.o
  CC [M]  /home/kkk/rtl8188eus/core/rtw_rm_fsm.o
  CC [M]  /home/kkk/rtl8188eus/core/efuse/rtw_efuse.o
  CC [M]  /home/kkk/rtl8188eus/hal/hal_intf.o
/home/kkk/rtl8188eus/hal/hal_intf.c: In function 'c2h_handler':
/home/kkk/rtl8188eus/hal/hal_intf.c:942:3: warning: empty declaration [enabled by default]
/home/kkk/rtl8188eus/hal/hal_intf.c:942:3: warning: ISO C90 forbids mixed declarations and code [-Wdeclaration-after-statement]
/home/kkk/rtl8188eus/hal/hal_intf.c: At top level:
cc1: warning: unrecognized command line option "-Wno-date-time" [enabled by default]
cc1: warning: unrecognized command line option "-Wno-enum-conversion" [enabled by default]
cc1: warning: unrecognized command line option "-Wno-sometimes-uninitialized" [enabled by default]
cc1: warning: unrecognized command line option "-Wno-cast-function-type" [enabled by default]
  CC [M]  /home/kkk/rtl8188eus/hal/hal_com.o
  CC [M]  /home/kkk/rtl8188eus/hal/hal_com_phycfg.o
  CC [M]  /home/kkk/rtl8188eus/hal/hal_phy.o
  CC [M]  /home/kkk/rtl8188eus/hal/hal_dm.o
  CC [M]  /home/kkk/rtl8188eus/hal/hal_dm_acs.o
  CC [M]  /home/kkk/rtl8188eus/hal/hal_btcoex_wifionly.o
  CC [M]  /home/kkk/rtl8188eus/hal/hal_btcoex.o
  CC [M]  /home/kkk/rtl8188eus/hal/hal_mp.o
  CC [M]  /home/kkk/rtl8188eus/hal/hal_mcc.o
  CC [M]  /home/kkk/rtl8188eus/hal/hal_hci/hal_usb.o
  CC [M]  /home/kkk/rtl8188eus/hal/led/hal_led.o
  CC [M]  /home/kkk/rtl8188eus/hal/led/hal_usb_led.o
  CC [M]  /home/kkk/rtl8188eus/hal/HalPwrSeqCmd.o
  CC [M]  /home/kkk/rtl8188eus/hal/rtl8188e/Hal8188EPwrSeq.o
  CC [M]  /home/kkk/rtl8188eus/hal/rtl8188e/rtl8188e_xmit.o
  CC [M]  /home/kkk/rtl8188eus/hal/rtl8188e/rtl8188e_sreset.o
  CC [M]  /home/kkk/rtl8188eus/hal/rtl8188e/rtl8188e_hal_init.o
  CC [M]  /home/kkk/rtl8188eus/hal/rtl8188e/rtl8188e_phycfg.o
  CC [M]  /home/kkk/rtl8188eus/hal/rtl8188e/rtl8188e_rf6052.o
  CC [M]  /home/kkk/rtl8188eus/hal/rtl8188e/rtl8188e_dm.o
  CC [M]  /home/kkk/rtl8188eus/hal/rtl8188e/rtl8188e_rxdesc.o
[... 52 lines cut ...]
  CC [M]  /home/kkk/rtl8188eus/hal/phydm/phydm_math_lib.o
  CC [M]  /home/kkk/rtl8188eus/hal/phydm/phydm_api.o
  CC [M]  /home/kkk/rtl8188eus/hal/phydm/phydm_pow_train.o
  CC [M]  /home/kkk/rtl8188eus/hal/phydm/halrf/halrf.o
  CC [M]  /home/kkk/rtl8188eus/hal/phydm/halrf/halphyrf_ce.o
  CC [M]  /home/kkk/rtl8188eus/hal/phydm/halrf/halrf_powertracking_ce.o
  CC [M]  /home/kkk/rtl8188eus/hal/phydm/halrf/halrf_powertracking.o
  CC [M]  /home/kkk/rtl8188eus/hal/phydm/halrf/halrf_kfree.o
  CC [M]  /home/kkk/rtl8188eus/hal/phydm/rtl8188e/halhwimg8188e_mac.o
  CC [M]  /home/kkk/rtl8188eus/hal/phydm/rtl8188e/halhwimg8188e_bb.o
  CC [M]  /home/kkk/rtl8188eus/hal/phydm/rtl8188e/halhwimg8188e_rf.o
  CC [M]  /home/kkk/rtl8188eus/hal/phydm/halrf/rtl8188e/halrf_8188e_ce.o
  CC [M]  /home/kkk/rtl8188eus/hal/phydm/rtl8188e/phydm_regconfig8188e.o
  CC [M]  /home/kkk/rtl8188eus/hal/phydm/rtl8188e/hal8188erateadaptive.o
  CC [M]  /home/kkk/rtl8188eus/hal/phydm/rtl8188e/phydm_rtl8188e.o
  CC [M]  /home/kkk/rtl8188eus/platform/platform_ops.o
  CC [M]  /home/kkk/rtl8188eus/core/rtw_mp.o
  LD [M]  /home/kkk/rtl8188eus/8188eu.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /home/kkk/rtl8188eus/8188eu.mod.o
  LD [M]  /home/kkk/rtl8188eus/8188eu.ko
make[1]: 디렉터리 '/home/kkk/Main/kernel' 나감
kkk@kkk-Lenovo-ideapad-100-15IBY:~/rtl8188eus$ ls -al 8188eu.ko
-rw-rw-r-- 1 kkk kkk 1625170  1월 19 12:30 8188eu.ko

kangz543g, 2024-01-19:
THE END

kangz543g, 2024-01-19:
ch340g usb installation

#!/bin/sh

# Check if the USB device with ID 1a86:7523 is connected
if lsusb | grep -q "1a86:7523"; then
    echo "USB device 1a86:7523 found."

    # Load the usbserial module
    modprobe usbserial

    # Check if the usbserial module is loaded successfully
    if lsmod | grep -q usbserial; then
        echo "usbserial module loaded successfully"
    else
        echo "Failed to load usbserial module"
        exit 1
    fi

    # Insert the ch34x.ko module
    insmod /lib/modules/3.10.14__isvp_turkey_1.0__/kernel/drivers/usb/serial/ch34x.ko

    # Check if the ch34x.ko module is loaded successfully
    if lsmod | grep -q ch34x; then
        echo "ch34x module loaded successfully"
    else
        echo "Failed to load ch34x module"
        exit 1
    fi

    echo "Modules loaded successfully"
else
    echo "USB device 1a86:7523 not found."
fi

exit 0

/dev/ttyUSB0    OK

kangz543g, 2024-01-19:
U-Boot 2013.07-g05f0cca (Oct 10 2023 - 00:05:13)

Board: ISVP (Ingenic XBurst T21 SoC)
DRAM:  64 MiB
Top of RAM usable for U-Boot at: 84000000
Reserving 473k for U-Boot at: 83f88000
Reserving 32832k for malloc() at: 81f78000
Reserving 32 Bytes for Board Info at: 81f77fe0
Reserving 124 Bytes for Global Data at: 81f77f64
Reserving 256k for boot params() at: 81f37f64
Stack Pointer at: 81f37f48
Now running in RAM - U-Boot at: 83f88000
MMC:   msc: 0
the manufacturer ef
SF: Detected W25Q128

In:    serial
Out:   serial
Err:   serial
Net:   ====>phy 0:0x0-0x0 found
SPEED:0, DUPLEX:0
Jz4775-9161
Autoupdate... 
Interface:  MMC
  Device 0: Vendor: Man 000003 Snr 8b3b8601 Rev: 13.10 Prod: SS08G0
            Type: Removable Hard Disk
            Capacity: 7386.0 MB = 7.2 GB (15126528 x 512)
Filesystem: FAT32 "NO NAME    "
the manufacturer ef
SF: Detected W25Q128

reading autoupdate-uboot.img
reading autoupdate-kernel.img
reading autoupdate-rootfs.img
reading autoupdate-full.img
Press Ctrl-c to abort autoboot... 1 
the manufacturer ef
SF: Detected W25Q128

--->probe spend 4 ms
SF: 3145728 bytes @ 0x50000 Read: OK
--->read spend 954 ms
## Booting kernel from Legacy Image at 80600000 ...
   Image Name:   Linux-3.10.14__isvp_turkey_1.0__
   Image Type:   MIPS Linux Kernel Image (lzma compressed)
   Data Size:    1578073 Bytes = 1.5 MiB
   Load Address: 80010000
   Entry Point:  8035e820
   Verifying Checksum ... OK
   Uncompressing Kernel Image ... OK

Starting kernel ...

[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 3.10.14__isvp_turkey_1.0__ (runner@fv-az532-920) (buildroot-gcc-12.3.0) #2 PREEMPT Sat Jan 6 15:49:53 UTC 2024
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU0 RESET ERROR PC:801BDD04
[    0.000000] [<801bdd04>] __delay+0x4/0x10
[    0.000000] CPU0 revision is: 00d00100 (Ingenic Xburst)
[    0.000000] FPU revision is: 00b70000
[    0.000000] CCLK:864MHz L2CLK:432Mhz H0CLK:225MHz H2CLK:225Mhz PCLK:112Mhz
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 0047b000 @ 00010000 (usable)
[    0.000000]  memory: 00035000 @ 0048b000 (usable after init)
[    0.000000] User-defined physical RAM map:
[    0.000000]  memory: 02800000 @ 00000000 (usable)
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x00000000-0x027fffff]
[... 139 lines cut ...]
WEXT: Could not set interface 'wlan0' UP
wlan0: Failed to initialize driver interface
wlan0: CTRL-EVENT-DSCP-POLICY clear_all
Starting ntpd: OK
Starting dropbear sshd: OK
Production mode
Starting httpd: OK
Starting mini-snmpd: DISABLED, OK
Starting telnetd: DISABLED, OK
Starting crond: OK

Loading of kernel modules...
ingenic: Get data from environment and set SENSOR as jxf37
--------------------
ISP_PARAM: isp_clk=90000000
SENSOR: jxf37
SENSOR_PARAM: sensor_gpio_func=0
--------------------
Starting rc.local
ingenic: Loading video system has started...
Starting majestic: OK
Watchdog: System watchdog disabled, unsupported vendor or Majestic Watchdog enabled

Welcome to OpenIPC
openipc-t21 login:

kangz543g, 2024-01-21:
micropython
root@openipc-t21:/mnt/mmcblk0p1/python/bin# ./micropython
MicroPython v1.19.1 on 2024-01-21; linux [GCC 12.3.0] version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> 
>>> help
<function>
>>> 
>>> import time
>>> 
>>> while True:
...     print("Hello, MicroPython!")
...     time.sleep(1)
... 
Hello, MicroPython!
Hello, MicroPython!
Hello, MicroPython!
Hello, MicroPython!
Hello, MicroPython!
Hello, MicroPython!
Hello, MicroPython!
Hello, MicroPython!

[11 more replies]