OpenIPC/firmware #1233: Stuttering video and disconnections on gk7205v200 - imx307

Yersi88, 2024-01-05:
I flashed OpenIPC on my gk7205v200. But I can't get stable video. 

1) If I open a video stream in `mpv rtsp://root:pass@192.168.80.22/stream=0` the video plays for a few seconds smoothly and then it freezes up. 
2) I added the camera to my Frigate NVR and I get very frequent connection timeouts: 
```
2024-01-05 21:03:53.792058689  [2024-01-05 21:03:53] ffmpeg.front_right.detect      ERROR   : rtsp://*:*@192.168.80.22/stream=1: Connection timed out
2024-01-05 21:03:53.792139833  [2024-01-05 21:03:53] ffmpeg.front_right.detect      ERROR   :     Last message repeated 1 times
```

What I tried so far: 
I had the following errors in openipc logs: 
`user.warn majestic[830]: Slow network for NETIP TCP transmit, skipping data`
This error went away after I increased the system.buffer: 4096 
However, this did nothing to address the choppy video and disconnections. 

Some logs: 
```
Jan  5 21:11:32 openipc-gk7205v200 authpriv.info dropbear[680]: Running in background
Jan  5 21:11:32 openipc-gk7205v200 cron.info crond[696]: crond (busybox 1.36.1) started, log level 8
Jan  5 21:11:32 openipc-gk7205v200 daemon.info goke: Get data from U-Boot ENV and set SENSOR as imx307
Jan  5 21:11:32 openipc-gk7205v200 kern.warn kernel: open_sys_config: loading out-of-tree module taints kernel.
Jan  5 21:11:32 openipc-gk7205v200 kern.emerg kernel: ==== online_flag=0, cmos_yuv_flag=0, sensor=imx307, chip=gk7205v200, board=demo====
Jan  5 21:11:32 openipc-gk7205v200 kern.emerg kernel: ==== g_quick_start_flag=0 ====
Jan  5 21:11:32 openipc-gk7205v200 kern.emerg kernel: sysconfig init success!
Jan  5 21:11:32 openipc-gk7205v200 kern.emerg kernel: Module media: init ok
Jan  5 21:11:32 openipc-gk7205v200 kern.info kernel: Media Memory Zone Manager
Jan  5 21:11:32 openipc-gk7205v200 kern.emerg kernel: osal 1.0 init success!
Jan  5 21:11:32 openipc-gk7205v200 kern.warn kernel: gk7205v200_base: module license 'Proprietary' taints kernel.
Jan  5 21:11:32 openipc-gk7205v200 kern.warn kernel: Disabling lock debugging due to kernel taint
Jan  5 21:11:32 openipc-gk7205v200 kern.emerg kernel: load sys.ko OK!
Jan  5 21:11:32 openipc-gk7205v200 kern.emerg kernel: load region.ko OK!
Jan  5 21:11:32 openipc-gk7205v200 kern.emerg kernel: load vgs.ko OK!
Jan  5 21:11:33 openipc-gk7205v200 kern.emerg kernel: load vi.ko OK !
Jan  5 21:11:33 openipc-gk7205v200 kern.emerg kernel: ISP Mod init!
Jan  5 21:11:33 openipc-gk7205v200 kern.emerg kernel: load vpss.ko OK!
Jan  5 21:11:33 openipc-gk7205v200 kern.emerg kernel: load chnl.ko OK!
Jan  5 21:11:33 openipc-gk7205v200 kern.emerg kernel: load vedu.ko OK!
Jan  5 21:11:33 openipc-gk7205v200 kern.emerg kernel: load rc.ko OK!
Jan  5 21:11:33 openipc-gk7205v200 kern.emerg kernel: load venc.ko OK!
Jan  5 21:11:33 openipc-gk7205v200 kern.emerg kernel: load h264e.ko OK!
Jan  5 21:11:33 openipc-gk7205v200 kern.emerg kernel: load h265e.ko OK!
Jan  5 21:11:33 openipc-gk7205v200 kern.emerg kernel: load jpege.ko OK!
Jan  5 21:11:33 openipc-gk7205v200 kern.emerg kernel: load ive.ko OK!
Jan  5 21:11:33 openipc-gk7205v200 kern.emerg kernel: Load sensor_spi.ko OK !
Jan  5 21:11:33 openipc-gk7205v200 kern.emerg kernel: load mipi_rx driver successful!
Jan  5 21:11:33 openipc-gk7205v200 kern.emerg kernel: register dev
Jan  5 21:11:33 openipc-gk7205v200 kern.emerg kernel: Watchdog Timer: 0.01 initialized. default_margin=60 sec (nodeamon= 0)
Jan  5 21:11:33 openipc-gk7205v200 kern.emerg kernel: wtdg init ok. ver=Nov 22 2023, 10:49:40.
Jan  5 21:11:33 openipc-gk7205v200 daemon.info goke: Loading video system has started...
Jan  5 21:11:33 openipc-gk7205v200 daemon.info : starting pid 861, tty '/dev/console': '/sbin/getty -L console 0 vt100 '
Jan  5 21:11:34 openipc-gk7205v200 user.info majestic[853]: Majestic Lite for HiSilicon, version master+b5b5dfe, built on 2023-11-17
Jan  5 21:11:34 openipc-gk7205v200 user.info majestic[853]: Using /etc/majestic.yaml as main configuration
Jan  5 21:11:34 openipc-gk7205v200 user.debug majestic[853]: Detected Watchdog
Jan  5 21:11:34 openipc-gk7205v200 user.debug majestic[853]: Watchdog timeout set to 10 seconds
Jan  5 21:11:34 openipc-gk7205v200 user.info majestic[853]: Using /etc/sensors/imx307_i2c_2l_1080p.ini sensor configuration file
Jan  5 21:11:35 openipc-gk7205v200 user.info majestic[853]: linear mode
Jan  5 21:11:35 openipc-gk7205v200 user.info majestic[853]: ==============================================================
Jan  5 21:11:35 openipc-gk7205v200 user.info majestic[853]: =====Sony imx307_2l sensor 1080P30fps(MIPI) init success!=====
Jan  5 21:11:35 openipc-gk7205v200 user.info majestic[853]: ==============================================================
Jan  5 21:11:35 openipc-gk7205v200 user.info majestic[853]: Loading IQ profile /etc/sensors/iq/default.ini
Jan  5 21:11:35 openipc-gk7205v200 user.info majestic[853]: HTTP server started on :::80
Jan  5 21:11:35 openipc-gk7205v200 user.info majestic[853]: RTSP server started on port 554
Jan  5 21:11:41 openipc-gk7205v200 kern.err kernel: goke-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
Jan  5 21:11:41 openipc-gk7205v200 kern.err kernel: goke-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
Jan  5 21:11:41 openipc-gk7205v200 kern.err kernel: goke-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
Jan  5 21:11:41 openipc-gk7205v200 kern.err kernel: goke-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
Jan  5 21:11:41 openipc-gk7205v200 kern.err kernel: goke-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
Jan  5 21:11:41 openipc-gk7205v200 kern.err kernel: goke-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
Jan  5 21:11:41 openipc-gk7205v200 kern.err kernel: goke-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
Jan  5 21:11:41 openipc-gk7205v200 kern.err kernel: goke-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
Jan  5 21:11:41 openipc-gk7205v200 kern.err kernel: goke-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
Jan  5 21:11:41 openipc-gk7205v200 kern.err kernel: goke-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
Jan  5 21:11:41 openipc-gk7205v200 kern.err kernel: goke-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
Jan  5 21:11:41 openipc-gk7205v200 kern.err kernel: goke-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
Jan  5 21:11:41 openipc-gk7205v200 kern.err kernel: goke-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
Jan  5 21:11:41 openipc-gk7205v200 kern.err kernel: goke-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
Jan  5 21:11:41 openipc-gk7205v200 kern.err kernel: goke-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
Jan  5 21:11:41 openipc-gk7205v200 kern.err kernel: goke-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
Jan  5 21:11:41 openipc-gk7205v200 kern.err kernel: goke-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
Jan  5 21:11:41 openipc-gk7205v200 kern.err kernel: goke-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
Jan  5 21:11:41 openipc-gk7205v200 kern.err kernel: goke-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
Jan  5 21:11:41 openipc-gk7205v200 kern.err kernel: goke-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
Jan  5 21:11:41 openipc-gk7205v200 kern.err kernel: goke-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
Jan  5 21:11:41 openipc-gk7205v200 kern.err kernel: goke-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
Jan  5 21:11:41 openipc-gk7205v200 kern.err kernel: goke-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
Jan  5 21:11:41 openipc-gk7205v200 kern.err kernel: goke-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
```

Any help appreciated! 
Many thanks

Yersi88, 2024-04-03:
> Did you ever get this figure out? I can't even get mine to display from RTSP on blue iris. Have not tried on frigate yet.

I did not. Pretty much gave up on OpenIPC due to lack of support.

cwtufts, 2024-04-04:
So I actually got mine figured out. It took forever but finally got it working in frigate now I'm working on 2 way audio

Yersi88, 2024-04-04:
> So I actually got mine figured out. It took forever but finally got it working in frigate now I'm working on 2 way audio

Nice! Do you mind sharing your configs?

closed 2026-09-04 by openipc-ai