OpenIPC/firmware #1238: hi3518ev200 runs in kernel panic since two weeks or so

aruhri, 2024-01-06:
Making an upgrade from the camera ui it does not come up any more.
The console log via serial connection shows a kernel panic.

I flashed kernel and rootfs whith recent downloads (Jan 6), but the kernel panic remains.

```
List of all partitions:
0100           65536 ram0  (driver?)
0101           65536 ram1  (driver?)
0102           65536 ram2  (driver?)
0103           65536 ram3  (driver?)
0104           65536 ram4  (driver?)
0105           65536 ram5  (driver?)
0106           65536 ram6  (driver?)
0107           65536 ram7  (driver?)
0108           65536 ram8  (driver?)
0109           65536 ram9  (driver?)
010a           65536 ram10  (driver?)
010b           65536 ram11  (driver?)
010c           65536 ram12  (driver?)
010d           65536 ram13  (driver?)
010e           65536 ram14  (driver?)
010f           65536 ram15  (driver?)
1f00             256 mtdblock0  (driver?)
1f01              64 mtdblock1  (driver?)
1f02            3072 mtdblock2  (driver?)
1f03           10240 mtdblock3  (driver?)
1f04            2752 mtdblock4 random: fast init done
 (driver?)
b300        15558144 mmcblk0  driver: mmcblk
  b301          716800 mmcblk0p1 9fe658f6-01
  b302          819200 mmcblk0p2 9fe658f6-02
No filesystem could mount root, tried:  squashfs

Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(31,3)
CPU: 0 PID: 1 Comm: swapper Not tainted 4.9.37 #1
Hardware name: Hisilicon Hi3518EV20X (Flattened Device Tree)
[<c000e42c>] (unwind_backtrace) from [<c000c8f0>] (show_stack+0x10/0x14)
[<c000c8f0>] (show_stack) from [<c005eea8>] (panic+0xa4/0x22c)
[<c005eea8>] (panic) from [<c04991d4>] (mount_block_root+0x1d8/0x270)
[<c04991d4>] (mount_block_root) from [<c04995dc>] (prepare_namespace+0x174/0x1b4)
[<c04995dc>] (prepare_namespace) from [<c0498d6c>] (kernel_init_freeable+0x158/0x1b8)
[<c0498d6c>] (kernel_init_freeable) from [<c0394900>] (kernel_init+0x8/0xec)
[<c0394900>] (kernel_init) from [<c000a200>] (ret_from_fork+0x14/0x34)
Rebooting in 20 seconds..
```

In the logs I remarked that the partition table seems to have changed:

```
hisi-sfc hisi_spi_nor.0: SPI Nor ID Table Version 1.0
hisi-sfc hisi_spi_nor.0: all blocks is unlocked.
hisi-sfc hisi_spi_nor.0: w25q128 (Chipsize 16 Mbytes, Blocksize 64KiB)
5 cmdlinepart partitions found on MTD device hi_sfc
5 cmdlinepart partitions found on MTD device hi_sfc
Creating 5 MTD partitions on "hi_sfc":
0x000000000000-0x000000040000 : "boot"
0x000000040000-0x000000050000 : "env"
0x000000050000-0x000000350000 : "kernel"
0x000000350000-0x000000d50000 : "rootfs"
0x000000d50000-0x000001000000 : "rootfs_data"
SPI Nand ID Table Version 2.6
Cannot found a valid SPI Nand Device
hisi_spi_nand_probe(171): Error: driver probe, result: -19
 (null): /soc/mdio@10091100/ethernet-phy@1 has invalid PHY address
hisi_femac_get_phy_addr: get PHY address failed!
libphy: hisi_femac_mii_bus: probed
```
In the install [documentation](https://github.com/OpenIPC/device-cip-37210/blob/main/README.md#flashing-openipc-rootfs) it says the partition for the rootfs starts at `0x250000`.
But even, when I change to that value when flashing the rootfs I receive the kernel-panic.

themactep, 2024-01-07:
You have flashed a 16mb image. You need to run `run setnor16m` in uboot console to change partitioning in bootcmd.

aruhri, 2024-01-07:
I think thats not the problem because I have run that command. Somehow the addresses in the  documentation do not fit to the output in the kernel-message. I really followed the docs step by step.

themactep, 2024-01-07:
The documentation is for lite image, your recent patritioning is for ultimate. If you flashed kernel and rootfs as separate bins following that guide then most likely you need to revert to lite partitioning. Run `run setnor8m` and see if it finds the kernel that way.

aruhri, 2024-01-13:
I started from beginning following the instructions, downloaded the lite image. Whether I format as 8m or 16m I run in a Kernel-panic.
When flashing the ultimate image formatted as 16m with corrected addresses, I run too in a kernel panic.
Which means: the kernel is loading, but at a specific state it panics.
Investigating further.

But to say clearly: there is a fault in the docs. It is for lite and if lite needs 8m formatting, then it should say so. but it says `run setnor16m`.