OpenIPC/firmware #1273: GK7205V300 + sc223a_4l for FPV

Schildkroet, 2024-01-21:
Hello,

i have a GK7205V300 + sc5239p (detected as 223a). When i flash "Ultimate" FW, the camera works, if i set config "/etc/sensors/4l_sc223a_i2c_1080p.ini".
When i flash "FPV" FW, it doesn't work, because the driver seems to be not included.

Would it be possible to add the driver to V300 like it is done here?
https://github.com/OpenIPC/firmware/commit/dcb99dc3aec422c286e7ea577ca7e6b8c7047df7

RoboSchmied, 2024-01-21:
The installation of sensor drivers is described [here](https://github.com/OpenIPC/wiki/blob/master/en/firmware-sensor-install-sc223a.md).

Schildkroet, 2024-01-21:
But there are no files for my specific SoC.
Using other files broke my system

RoboSchmied, 2024-01-21:
v300 uses the same sensor files as [v200](https://github.com/OpenIPC/firmware/tree/master/general/package/goke-osdrv-gk7205v200/files/sensor). 
Please show the output of majestic (fpv).

Schildkroet, 2024-01-21:
How to show output of majestic?

RoboSchmied, 2024-01-21:
ssh to cam 
```
killall majestic
majestic
```

Schildkroet, 2024-01-21:
Original System:
[image]

After installing libs:
[image]

At least my system didnt crash this time, but i still get no output

RoboSchmied, 2024-01-21:
Thank you. Please compare your "Isp_FrameRate" in file /etc/sensors/sc223a_i2c_1080p.ini  with your video0.fps in /etc/majestic.yaml.

Or show me both files.

Schildkroet, 2024-01-21:
Both are set to 30fps

RoboSchmied, 2024-01-21:
please try setting them to 25 fps

Schildkroet, 2024-01-21:
Both are set to 25:
[image]

RoboSchmied, 2024-01-21:
You said it worked on ultimate using the 4l driver.  (4l_sc223a_i2c_1080p.ini and [libsns_sc223a_4l.so](https://github.com/OpenIPC/firmware/raw/master/general/package/goke-osdrv-gk7205v200/files/sensor/libsns_sc223a_4l.so))
You should try them.

delete the other sc223a files before

Schildkroet, 2024-01-21:
No, still not working (same behaviour)

RoboSchmied, 2024-01-21:
Please show your /etc/majestic.yaml.

Schildkroet, 2024-01-21:
```
system:
  webAdmin: disabled
  buffer: 1024
image:
  mirror: false
  flip: false
  rotate: none
  contrast: 50
  hue: 50
  saturation: 50
  luminance: 50
osd:
  enabled: false
  template: "%d.%m.%Y %H:%M:%S"
nightMode:
  enabled: false
records:
  enabled: false
  path: /mnt/mmcblk0p1/%F/%H.mp4
  maxUsage: 95
video0:
  enabled: true
  size: 1920x1080
  fps: 25
  bitrate: 5120
  codec: h264
  rcMode: cbr
  gopSize: 1.5
video1:
  enabled: false
jpeg:
  enabled: false
mjpeg:
  size: 640x360
  fps: 5
  bitrate: 1024
audio:
  enabled: false
  volume: auto
  srate: 8000
rtsp:
  enabled: true
  port: 554
hls:
  enabled: false
youtube:
  enabled: false
motionDetect:
  enabled: false
  visualize: false
  debug: false
ipeye:
  enabled: false
watchdog:
  enabled: true
  timeout: 300
isp:
 sensorConfig: /etc/sensors/4l_sc223a_i2c_1080p.ini
 slowShutter: disabled
netip:
  enabled: false
outgoing:
  enabled: true
  server: udp://127.0.0.1:5600

```

RoboSchmied, 2024-01-21:
Remove (or comment out) everything in the video0 section but not "enabled: true".

Schildkroet, 2024-01-21:
Still same Timeout error

RoboSchmied, 2024-01-21:
Please try the sc223a (not 4l) drivers again with those majestic settings.

You should be able to find the working files (.ini and .so and majestic.yaml) in your ultimate image.

Schildkroet, 2024-01-21:
This is the working output on Ultimate:
[image]

Is this helpful?

Schildkroet, 2024-01-21:
I set up my fpv image again with ultimate settings, but now i get this error:
[image]

Schildkroet, 2024-01-21:
So i have either a working wifi or camera

Problem seems to be somewhere with mipi
`[mipi]
lane_id = 0|1|-1|-1|-1|-1|-1|-1|      ;lane_id: -1 - disable`

if you change the first "1" to 2 (like in other config), wifi works, but not camera...

Schildkroet, 2024-01-21:
This driver doesn't work (Timeout error as shown before). I have to use 4l

Schildkroet, 2024-01-21:
It seems to cause problems with wifibroadcast. Wifi driver is loaded and i can start wifibroadcast. But as soon as i start majestetic, wifibroadcast reports a socket error.
[image]

RoboSchmied, 2024-01-21:
This is a new issue. (maybe injection capable drivers are not in the fpv image)
Please post your majestic.yaml video0 part that solved the sensor problem.

Thank you

Schildkroet, 2024-01-21:
It seems majestic is crashing wifibroadcast when connecting. If i change port 5600 to something else (only in 1 config), i still get the connection refused, but at least nothing is crashing.

Schildkroet, 2024-01-21:
```
system:
  webAdmin: enabled
  buffer: 1024
image:
  mirror: false
  flip: false
  rotate: none
  contrast: 50
  hue: 50
  saturation: 50
  luminance: 50
osd:
  enabled: false
  template: "%d.%m.%Y %H:%M:%S"
nightMode:
  enabled: false
records:
  enabled: false
  path: /mnt/mmcblk0p1/%F/%H.mp4
  maxUsage: 95
video0:
  enabled: true
  crop: 0x0x1280x720
video1:
  enabled: false
jpeg:
  enabled: false
mjpeg:
  size: 640x360
  fps: 5
  bitrate: 1024
audio:
  enabled: false
  volume: auto
  srate: 8000
rtsp:
  enabled: true
  port: 554
hls:
  enabled: false
youtube:
  enabled: false
motionDetect:
  enabled: false
  visualize: false
  debug: false
ipeye:
  enabled: false
watchdog:
  enabled: true
  timeout: 300
isp:
  slowShutter: disabled
  sensorConfig: /etc/sensors/4l_sc223a_i2c_1080p.ini
netip:
  enabled: false
outgoing:
  enabled: true
  server: udp://127.0.0.1:5600
```

Schildkroet, 2024-01-21:
Wifi:
```
usb 1-1: ath9k_htc: Firmware ath9k_htc/htc_9271-1.4.0.fw requested
usbcore: registered new interface driver ath9k_htc
usb 1-1: ath9k_htc: Transferred FW: ath9k_htc/htc_9271-1.4.0.fw, size: 51008
ath9k_htc 1-1:1.0: ath9k_htc: HTC initialized with 33 credits
ath9k_htc 1-1:1.0: ath9k_htc: FW Version: 1.4
ath9k_htc 1-1:1.0: FW RMW support: On
ath: EEPROM regdomain: 0x0
ath: EEPROM indicates default country code should be used
ath: doing EEPROM country->regdmn map search
ath: country maps to regdmn code: 0x3a
ath: Country alpha2 being used: US
ath: Regpair used: 0x3a
ieee80211 phy0: Atheros AR9271 Rev:1
ATH: TX Power set: 20
ATH: TX Power set: 20
ATH: TX Power set: 20
ATH: TX Power set: 20
ATH: TX Power set: 20
device wlan0 entered promiscuous mode
usb 1-1: USB disconnect, device number 2
usb 1-1: ath9k_htc: USB layer deinitialized
usb 1-1: new high-speed USB device number 3 using xhci-hcd
usb 1-1: ath9k_htc: Firmware ath9k_htc/htc_9271-1.4.0.fw requested
usb 1-1: ath9k_htc: Transferred FW: ath9k_htc/htc_9271-1.4.0.fw, size: 51008
ath9k_htc 1-1:1.0: ath9k_htc: HTC initialized with 33 credits
ath9k_htc 1-1:1.0: ath9k_htc: FW Version: 1.4
ath9k_htc 1-1:1.0: FW RMW support: On
ath: EEPROM regdomain: 0x0
ath: EEPROM indicates default country code should be used
ath: doing EEPROM country->regdmn map search
ath: country maps to regdmn code: 0x3a
ath: Country alpha2 being used: US
ath: Regpair used: 0x3a
ieee80211 phy1: Atheros AR9271 Rev:1
```

Schildkroet, 2024-01-22:
Turns out the injection problem was a power supply issue.
Everything seems to work now.
Thanks for the help

closed 2024-01-22 by Schildkroet