OpenIPC/firmware #1276: Hi, this platform might be interested in adding support for Novatek NT98528 as it is not listed

SidWu04, 2024-01-23:
Novatek NT98528 is a SoC for AI or IP-CAM. It can be used in a variety of edge computing solutions. It has built-in H264/H265/MJPEG encoder.
And there is a complete SDK, the core version is 4.19.91, I saw that OpenIPC itself has listed Novatek NT98566, and its SDK is compatible with the NT9852x series,
Is it possible to consider adding more efficient NT98528 Soc support?

If there is OpenIPC support, I will use it for complete testing of ONVIF & RTSP.

SidWu04, 2024-01-23:
[image]
[image]
Attached are 2 more pictures for reference

widgetii, 2024-01-23:
> there is a complete SDK, the core version is 4.19.91

Do you have the SDK?

SidWu04, 2024-01-23:
I will ask Novatek supplier about this again

ppyzik, 2025-07-29:
https://github.com/hn/reolink-camera
Just leaving it here for future research