OpenIPC/firmware #1286: Unsupported sensor

29ru, 2024-01-29:
Hello!
I have a camera with an SSC335DE processor. Based on the Rostelecom firmware log, the sensor in the [SP2329](https://github.com/OpenIPC/firmware/files/14084387/sp2329.zip) camera. OpenIPC firmware does not support this sensor. I loaded the driver and profile extracted from the Rostelecom firmware into the camera, but the kernel still reports that 'UNSUPPORTED sensor'.
Please add SP2329 sensor support to the firmware.
Best regards, 29ru

FlyRouter, 2024-02-01:
Good afternoon
Thank you for your interest in our project.
Unfortunately, we do not have this sensor on our laboratory benches or data on it.
Please provide all the information you have, the binary driver and the calibration file.
To speed up communication, you can use our chats in Telegram.
https://openipc.org/our-channels

29ru, 2024-02-01:
Good day
binary driver and calibration in archive [SP2329](https://github.com/OpenIPC/firmware/files/14084387/sp2329.zip)
posted it in the initial post

29ru, 2024-02-01:
**modeprobe** still reports that driver of SP2329 module not found in files modules.dep and modules.dep.bin.

29ru, 2024-02-08:
It seems **majestic** cannot detect the sensor (**ipcinfo** and **ipctool** also do not detect the name of the sensor). How to force **majestic** to use a specific driver?

29ru, 2024-02-08:
> `SENSOR=sp2329 majestic`

in majectic.yaml or env?

29ru, 2024-02-08:
> > > `SENSOR=sp2329 majestic`
> > 
> > 
> > in majectic.yaml or env?
> 
> run this command

no effect. majestic: Canot query sensor index

29ru, 2024-02-08:
> Is sensor driver loaded successfully in the beginning? Did you get any error message when insmod the driver? what does lsmod say?

lsmod no comments

29ru, 2024-02-08:
> > > Is sensor driver loaded successfully in the beginning? Did you get any error message when insmod the driver? what does lsmod say?
> > 
> > 
> > lsmod no comments
> 
> nothing with lsmod? Can't be

sorry, I typed it wrong

29ru, 2024-02-08:
root@openipc-ssc335de:~# lsmod
Module                  Size  Used by    Tainted: P
mi_venc               159334  0
mi_vpe                132610  0
mi_vif                 39235  0
mi_divp                48536  0
mi_shadow              34375  0
mi_sensor              21952  0
mi_ao                  72914  0
mi_ai                 194338  0
mi_rgn                 74250  2 mi_vpe,mi_divp
mi_sys                415223  9 mi_venc,mi_vpe,mi_vif,mi_divp,mi_shadow,mi_sensor,mi_ao,mi_ai,mi_rgn
mi_common               5818 10 mi_venc,mi_vpe,mi_vif,mi_divp,mi_shadow,mi_sensor,mi_ao,mi_ai,mi_rgn,mi_sys
mhal                 1479105  9 mi_venc,mi_vpe,mi_vif,mi_divp,mi_sensor,mi_ao,mi_ai,mi_rgn,mi_sys
exfat                  48907  0

root@openipc-ssc335de:~# SENSOR=sp2329 majestic
22:49:45  <       majestic> [    main] main@213                      Majestic Ultimate for SigmaStar, version master+cfd55b3, built on 2024-01-21
22:49:45  <       majestic> [watchdog] watchdog_start@101            Detected infinity Watchdog
22:49:45  <       majestic> [watchdog] watchdog_start@121            Watchdog timeout set to 300 seconds
22:49:45  <       majestic> [   httpd] new_http_server@375           HTTP server started on 0.0.0.0:80
22:49:45  <       majestic> [app_conf] load_config@106               Using /etc/majestic.yaml as main configuration
22:49:45  <       majestic> [     sdk] start_sdk@1042                Initializing Sigmastar SDK, mi_sys_version_2.18.1
22:49:45  <       majestic> [     sdk] mi_snr_get_resolution@393     Cannot query sensor index
22:49:45  <       majestic> [     sdk] mi_snr_init@409               Cannot initialize sensor
22:49:45  <       majestic> [     sdk] mi_vif_init@439               Cannot enable sensor
22:49:45  <       majestic> [     sdk] start_sdk@1051                Error initializing vif
22:49:45  <       majestic> [    main] main@253                      Cannot start vendor sdk, continue Webserver

29ru, 2024-02-08:
root@openipc-ssc335de:~# majestic
22:03:33  <       majestic> [    main] main@213                      Majestic Ultimate for SigmaStar, version master+cfd55b3, built on 2024-01-21
22:03:33  <       majestic> [watchdog] watchdog_start@101            Detected infinity Watchdog
22:03:33  <       majestic> [watchdog] watchdog_start@121            Watchdog timeout set to 300 seconds
22:03:33  <       majestic> [   httpd] new_http_server@375           HTTP server started on 0.0.0.0:80
22:03:33  <       majestic> [app_conf] load_config@106               Using /etc/majestic.yaml as main configuration
22:03:33  <       majestic> [     sdk] start_sdk@1042                Initializing Sigmastar SDK, mi_sys_version_2.18.1
22:03:33  <       majestic> [     sdk] mi_snr_get_resolution@397     Sensor index 0: 1928x1088@30fps
22:03:33  <       majestic> [     log] printf@268                    CUS3A ver : Sigmastar Module misc_cus3a version: project_commit.a8e28a0 sdk_commit.d0df0a8 build_time.20201022150139
22:03:33  <       majestic> [     log] printf@268                    [MI_ISP_CUS3A_Enable] AE = 1, AWB = 0, AF = 0
22:03:33  <       majestic> [     log] printf@268                    change isproot : /etc/firmware
22:03:33  <       majestic> [     log] printf@268                    Load iq file /etc/firmware/iqfile0.bin
22:03:33  <       majestic> [     log] printf@268                    [MI_ISP_CUS3A_Enable] AE = 1, AWB = 1, AF = 0
22:03:33  <       majestic> [     sdk] mi_venc_create_chn@701        Channel 0: 1928x1088@30fps
22:03:33  <       majestic> [     sdk] mi_venc_create_chn@702        - [H265, 4096Kb, RC2, GOP30]
22:03:33  <       majestic> [     sdk] mi_venc_create_chn@754        - [minQp: 12, maxQp: 48, qpDelta: -4]
22:03:33  <       majestic> [     sdk] mi_venc_create_jpeg@593       Channel 2: 1928x1088@5fps
22:03:33  <       majestic> [     sdk] mi_venc_create_jpeg@594       - [JPEG, Q52]
22:03:33  <        3A_Proc> [     log] printf@268                    [sstar ae_init] isp_gain_max = 0x3fff, sgl_min = 0x400, sgl_max = 0x3e00, sgs_min  0x400, sgs_max = 0x3e00
22:03:33  <        3A_Proc> [     log] printf@268                    [sstar ae_init], shl_min = 60 us, shl_max = 333333 us, shl_step = 30057 us, shs_min = 60 us, shs_max = 333333 us, shs_step = 30057 us
22:03:33  <        3A_Proc> [     log] printf@268                    [sstar ae_init], FNum = 18, fps = 30
22:03:33  <        3A_Proc> [     log] printf@268                    change isproot : /etc/firmware
22:03:33  <        3A_Proc> [     log] printf@268                    Load iq file /etc/firmware/iqfile0.bin
22:03:34  <       majestic> [     sdk] mi_isp_get_file@109           Load isp file /etc/sensors/sp2329.bin
22:03:34  <       majestic> [     log] printf@268                    [MAJOR]: ispversion(1) in sdk, ispversion(0) in binfile.
22:03:34  <       majestic> [     log] printf@268                    [MINOR]: ispversion(6) in sdk, ispversion(0) in binfile.
22:03:34  <       majestic> [    puts]                               ooooops,SDK & iqbinfile major version is not match.
22:03:34  <       majestic> [    puts]                               warning warning!!!SDK & iqbinfile minor version is not match.
22:03:34  <       majestic> [    puts]                               Key check failed!
22:03:34  <       majestic> [     log] printf@268                    Magic_key = 0x0, User_key = 0x4d2
22:03:34  <       majestic> [     log] printf@268                    ******************** Load api bin Fail - ERROR CODE (0xffffffff) ********************
22:03:34  <       majestic> [     sdk] start_sdk@1088                Error loading isp file
22:03:34  <       majestic> [    main] main@253                      Cannot start vendor sdk, continue Webserver

29ru, 2024-02-08:
> seems calibration file is password protected, try deleting `/etc/sensors/sp2329.bin` then run majestic again

It's work! Thank You for support!

P.S. In HexEdit sp2329.bin similar to others

29ru, 2024-02-08:
Files sp2329.bin and /etc/firmware/iqfile0.bin are equivalent

closed 2024-02-09 by 29ru

dos-007, 2024-03-30:
Hi, tell me why gedit /usr/bin/load_setmaster, add sp2329 into the set_sensor function lsmod null

dos-007, 2024-03-30:
root@openipc-ssc335de:~# lsmod
Module                  Size  Used by    Not tainted

dos-007, 2024-03-30:
root@openipc-ssc335de:~# majestic
21:35:47  <       majestic> [    main] main@240                      Majestic Ultimate for SigmaStar, version master+d389254, built on 2024-03-17
21:35:47  <       majestic> [watchdog] watchdog_start@91             Detected infinity Watchdog
21:35:47  <       majestic> [watchdog] watchdog_start@111            Watchdog timeout set to 300 seconds
21:35:47  <       majestic> [   httpd] start_web_server@387          Set server path: /var/www
21:35:47  <       majestic> [   httpd] new_http_server@375           HTTP server started on 0.0.0.0:80
21:35:48  <       majestic> [    main] init_sdk@66                   Loading main configuration: /etc/majestic.yaml
21:35:48  <       majestic> [     sdk] start_sdk@1058                Initializing Sigmastar SDK, mi_sys_version_2.18.1
21:35:48  <       majestic> [     log] printf@292                    [sys] failed to open /dev/mi/sys!(Bad file descriptor)
21:35:48  <       majestic> [     log] printf@292                    [sys] failed to ioctl 0x80046900!(Bad file descriptor)
21:35:48  <       majestic> [     sdk] start_sdk@1062                Error initializing sys
21:35:48  <       majestic> [    main] init_sdk@73                   Cannot start SDK
^C21:35:49  <       majestic> [  levent] signal_cb@53                  Interrupt signal received
21:35:49  <       majestic> [    main] main@277                      Shutdown main thread