OpenIPC/firmware #1290: Support for OpenIPC for RV1106 & RK3566

232485, 2024-01-31:
I have been working on a system streams from multiple cameras apart from sensors. I was considering using OpenIPC. I need help in understanding if OpenIPC is the right fit for such a use case.

1. *RV1106* at camera source for video streaming, Object or scene detection at the edge for single channel system in addition to audio streaming bi-directional.
2. *RK3566* at camera source for video streaming, Object or scene detection at the edge for multiple channel system in addition to audio streaming bi-directional.

Any alternatives, Evaluation boards to help achieve this will be helpful.

FlyRouter, 2024-02-01:
Good afternoon
Thank you for your interest in our project.
Unfortunately, at the moment we do not have free specialists and resources to solve your problems.

Pi-Boss, 2024-10-23:
> I have been working on a system streams from multiple cameras apart from sensors. I was considering using OpenIPC. I need help in understanding if OpenIPC is the right fit for such a use case.
>
> 1. [...]

我正在制作研究rv1106运行openipc的可能性

closed 2026-09-04 by openipc-ai