OpenIPC/firmware #1295: majestic can not set GPIO pin of audio speaker

githubtmp7, 2024-02-03:
Прописал в majestic GPIO pin of audio speaker 37, в отладочных сообщениях вижу попытку установки нужного значения
majestic> [    gpio] set_gpio@25                   set_gpio(37, 1)
но фактического  значение gpio не меняется и звук не появляется.
Если в тоже самое время установить значение вручную командой ipctool gpio set 37 1 , то звук появляется.
Для параметров GPIO pin1 of signal for IRcut filter и GPIO pin to turn on night mode illumination majestic меняет значения gpio корректно.

По предложению сообщества в качестве теста,  попробовал прописать значение 37 для GPIO pin1 of signal for IRcut filter.
При старте majestic с такой настройкой появляется сообщение:
gpio-37 (sysfs): _gpiod_direction_output_raw: tried to set a GPIO tied to an IRQ as output

root@openipc-hi3518ev200:/tmp#majestic -v
master+a2623bc, 2024-01-29

root@openipc-hi3518ev200:/tmp# ipctool
---
chip:
  vendor: HiSilicon
  model: 3518EV200
board:
  vendor: OpenIPC
  version: 2.4.01.30
  possible-IR-cut-GPIO: 37,50
ethernet:
  mac: "7a:a0:dc:66:3b:78"
  u-mdio-phyaddr: 1
  phy-id: 0x0000ffff
  d-mdio-phyaddr: 0
rom:
- type: nor
  block: 64K
  partitions:
    - name: boot
      size: 0x40000
      sha1: 57ea3977
    - name: env
      size: 0x10000
      sha1: 9fb59454
      contains:
        - name: uboot-env
          offset: 0x0
    - name: kernel
      size: 0x300000
      sha1: a42a613c
    - name: rootfs
      size: 0xa00000
      path: /,squashfs
      sha1: 7ac389f5
    - name: rootfs_data
      size: 0x2b0000
      path: /overlay,jffs2,rw
  size: 16M
  addr-mode: 3-byte
ram:
  total: 64M
  media: 26M
firmware:
  u-boot: "2010.06-dirty (Jan 29 2024 - 11:57:03)"
  kernel: "4.9.37 (Tue Jan 30 11:12:24 UTC 2024)"
  toolchain: buildroot-gcc-12.3.0
  sdk: "Hi3518EV200_MPP_V1.0.5.0 B060 Release (Mar 21 2018, 08:56:43)"
  main-app: majestic
sensors:
- vendor: SmartSens
  model: SC2035
  control:
    bus: 0
    type: i2c
    addr: 0x60
  data:
    type: DC
  clock: 24MHz

root@openipc-hi3518ev200:/tmp# ipctool reginfo
muxctrl_reg0 0x200f0000 0x1 GPIO0_4 [SENSOR_CLK]
muxctrl_reg1 0x200f0004 0 [SENSOR_RSTN] GPIO0_5
muxctrl_reg2 0x200f0008 0x1 GPIO0_6 [FLASH_TRIG] SFC_EMMC_BOOT_MODE SPI1_CSN1 VI_VS_BT1120
muxctrl_reg3 0x200f000c 0 [GPIO0_7] SHUTTER_TRIG SFC_DEVICE_MODE VI_HS_BT1120
muxctrl_reg4 0x200f0010 0x1 GPIO2_0 [RMII_CLK] VO_CLK SDIO1_CCLK_OUT
muxctrl_reg5 0x200f0014 0x1 GPIO2_1 [RMII_TX_EN] VO_VS SDIO1_CARD_DETECT
muxctrl_reg6 0x200f0018 0x1 GPIO2_2 [RMII_TXD0] VO_DATA5 SDIO1_CWPR
muxctrl_reg7 0x200f001c 0x1 GPIO2_3 [RMII_TXD1] VO_DE SDIO1_CDATA1
muxctrl_reg8 0x200f0020 0x1 GPIO2_4 [RMII_RX_DV] VO_DATA7 SDIO1_CDATA0
muxctrl_reg9 0x200f0024 0x1 GPIO2_5 [RMII_RXD0] VO_DATA2 SDIO1_CDATA3
muxctrl_reg10 0x200f0028 0x1 GPIO2_6 [RMII_RXD1] VO_DATA3 SDIO1_CCMD
muxctrl_reg11 0x200f002c 0x1 GPIO2_7 [EPHY_RST] BOOT_SEL VO_HS SDIO1_CARD_POWER_EN
muxctrl_reg12 0x200f0030 0 [GPIO0_3] SPI1_CSN1 VO_DATA0
muxctrl_reg13 0x200f0034 0x1 GPIO3_0 [EPHY_CLK] VO_DATA1 SDIO1_CDATA2
muxctrl_reg14 0x200f0038 0x1 GPIO3_1 [MDCK] BOOTROM_SEL VO_DATA6
muxctrl_reg15 0x200f003c 0x1 GPIO3_2 [MDIO] VO_DATA4
muxctrl_reg16 0x200f0040 0x2 GPIO3_3 SPI0_SCLK [I2C0_SCL]
muxctrl_reg17 0x200f0044 0x2 GPIO3_4 SPI0_SDO [I2C0_SDA]
muxctrl_reg18 0x200f0048 0 [GPIO3_5] SPI0_SDI
muxctrl_reg19 0x200f004c 0 [GPIO3_6] SPI0_CSN
muxctrl_reg20 0x200f0050 0 [GPIO3_7] SPI1_SCLK I2C1_SCL
muxctrl_reg21 0x200f0054 0 [GPIO4_0] SPI1_SDO I2C1_SDA
muxctrl_reg22 0x200f0058 0 [GPIO4_1] SPI1_SDI
muxctrl_reg23 0x200f005c 0 [GPIO4_2] SPI1_CSN0
muxctrl_reg24 0x200f0060 0 [GPIO4_3] I2C2_SDA
muxctrl_reg25 0x200f0064 0 [GPIO4_4] I2C2_SCL
muxctrl_reg26 0x200f0068 0 [GPIO4_5] USB_OVRCUR
muxctrl_reg27 0x200f006c 0 [GPIO4_6] USB_PWREN
muxctrl_reg28 0x200f0070 0 [GPIO0_0] IR_IN TEMPER_DQ
muxctrl_reg29 0x200f0074 0 [GPIO0_1] TEMPER_DQ
muxctrl_reg30 0x200f0078 0 [GPIO0_2] TEMPER_DQ
muxctrl_reg31 0x200f007c 0x1 GPIO1_0 [VI_DATA13] I2S_BCLK_TX PWM0
muxctrl_reg32 0x200f0080 0x1 GPIO1_1 [VI_DATA10] I2S_SD_TX UART1_TXD
muxctrl_reg33 0x200f0084 0x1 GPIO1_2 [VI_DATA12] I2S_MCLK UART1_CTSN
muxctrl_reg34 0x200f0088 0x1 GPIO1_3 [VI_DATA11] I2S_WS_TX UART2_RXD
muxctrl_reg35 0x200f008c 0x2 GPIO1_4 VI_DATA15 [VI_VS_SEN] I2S_WS_RX UART1_RXD
muxctrl_reg36 0x200f0090 0x2 GPIO1_5 VI_DATA14 [VI_HS_SEN] I2S_BCLK_RX UART1_RTSN
muxctrl_reg37 0x200f0094 0x1 GPIO1_6 [VI_DATA9] I2S_SD_RX UART2_TXD
muxctrl_reg38 0x200f0098 0x1 GPIO1_7 [SDIO0_CARD_POWER_EN]
muxctrl_reg39 0x200f009c 0 [GPIO4_7] SDIO0_CARD_DETECT
muxctrl_reg40 0x200f00a0 0 [GPIO5_0] SDIO0_CWPR
muxctrl_reg41 0x200f00a4 0 [GPIO5_1] SDIO0_CCLK_OUT
muxctrl_reg42 0x200f00a8 0 [GPIO5_2] SDIO0_CCMD
muxctrl_reg43 0x200f00ac 0 [GPIO5_3] SDIO0_CDATA0
muxctrl_reg44 0x200f00b0 0x2 TEST_CLK SDIO0_CDATA1 [GPIO5_4]
muxctrl_reg45 0x200f00b4 0 [GPIO5_5] SDIO0_CDATA2
muxctrl_reg46 0x200f00b8 0 [GPIO5_6] SDIO0_CDATA3
muxctrl_reg47 0x200f00bc 0 [GPIO5_7] EMMC_DATA6 I2S_SD_TX UART1_RTSN
muxctrl_reg48 0x200f00c0 0 [GPIO6_0] EMMC_DATA5 I2S_WS_TX UART1_RXD
muxctrl_reg49 0x200f00c4 0 [GPIO6_1] EMMC_DATA7 I2S_MCLK UART1_CTSN
muxctrl_reg50 0x200f00c8 0 [GPIO6_2] EMMC_DS I2S_SD_RX UART1_TXD
muxctrl_reg51 0x200f00cc 0 [GPIO6_3] EMMC_DATA1 UART2_RXD
muxctrl_reg52 0x200f00d0 0 [GPIO6_4] EMMC_DATA2 I2S_BCLK_TX UART2_TXD
muxctrl_reg53 0x200f00d4 0 [GPIO6_5] JTAG_TRSTN SPI1_CSN1 I2S_MCLK
muxctrl_reg54 0x200f00d8 0 [GPIO6_6] JTAG_TCK SPI1_SCLK I2S_WS_TX I2C1_SCL
muxctrl_reg55 0x200f00dc 0 [GPIO6_7] JTAG_TMS SPI1_CSN0 I2S_SD_TX
muxctrl_reg56 0x200f00e0 0 [GPIO7_0] JTAG_TDO SPI1_SDO I2S_SD_RX I2C1_SDA
muxctrl_reg57 0x200f00e4 0 [GPIO7_1] JTAG_TDI SPI1_SDI I2S_BCLK_TX
muxctrl_reg58 0x200f00e8 0x1 PMC_PWM [GPIO7_2] PWM0
muxctrl_reg59 0x200f00ec 0x1 PWM1 [GPIO7_3]
muxctrl_reg60 0x200f00f0 0x1 PWM2 [GPIO7_4] SPI1_CSN1
muxctrl_reg61 0x200f00f4 0x1 PWM3 [GPIO7_5]
muxctrl_reg62 0x200f00f8 0x1 SAR_ADC_CH0 [GPIO7_6]
muxctrl_reg63 0x200f00fc 0x1 SAR_ADC_CH1 [GPIO7_7]
muxctrl_reg64 0x200f0100 0x1 SAR_ADC_CH2 [GPIO8_0]
muxctrl_reg65 0x200f0104 0x1 SAR_ADC_CH3 [GPIO8_1]
root@openipc-hi3518ev200:/tmp#

githubtmp7, 2024-02-07:
Пересобрал ядро с поддержкой debugfs
попробовал на последнем доступном релизе
вывод cat /sys/kernel/debug/gpio
показывает, что majestic почему-то переводит gpio 37 в режим in lo IRQ
вероятно это происходит на этапе инициализации при загрузке аудио драйвера, majestic пытается записать 0  gpio 37

Коллеги кто имеет доступ к исходникам посмотрите пожалуйста. 

gpiochip4: GPIOs 32-39, parent: amba/20180000.gpio_chip, 20180000.gpio_chip:
 gpio-33  (                    |sysfs               ) out hi
 gpio-37  (                    |sysfs               ) in  lo IRQ

gpiochip5: GPIOs 40-47, parent: amba/20190000.gpio_chip, 20190000.gpio_chip:

gpiochip6: GPIOs 48-55, parent: amba/201a0000.gpio_chip, 201a0000.gpio_chip:
 gpio-50  (                    |sysfs               ) out lo

<14>Feb  7 15:06:42 majestic[6125]: Majestic Ultimate for HiSilicon, version master+c0a1beb, built on 2024-02-02
<15>Feb  7 15:06:42 majestic[6125]: Main audio codec is opus (other options: mp3, aac, pcm, alaw, ulaw)
<15>Feb  7 15:06:42 majestic[6125]: Detected Hisilicon Watchdog
<15>Feb  7 15:06:42 majestic[6125]: Watchdog timeout set to 300 seconds
<14>Feb  7 15:06:42 majestic[6125]: Using /etc/majestic.yaml as main configuration
<15>Feb  7 15:06:42 majestic[6125]: Main audio codec is opus (other options: mp3, aac, pcm, alaw, ulaw)
<14>Feb  7 15:06:43 majestic[6125]: Using /etc/sensors/sc2035_i2c_1080p.ini sensor configuration file
<15>Feb  7 15:06:43 majestic[6125]: App was built with MPP version: Hi3518EV200_MPP_V1.0.5.0.B060 Release
<15>Feb  7 15:06:43 majestic[6125]: Current MPP version: HI_VERSION=Hi3518EV200_MPP_V1.0.5.0 B060 Release
<15>Feb  7 15:06:43 majestic[6125]: sensor sc2035
<15>Feb  7 15:06:43 majestic[6125]:   input_mode CMOS_18V, WDR NONE
<15>Feb  7 15:06:43 majestic[6125]:   dev [1920x1080]@0x0 20fps, BGGR
<15>Feb  7 15:06:43 majestic[6125]: trying to load /usr/lib/sensors/libsns_sc2035.so
<15>Feb  7 15:06:43 majestic[6125]: Free MMZ mem before allocation: 26436KB
<15>Feb  7 15:06:43 majestic[6125]: VB configuration:
<15>Feb  7 15:06:43 majestic[6125]:   [0]: 3110400 x 3
<15>Feb  7 15:06:43 majestic[6125]:   [1]: 4147200 x 1
<15>Feb  7 15:06:43 majestic[6125]: Free MMZ mem after allocation: 13260KB
<15>Feb  7 15:06:43 majestic[6125]: Sensor driver loaded
<15>Feb  7 15:06:43 majestic[6125]: SDK is in 'offline' mode
<14>Feb  7 15:06:43 majestic[6125]: linear mode
<15>Feb  7 15:06:43 majestic[6125]: cmos_set_image_mode.............................655....
<14>Feb  7 15:06:43 majestic[6125]: =========================================================
<14>Feb  7 15:06:43 majestic[6125]: ===sc2035 sensor 1080P30fps(Parallel port) init success222!=====
<14>Feb  7 15:06:43 majestic[6125]: =========================================================
<15>Feb  7 15:06:43 majestic[6125]: H.264 1920x1080 20fps 4096Kbit  20 GOP
<15>Feb  7 15:06:43 majestic[6125]: u32IntTime:0x10...........0x1........0..
<14>Feb  7 15:06:43 majestic[6125]: Loading IQ profile /etc/sensors/iq/default.ini
<15>Feb  7 15:06:43 majestic[6125]:   new venc: 0   vpss_grp: 0,   vpss_chn: 0
<15>Feb  7 15:06:43 majestic[6125]:   venc took 7704KB of MMZ [1518]
<15>Feb  7 15:06:43 majestic[6125]: u32IntTime:0x14...........0x1........0x40..
<15>Feb  7 15:06:43 majestic[6125]: u32IntTime:0x1a...........0x1........0xa0..
<15>Feb  7 15:06:43 majestic[6125]: u32IntTime:0x22...........0x2........0x20..
<15>Feb  7 15:06:43 majestic[6125]: i:0..g_stSnsRegsInfo:0x2  g_stPreSnsRegsInfo:0x1.....
<14>Feb  7 15:06:43 majestic[6125]: #####################HI_TRUE....
<15>Feb  7 15:06:43 majestic[6125]: OSD initialized
<15>Feb  7 15:06:43 majestic[6125]: u32IntTime:0x2c...........0x2........0xc0..
<15>Feb  7 15:06:44 majestic[6125]:   OSD took 176KB of MMZ
<15>Feb  7 15:06:44 majestic[6125]: u32IntTime:0x39...........0x3........0x90..
<15>Feb  7 15:06:44 majestic[6125]: Ai(0,0) bind to AencChn:0 ok!
<15>Feb  7 15:06:44 majestic[6125]: i:0..g_stSnsRegsInfo:0x3  g_stPreSnsRegsInfo:0x2.....
<14>Feb  7 15:06:44 majestic[6125]: #####################HI_TRUE....
<15>Feb  7 15:06:44 majestic[6125]:   Audio took 328KB of MMZ
<15>Feb  7 15:06:44 majestic[6125]: u32IntTime:0x49...........0x4........0x90..
<15>Feb  7 15:06:44 majestic[6125]: i:0..g_stSnsRegsInfo:0x4  g_stPreSnsRegsInfo:0x3.....
<14>Feb  7 15:06:44 majestic[6125]: #####################HI_TRUE....
<15>Feb  7 15:06:44 majestic[6125]: set_gpio(37, 0)
<15>Feb  7 15:06:44 majestic[6125]: bind adec:0 to ao(0,0) ok 
<14>Feb  7 15:06:44 majestic[6125]: Image tuning task started
<15>Feb  7 15:06:44 majestic[6125]: HiSilicon SDK started

widgetii, 2024-02-07:
Просьба проверить свежую сборку Majestic

githubtmp7, 2024-02-08:
Добрый день!
Для моей платформы новая версия штатным образом не доступна, поделитесь ссылкой попробую 
updatemajestic.sh 
Majestic Updater

Retrieving update info
from http://openipc.s3-eu-west-1.amazonaws.com/majestic.hi3516cv200.ultimate.master.tar.meta
RESULT:

Comparing versions
Installed: master+c0a1beb, 2024-02-02
Available: master+1419d09, 2024-02-07

Updating Majestic
from http://openipc.s3-eu-west-1.amazonaws.com/majestic.hi3516cv200.ultimate.master.tar.bz2
Killing Majestic process
RESULT:
RESULT:
RESULT:
Done.
Majestic master+c0a1beb, 2024-02-02 installed in overlay.

widgetii, 2024-02-09:
Релизы публикуются на канале https://t.me/openipc_dev

closed 2024-02-16 by githubtmp7