OpenIPC/firmware #1315: FIrmware upgrade bump is missing

vinibali, 2024-02-15:
Hello there!
I wanted to upgrade the firmware trough Majestic on my gk7205v210 camera.
Originally it was installed with the following version:

```
# cat /etc/*release
NAME=Buildroot
VERSION=-g570c995a-dirty
ID=buildroot
VERSION_ID=2021.02.12
PRETTY_NAME="Buildroot 2021.02.12"
OPENIPC_VERSION=2.3.05.06
GITHUB_VERSION="master+570c995a, 2023-05-06"
BUILD_OPTION=lite
# uname -a
Linux openipc-gk7205v210 4.9.37 #1 Sat May 6 00:35:21 UTC 2023 armv7l GNU/Linux
```
Firmware versions according to Majestic update UI:
```
Firmware
Version

Installed
    2.3.05.06
On GitHub
    2.3.02.15
```
I assume the correct version number on the GitHub should be: 2.4.02.15
Best regards

vinibali, 2024-02-18:
MIght relates to:
https://github.com/OpenIPC/webui/issues/251

widgetii, 2026-10-20:
I checked this against current firmware and it no longer reproduces.

The "On GitHub" figure came from the old WebUI, which read a GitHub release label. That label could be older than the build already on the camera. That code is gone. The current majestic-webui (the `dist` release the firmware package ships) takes the latest version from the board's own build manifest through `sysupgrade --list-builds`. It then compares that build with the installed one by revision and build date, not by a version string. The Update page, the update notice and the scheduled auto-update all share that one comparison.

Checked on a lab hi3516av300 (`hi3516av300_lite`, 2.6.10.13):

```
# sysupgrade --list-builds
Available builds for hi3516av300_lite (newest first):
    nightly-20261019-5e914f7
    nightly-20261018-cae8b09
    ...
```

The comparison, run on the camera against that latest build:

```
master+570c995a, 2023-05-06 -> true    (the build in this report: update offered)
master+5e914f7, 2026-10-19  -> false   (already current)
master+e8d577a, 2026-10-17  -> true
local+build, 2026-10-13     -> null    (local build: image offered, not called newer)
```

A camera on an old build is now offered the actual newest image for its board, and a camera that is already current is told so. Closing. If you still see a wrong version on a current nightly, please reopen and include `cat /etc/os-release` and the output of `sysupgrade --list-builds`.

closed 2026-10-20 by widgetii