OpenIPC/firmware #1325: jco h42 module cpu ingenic T31n，sensor jxq03，no video。

catulacnpoly, 2024-02-25:
the module is jco h42 the cpu is ingenic t31n，cmos sensor is jxq03，at the first I try to flash it use the openipc official mothed，but failed，then I flash it use a ch341 programmer to do it well，it can boot the openipc_lite,but sadly,I got no video out at preview page,it seems jxq03 works well,and majestic works well too,someone tell me what the reason?

I copy the screen to youtube：https://youtu.be/0ePk9cPmWaU

boot info
[image]

U-Boot SPL 2013.07-g05f0cca (Oct 10 2023 - 00:04:03)
Timer init
CLK stop
PLL init
pll_init:366
pll_cfg.pdiv = 10, pll_cfg.h2div = 5, pll_cfg.h0div = 5, pll_cfg.cdiv = 1, pll_c                                                                             fg.l2div = 2
nf=118 nr = 1 od0 = 1 od1 = 2
cppcr is 07605100
CPM_CPAPCR 0750510d
nf=84 nr = 1 od0 = 1 od1 = 2
cppcr is 05405100
CPM_CPMPCR 07d0590d
nf=100 nr = 1 od0 = 1 od1 = 2
cppcr is 06405100
CPM_CPVPCR 0640510d
cppcr 0x9a773310
apll_freq 1404000000
mpll_freq 1000000000
vpll_freq = 1200000000
ddr sel mpll, cpu sel apll
ddrfreq 500000000
cclk  1404000000
l2clk 702000000
h0clk 200000000
h2clk 200000000
pclk  100000000
CLK init
SDRAM init
sdram init start
ddr_inno_phy_init ..!
phy reg = 0x00000007, CL = 0x00000007
ddr_inno_phy_init ..! 11:  00000004
ddr_inno_phy_init ..! 22:  00000006
ddr_inno_phy_init ..! 33:  00000006
REG_DDR_LMR: 00000210
REG_DDR_LMR: 00000310
REG_DDR_LMR: 00000110
REG_DDR_LMR, MR0: 00f73011
T31_0x5: 00000007
T31_0x15: 0000000c
T31_0x4: 00000000
T31_0x14: 00000002
INNO_TRAINING_CTRL 1: 00000000
INNO_TRAINING_CTRL 2: 000000a1
T31_cc: 00000003
INNO_TRAINING_CTRL 3: 000000a0
T31_118: 0000003c
T31_158: 0000003c
T31_190: 00000021
T31_194: 0000001f
jz-04 :  0x00000051
jz-08 :  0x000000a0
jz-28 :  0x00000024
DDR PHY init OK
INNO_DQ_WIDTH   :00000003
INNO_PLL_FBDIV  :00000014
INNO_PLL_PDIV   :00000005
INNO_MEM_CFG    :00000051
INNO_PLL_CTRL   :00000018
INNO_CHANNEL_EN :0000000d
INNO_CWL        :00000006
INNO_CL         :00000007
DDR Controller init
[... 219 lines cut ...]
udhcpc: broadcasting discover
udhcpc: broadcasting discover
udhcpc: broadcasting discover
udhcpc: broadcasting discover
udhcpc: no lease, forking to background
Starting ntpd: OK
Starting dropbear sshd: OK
Production mode
Starting httpd: OK
Starting mini-snmpd: DISABLED, OK
Starting telnetd: DISABLED, OK
Starting crond: OK

Loading of kernel modules...
ingenic: Get data from sinfo and set SENSOR as jxq03
ingenic: Write detected jxq03 to U-Boot ENV
--------------------
AVPU_PARAM: clk_name=vpll avpu_clk=400000000
ISP_PARAM: isp_clk=150000000
SENSOR: jxq03
SENSOR_PARAM:
--------------------
Starting telegram-bot.sh: OK
ingenic: Loading video system has started...
Starting majestic: OK

cronyx, 2024-02-26:
try run this commands via ssh and reboot:

```
cli -s .isp.blkCnt 1
fw_setenv osmem 39M
fw_setenv rmem 25M@0x2700000
```

catulacnpoly, 2024-02-27:
Thank you cronyx ,I got the video preview.

closed 2024-02-27 by catulacnpoly