OpenIPC/firmware #1327: WebRTC / go2RTC - Almost no documentation to get it working

mogupta, 2024-02-25:
Hello everyone
I love this project. I was able to flash the provided bin file to Hi3518EV200 , the board works fine.
Also tried building on ubuntu and installing it on the board, works fine. 

How to get WebRTC to work with Amazon kinesis ? I am looking for peer to peer video steaming solution. 
Tried to install go2rtc by enabling the package in the config file but unable to make. it ends up with the following error 

[image]

widgetii, 2024-02-25:
Unfortunately, it won't work on such ancient boards due to memory limitations. Consider upgrading your hardware or using a separate server in between

closed 2024-02-25 by widgetii

mogupta, 2024-02-25:
Hi @widgetii can you suggest a economical board ? In Gok or hisilicon ?

How we get amazon kinesis working ? Cannot build the image

widgetii, 2024-02-25:
You should write your own code based on their official library https://github.com/awslabs/amazon-kinesis-video-streams-webrtc-sdk-c

We've done support for one commercial clients, but this solution will never be open sourced

mogupta, 2024-02-25:
1. Aws producer and AWS webrtc are already included as packages right ? Are they not useful ? 
2. How to build image with go2rtc enabled ? Getting error message , please see first comment for screenshot   . 
3. Openipc/webrtc repo is of no use ?
4. Go2rtc will work ? If I upgrade the hardware

widgetii, 2024-02-26:
I've never worked with go2rtc so cannot say anything about it

AWS libraries are libraries, you cannot use them without writing a glue code