OpenIPC/firmware #1328: 君正T31ZX烧录OpenIPC启动问题

ljc111111111111111111111111111111, 2024-02-27:
君正T31ZX烧录OpenIPC固件后重新启动出现问题，按照安装指南操作
[image]
重启进入不了系统，一直rebooting，
[image]

cronyx, 2024-02-27:
登录到 u-boot shell 并运行命令 run setnor16m

ljc111111111111111111111111111111, 2024-02-27:
> 登录到 u-boot shell 并运行命令 run setnor16m
[image]
没有此指令

cronyx, 2024-02-27:
据我所知，你还没有刷新固件，因为你还有库存的u-boot

cronyx, 2024-02-27:
1.
```
setenv ipaddr 192.168.1.10; setenv serverip 192.168.1.254
mw.b 0x80600000 0xff 0x1000000
fatload mmc 0:1 0x80600000 openipc-t31zx-ultimate-16mb.bin
sf erase 0x0 0x1000000; sf write 0x80600000 0x0 0x1000000
reset
```

刷机成功后，您应该看到 OpenIPC# 而不是 isvp_t31#

2.
`run setnor16m`

ljc111111111111111111111111111111, 2024-02-27:
就是没有成功，一直在rebooting，那个isvp_t31#是我重新通过启动卡进入的Uboot shell界面

cronyx, 2024-02-27:
在u-boot openipc中，按ctrl+c组合键停止下载。 需要从 u-boot openipc（运行 printenv）而不是从库存获取附加信息。

ljc111111111111111111111111111111, 2024-02-28:
怎么进入u-boot openipc中呢？按照这里的操作`setenv ipaddr 192.168.1.10; setenv serverip 192.168.1.254
mw.b 0x80600000 0xff 0x1000000
fatload mmc 0:1 0x80600000 openipc-t31zx-ultimate-16mb.bin
sf erase 0x0 0x1000000; sf write 0x80600000 0x0 0x1000000
reset`后就一直在rebooting，按ctrl+c组合键也停不下来，如果说用启动卡进入uboot的话还是还是会进入它原来的固件，难道说制作启动卡时的固件就应该换吗？

ljc111111111111111111111111111111, 2024-02-28:
我不知道，我就是完全按照安装指南来的，但它就是一直rebooting，怎么也停不下来

ljc111111111111111111111111111111, 2024-02-28:
ok,感谢，成功进入

closed 2024-03-02 by ljc111111111111111111111111111111

jpfree, 2024-06-28:
我也是同样的问题，按流程走的，也在openipc的UBOOT下：
`
OpenIPC # printenv
baseaddr=0x80600000
baudrate=115200
bootargs=console=ttyS1,115200 panic=10 root=/dev/mtdblock3 init=/init mtdparts=jz_sfc:256k(boot),64k(env),2048k(kernel),${rootmtd}(rootfs),-(rootfs_data) mem=${osmem} rmem=${rmem} ${extras}
bootcmd=sf probe 0; setenv setargs setenv bootargs ${bootargs}; run setargs; sf read ${baseaddr} ${kernaddr} ${kernsize}; bootm ${baseaddr}
bootdelay=1
cmdnor=sf probe 0; setenv setargs setenv bootargs ${bootargs}; run setargs; sf read ${baseaddr} ${kernaddr} ${kernsize}; bootm ${baseaddr}
ethact=Jz4775-9161
ethaddr=00:00:23:34:45:66
extras=nogmac
gatewayip=192.168.1.1
ipaddr=192.168.1.10
kernaddr=0x50000
kernsize=0x200000
loads_echo=1
netmask=255.255.255.0
osmem=64M
rmem=64M@0x4000000
rootaddr=0x250000
rootmtd=5120k
rootsize=0x500000
sdcard=mmc rescan; setenv updatetool fatload mmc 0
serverip=192.168.1.254
soc=t31
stderr=serial
stdin=serial
stdout=serial
totalmem=128M
ubnor=sf probe 0; sf erase 0x0 ${kernaddr}; sf write ${baseaddr} 0x0 ${kernaddr}
uknor=${updatetool} ${baseaddr} uImage.${soc}; sf probe 0; sf erase ${kernaddr} ${kernsize}; sf write ${baseaddr} ${kernaddr} ${filesize}
updatetool=tftpboot
urnor=${updatetool} ${baseaddr} rootfs.squashfs.${soc}; sf probe 0; sf erase ${rootaddr} ${rootsize}; sf write ${baseaddr} ${rootaddr} ${filesize}
ver=U-Boot 2013.07-g90873ad (Jun 02 2024 - 14:59:49)

Environment size: 1408/65532 bytes
OpenIPC # run setnor16m
## Error: "setnor16m" not defined
OpenIPC #
`

RayPS, 2024-07-09:
一样的问题，请问大家解决了吗？
```
OpenIPC # run setnor16m
## Error: "sertnor16m" not defined
OpenIPC # <INTERRUPT>
OpenIPC # <INTERRUPT>
OpenIPC # <INTERRUPT>
OpenIPC # printenv 
baseaddr=0x21000000
baudrate=115200
bootargs=console=ttyS0,115200 panic=20 root=/dev/mtdblock3 init=/init mtdparts=NOR_FLASH:256k(boot),64k(env),2048k(kernel),${rootmtd}(rootfs),-(rootfs_data) LX_MEM=${memlx} mma_heap=mma_heap_name0,miu=0,sz=${memsz} cma=2M
bootcmd=setenv bootcmd ${cmdnor}; sf probe 0; saveenv; run bootcmd
bootdelay=0
cmdnor=sf probe 0; setenv setargs setenv bootargs ${bootargs}; run setargs; sf read ${baseaddr} ${kernaddr} ${kernsize}; bootm ${baseaddr}
ethact=sstar_emac
ethaddr=00:00:23:34:45:66
gatewayip=192.168.1.1
ipaddr=192.168.1.10
kernaddr=0x50000
kernsize=0x200000
memlx=0x3FE0000
memsz=0x2000000
netmask=255.255.255.0
rootaddr=0x250000
rootmtd=5120k
rootsize=0x500000
sdcard=setenv updatetool fatload mmc 0
serverip=192.168.1.1
soc=ssc337
stderr=serial
stdin=serial
stdout=serial
totalmem=64M
ubnor=sf probe 0; sf erase 0x0 ${kernaddr}; sf write ${baseaddr} 0x0 ${kernaddr}
uknor=${updatetool} ${baseaddr} uImage.${soc}; sf probe 0; sf erase ${kernaddr} ${kernsize}; sf write ${baseaddr} ${kernaddr} ${filesize}
updatetool=tftpboot
urnor=${updatetool} ${baseaddr} rootfs.squashfs.${soc}; sf probe 0; sf erase ${rootaddr} ${rootsize}; sf write ${baseaddr} ${rootaddr} ${filesize}
ver=U-Boot 2015.01 (Jun 02 2024 - 15:01:29)

Environment size: 1317/65532 bytes
OpenIPC # 
```

jpfree, 2024-07-16:
> 一样的问题，请问大家解决了吗？
>
> ```
> OpenIPC # run setnor16m
> ## Error: "sertnor16m" not defined
> OpenIPC # <INTERRUPT>
> OpenIPC # <INTERRUPT>
> OpenIPC # <INTERRUPT>
> OpenIPC # printenv
> baseaddr=0x21000000 [...]

找个老UBOOT刷入就可以了。