OpenIPC/firmware #1330: 打不开摄像机的 Web 界面问题

ljc111111111111111111111111111111, 2024-02-29:
已经成功烧录OpenIPC固件
[image]
，但打不开摄像机的 Web 界面
[image]
，我板子是T31ZX，sensor是GC2063，烧录上OpenIPC后其内sensor默认为GC2053，我不知道是sensor问题还是其他什么问题，如果是sensor问题，我将环境变量从原来GC2053改为GC2063，但打印ipcinfo依旧显示为GC2053，该怎么处理呢？希望大佬帮忙解决

ljc111111111111111111111111111111, 2024-03-01:
按您给的指令来一直卡住
[image]
端口换为80也不行
[image]

cronyx, 2024-03-01:
gc2063 = gc2053

ljc111111111111111111111111111111, 2024-03-01:
已经改回gc2053，也有驱动，
[image]
，但还是打不开web界面

ljc111111111111111111111111111111, 2024-03-01:
[image]

ljc111111111111111111111111111111, 2024-03-01:
应该是我还没有联网，OpenIPC有MT7862的WIFI驱动吗？还是说我mt7862的驱动编译进OpenIPC的固件中？

closed 2024-03-02 by ljc111111111111111111111111111111