OpenIPC/firmware #1333: Hi3518EV100 -  No Ethernet after flashing OpenIPC

ScreamingBeef, 2024-03-03:
Flashed the current available release for my SoC (Hi3518EV100) - whilst it boots okay, I am unable to get network access in uboot or OpenIPC. 

printenv in uboot, ifconfig in OpenIPC show expected values for network config. It appears that no packets are exchanged on the network, so I am suspecting network is not working at the lowest levels. On the network port itself, activity and link lights are lit, steady. 

This is my first time exploring an embedded system so am unsure where to start debugging. Happy to provide as much info as I can.

thegroove, 2024-03-03:
Can try this: https://github.com/OpenIPC/wiki/blob/master/en/trouble-network.md

ScreamingBeef, 2024-03-03:
Ah thank you, I feel a bit silly for not coming across that myself!

I'll test and report back

ScreamingBeef, 2024-03-03:
Tried the options in the network trouble page with no success. Should I raise this in Telegram instead of as a github issue?

NekoB0x, 2024-03-03:
Had the same issue, used stock uBoot as a workaround (stock has 256k env partition so the layout is different), used a guide from [HERE](https://github.com/straga/security_cam/blob/master/HI3518ev100-OpenIPC.md#flash-over-tftp)

Copying env variables from stock didn't work either:
phyaddru=0
phyaddrd=1
mdio_intf=rmii

Interestingly ipctool shows:
ethernet:
  u-mdio-phyaddr: 1
  phy-id: 0xffffffff
  d-mdio-phyaddr: 0
  phy-mode: mii

ScreamingBeef, 2024-03-03:
Thanks for the guide, I'll work through it this evening. Might try SPI to write a clean OpenIPC image too, i've made a lot of changes during testing and it's possible I've created my own conflicts.

shifttymike, 2025-01-07:
I hit the same issue on a HI3518ERBCV100 (hi3518ev100) and the following params worked for me:
`hieth.mdioifu=0 hieth.mdioifd=0 hieth.phyaddru=1 hieth.phyaddrd=2`

viniciusvillas, 2025-04-09:
Thank you so much, @shifttymike!! It also worked for me (HI3518CV100)!
If you don't mind me asking, how did you get to these parameters? I mean, the values for each one? Trial and error?
I'm currently compiling a "ultimate" version for my camera (a ZModo ZP-IBI13), since it has 16MB of RAM but there's no precompiled package for it - I just made a new configuration based on the HI3518EV200 ultimate configuration.

closed 2026-11-16 by openipc-ai