OpenIPC/firmware #1344: WIP: Add support for RTL8812EU-based Wi-Fi adapters for FPV firmware

libc0607, 2024-03-12:
Hi. 

The Chinese Wi-Fi module vendor LB-LINK had recently released a high-power Wi-Fi module BL-M8812EU2 based on RTL8812EU chipset costs only ~$10 (68 CNY). Here's the [product page](https://www.lb-link.com/product_36_183.html) and the [purchase link](https://item.taobao.com/item.htm?id=764510955987). The FEM chip inside is [KCT8528HE](https://datasheet5.oss-cn-shanghai.aliyuncs.com/datasheets/kxcomtech/KCT8528HE.pdf), with an impressive +29dBm (802.11g, 6Mbps, -9dB EVM) PA output and a +15.5dB (NF=1.7dB) LNA. 

I bought two a few weeks ago, and the tech support from LB-LINK send me a driver tar from Realtek. It seems that the chip is gonna be the replacement of RTL8812AU which has been marked as EOL. The original driver does not support packet injection with correct rate or MCS index, so I tried to merge the "injection patch" from svpcom/rtl8812au to this driver, which can be found at [here](https://github.com/libc0607/rtl88x2eu-20230815). 

The progress now is that the driver build good on kernel 6.5, 802.11g/n packets can be sent at the correct rate/MCS using my own packet injecting tool, and video transmission using wfb_tx and wfb_rx in my two virtual machines (see [here](https://www.youtube.com/watch?v=mWpVyfZnKYI) in youtube). The OpenIPC package makefile is avaliable at [here](https://github.com/libc0607/openipc-firmware/commit/cc990c07cc367915b74f74e87f02f199dfba2ac8).

The vendor also claimed that RTL8812EU supports 10MHz bandwidth, but there's no document and I don't know how to use it. The driver contains [a firmware](https://github.com/libc0607/rtl88x2eu-20230815/blob/v5.15.0.1/hal/rtl8822e/hal8822e_fw_10M.c) which seems to be the "10MHz firmware", but I don't have a ~6GHz spectrum analyzer either so I can't tell if it's really working -- It would be great if anyone could help.

I'm choosing to post it here now because this driver definitely needs more integration testing. If someone is also interested in this module, he can just start from here and does not need to patch the driver or write Makefile again. 

Of course I'll open a pull request when it becomes somehow usable and stable.

libc0607, 2024-04-02:
> @libc0607 Seems like nobody cares about Windows when it comes to WifiBroadcasting... Will it somehow be possible to make a Windows Driver also? Would love to try it out and recieve WifiBroadcasting to OBS Studio

Theoretically it's possible but I don't think anybody gonna do it. It requires a lot of work and also support from Realtek which is obviously impossible. 

You can try using an NVR (e.g. see Hi3536DV100 configs in this repo; or anything similar) as a video stream forwarder. It will act as an "abstraction layer" in the system that converts different Wi-Fi adaptors to standard ethernet UDP packets and then send them to ur Windows PC via a RJ45 connector.

widgetii, 2024-04-02:
> @libc0607 Seems like nobody cares about Windows when it comes to WifiBroadcasting... Will it somehow be possible to make a Windows Driver also? Would love to try it out and recieve WifiBroadcasting to OBS Studio

There is such driver https://github.com/buldo/WiFiDriver, but you're right that nobody cares about Windows for a long time

svpcom, 2024-04-02:
@libc0607 Is it possible to buy it on aliexpress? Are any other cards with this chipset exists on the market?

ETE-Design, 2024-04-02:
> There is such driver https://github.com/buldo/WiFiDriver, but you're right that nobody cares about Windows for a long time

Thanks... Do you know how to get it work with OpenIPC, and how to access the Streaming on a Windows PC?

widgetii, 2024-04-02:
None of OpenIPC's devs are using Windows now

libc0607, 2024-04-02:
> @libc0607 Is it possible to buy it on aliexpress? 

I thought there would be, but I couldn't find any after I registered. Maybe you can ask some of the BL-R8812AF1 sellers since the modules are from same vendor.
And I'm gonna ask their product manager tomorrow if he knows there's any reseller outside mainland China. 

> Are any other cards with this chipset exists on the market?

I'm afraid not. No search results in zh-cn internet either.

ETE-Design, 2024-04-02:
> None of OpenIPC's devs are using Windows now

So my best shoot will be to ask @buldo, or what do you think?

svpcom, 2024-04-02:
My point is that I want to invest my time only to products that are available on mass market.
It is possible to add injection to most of SoftMAC drivers, but hardware should be easy to buy and it should not be EOL at least for 5 years. Also it will be very good if vendor provides docs for internal card registers (like atheros done for ath9k). Without these specs it is very hard to add features needed for injection (for example: disabling acks, use rts/cts packet, disabling csma/ca, extended channel monitoring for noise and others)

> I thought there would be, but I couldn't find any after I registered. Maybe you can ask some of the BL-R8812AF1 sellers since the modules are from same vendor. And I'm gonna ask their product [...]

buldo, 2024-04-02:
@libc0607 if you want to receive on windows there are much better ways than my driver:
1. Use dedicated sbc for RX. Connect this sbc to windows via ethernet. Even $5 milk-v is enough. 
2. Use WSL with custom kernel and usb-ip pass-through. It enough to receive video stream.

ETE-Design, 2024-04-02:
> Theoretically it's possible but I don't think anybody gonna do it. It requires a lot of work and also support from Realtek which is obviously impossible.
>
> You can try using an NVR (e.g. see [...]

If using the NVR will I then need one NVR for each stream I like to recieve, or is it possible to recieve multiple streams on one NVR?

svpcom, 2024-04-02:
@libc0607 Could you upload datasheet for RTL8812EU here?

libc0607, 2024-04-03:
> @libc0607 Could you upload datasheet for RTL8812EU here?

BL-M8812EU2 (the module) datasheet: 
[BL-M8812EU2_datasheet_V1.0.1.0_231027(70003939).pdf](https://github.com/OpenIPC/firmware/files/14846490/BL-M8812EU2_datasheet_V1.0.1.0_231027.70003939.pdf)

There's no RTL8812EU-CG chip datasheet (at least for now).   
It's fairly new and the module has been selling in China for only 2 months ... leakers need some time XD

libc0607, 2024-04-03:
> If using the NVR will I then need one NVR for each stream I like to recieve, or is it possible to recieve multiple streams on one NVR?

Maybe you can try multiple Wi-Fi adaptors on one NVR?

ETE-Design, 2024-04-03:
> @libc0607 if you want to receive on windows there are much better ways than my driver:
> 
> 1. Use dedicated sbc for RX. Connect this sbc to windows via ethernet. Even $5 milk-v is enough.
> 2. Use WSL with custom kernel and usb-ip pass-through. It enough to receive video stream.

But with your driver you will get lowest latency cause you cut out one lead, or am I wrong?
Would prefer to use as clean setup as possible.
Also do you know if it is possible to recieve multiple broadcasting with only one receiver, or do I need one receiver for  each sender?

If using WSL is it then possible to push the signal to a Windows OBS Studio?

libc0607, 2024-04-04:
Driver update: 

The TX power setting of RTL8812EU is working. The total power consumption of the adaptor came to ~10W (5V/2.0A).   
Just quickly went through the code and I found that the setting is a bit different compared with RTL8812AU.   

Simply add ```rtw_tx_pwr_by_rate=0 rtw_tx_pwr_lmt_enable=0``` when ```insmod```,   
then the TX power can be directly set by ```iw dev wlan0 set txpower fixed <mBm>```.   
The range of <mBm> is 0~3150, which equals 50 times driver_txpower_override. 

Another update is about regulation. Add ```rtw_regd_src=1``` when ```insmod``` and the phy won't claim it's "self-managed", then you can use the channel you want.

hwanggz0, 2024-04-15:
here an 8812eu driver with monitor mode and injection: https://github.com/hwanggz0/rtl8812eu/

closed 2024-04-16 by libc0607

lida2003, 2024-12-16:
> then the TX power can be directly set by `iw dev wlan0 set txpower fixed <mBm>`.
> The range of is 0~3150, which equals 50 times driver_txpower_override.

```
There are two ways to set TX power.   
- ```driver_txpower_override``` in ```/etc/wfb.conf```. The range is ```0~63```
- ```iw dev <wlan0> set txpower fixed <mBm>```. The range is ```0~3150```, and can be set dynamically when transmitting.
```

Hi @libc0607 Where did you get those numbers ```0~63```  ```0~3150```, I didn't find in the spec([BL-M8812EU2_datasheet_V1.0.1.0_231027(70003939).pdf](https://github.com/OpenIPC/firmware/files/14846490/BL-M8812EU2_datasheet_V1.0.1.0_231027.70003939.pdf)) you provided.?

libc0607, 2024-12-16:
> Hi @libc0607 Where did you get those numbers `0~63` `0~3150`, I didn't find in the spec([BL-M8812EU2_datasheet_V1.0.1.0_231027(70003939).pdf](https://github.com/OpenIPC/firmware/files/14846490/BL-M8812EU2_datasheet_V1.0.1.0_231027.70003939.pdf)) you provided.?

It's an internal value in the driver. 
In most situations, the user does not need to consider the value -- TX power is calibrated by LB-LINK and then written into eFuse. But we override that (to unlock a bit).

svpcom, 2024-12-16:
@libc0607 Does I mean right that using `iw dev <wlan0> set txpower fixed <mBm>` is more preferred than `driver_txpower_override` because it abstract from card internals and operate real power values? Or this mBm value is also not real tx power value?

libc0607, 2024-12-16:
> @libc0607 Do I mean right that using `iw dev <wlan0> set txpower fixed <mBm>` is more preferred than `driver_txpower_override` because it abstract from card internals and operate real power values? Or this mBm value is also not real tx power value?

Both ```driver_txpower_override``` and ```mBm``` are not real dBm values. 

In OpenIPC, if set txpower by ```driver_txpower_override```, it will be converted to ```<mBm>``` (multiplied by 50). It was some ... backward compatible solution, to avoid adding the third txpower setting in ```/etc/wfb.conf``` to confuse users. ( 

The default eFuse value (table) is calibrated by LB-LINK to match the real dBm power in the module datasheet. (The values even differ in different center frequencies.) That's the only connection to the real world...

For example, I know someone tested RTL8812CU output:
https://github.com/JasonFreeLab/rtl8812cu?tab=readme-ov-file#ant0_5825mhz_20m_11n_mcs1_mbm1700

svpcom, 2024-12-16:
@libc0607 Is it possible to add some module parameter to the driver to extract these calibration values? Maybe it can help to solve 40MHz injection problem

libc0607, 2024-12-16:
> @libc0607 Is it possible to add some module parameter to the driver to extract these calibration values? Maybe it can help to solve 40MHz injection problem

If I understand it correctly, the txpower table can be read out by ```cat /proc/net/rtl88x2eu/wlan0/txpwr_total_dbm``` (after ```insmod```, before any ```iw set txpower```). 
```iw wlan0 set txpower fixed <mBm>``` will just overwrite all values in the table.

And ```iw wlan0 info``` shows the biggest value in this table.

lida2003, 2024-12-16:
> In most situations, the user does not need to consider the value -- TX power is calibrated by LB-LINK and then written into eFuse.

> If I understand it correctly, the default txpower table can be read out by `cat /proc/net/rtl88x2eu/wlan0/txpwr_total_dbm` (after `insmod`, before any `iw set txpower`).

I have 8812AU wifi card, just plugin, wfb-ng not started.
There is no such `/proc/net/rtl88x2...` dir. And I'm sure the driver is loaded, which can tell from `dmesg` and `lsmod | grep 88XXau_wfb`.

The reason I previously asked this question was to better understand and obtain manufacturer-calibrated data, which might be more linear or make it easier to [comprehend the adaptive link's profile configuration table](https://github.com/sickgreg/OpenIPC-Adaptive-Link/issues/10).

libc0607, 2024-12-16:
> I have 8812AU wifi card, just plugin, wfb-ng not started. There is no such `/proc/net/rtl88x2...` dir. And I'm sure the driver is loaded, which can tell from `dmesg` and `lsmod | grep 88XXau_wfb`. [...]

The current 8812au driver is based on Realtek driver v5.2.20. Realtek had not implemented ```/proc``` debug APIs at that time. ( 
Maybe you can read the eFuse and find the value? There are some documents about 8812au efuse in svpcom's 8812au repo.

As for TX power backoff -- there are some empirical values for different MCS indexes, e.g. -3dB from P1dB for MCS0, and -8dB for MCS7

lida2003, 2024-12-17:
I got this from 8812EU just right after `modprobe`.  Right now, I don't have any idea about these numbers.

@libc0607 How to understand the relationship between the parameters and the command `iw dev <wlan0> set txpower fixed <mBm>`?

- mBm: 0 ~ 3150

```
# cat /proc/net/rtl88x2eu/wlan0/txpwr_total_dbm
antenna_gain:     0
20MHz, cch20:0
rate         target = min((byr    + btc    + extra ), rlmt  , lmt   , ulmt  ) + tpc
   CCK_1M 1T     18 =    ((    18        0        0),     NA,     NA,     NA)        0
   CCK_2M 1T     18 =    ((    18        0        0),     NA,     NA,     NA)        0
  CCK5_5M 1T     18 =    ((    18        0        0),     NA,     NA,     NA)        0
  CCK_11M 1T     18 =    ((    18        0        0),     NA,     NA,     NA)        0
  OFDM_6M 1T     17 =    ((    17        0        0),     NA,     NA,     NA)        0
  OFDM_9M 1T     17 =    ((    17        0        0),     NA,     NA,     NA)        0
 OFDM_12M 1T     17 =    ((    17        0        0),     NA,     NA,     NA)        0
 OFDM_18M 1T     17 =    ((    17        0        0),     NA,     NA,     NA)        0
 OFDM_24M 1T     17 =    ((    17        0        0),     NA,     NA,     NA)        0
 OFDM_36M 1T     17 =    ((    17        0        0),     NA,     NA,     NA)        0
 OFDM_48M 1T     17 =    ((    17        0        0),     NA,     NA,     NA)        0
 OFDM_54M 1T     17 =    ((    17        0        0),     NA,     NA,     NA)        0
     MCS0 1T     16 =    ((    16        0        0),     NA,     NA,     NA)        0
     MCS1 1T     16 =    ((    16        0        0),     NA,     NA,     NA)        0
     MCS2 1T     16 =    ((    16        0        0),     NA,     NA,     NA)        0
     MCS3 1T     16 =    ((    16        0        0),     NA,     NA,     NA)        0
     MCS4 1T     16 =    ((    16        0        0),     NA,     NA,     NA)        0
     MCS5 1T     16 =    ((    16        0        0),     NA,     NA,     NA)        0
     MCS6 1T     16 =    ((    16        0        0),     NA,     NA,     NA)        0
     MCS7 1T     16 =    ((    16        0        0),     NA,     NA,     NA)        0
     MCS8 2T  19.01 =    (( 19.01        0        0),     NA,     NA,     NA)        0
     MCS9 2T  19.01 =    (( 19.01        0        0),     NA,     NA,     NA)        0
    MCS10 2T  19.01 =    (( 19.01        0        0),     NA,     NA,     NA)        0
    MCS11 2T  19.01 =    (( 19.01        0        0),     NA,     NA,     NA)        0
    MCS12 2T  19.01 =    (( 19.01        0        0),     NA,     NA,     NA)        0
    MCS13 2T  19.01 =    (( 19.01        0        0),     NA,     NA,     NA)        0
    MCS14 2T  19.01 =    (( 19.01        0        0),     NA,     NA,     NA)        0
    MCS15 2T  19.01 =    (( 19.01        0        0),     NA,     NA,     NA)        0
VHT1SMCS0 1T     16 =    ((    16        0        0),     NA,     NA,     NA)        0
VHT1SMCS1 1T     16 =    ((    16        0        0),     NA,     NA,     NA)        0
VHT1SMCS2 1T     16 =    ((    16        0        0),     NA,     NA,     NA)        0
VHT1SMCS3 1T     16 =    ((    16        0        0),     NA,     NA,     NA)        0
VHT1SMCS4 1T     16 =    ((    16        0        0),     NA,     NA,     NA)        0
VHT1SMCS5 1T     16 =    ((    16        0        0),     NA,     NA,     NA)        0
VHT1SMCS6 1T     16 =    ((    16        0        0),     NA,     NA,     NA)        0
VHT1SMCS7 1T     16 =    ((    16        0        0),     NA,     NA,     NA)        0
VHT1SMCS8 1T     16 =    ((    16        0        0),     NA,     NA,     NA)        0
VHT1SMCS9 1T     16 =    ((    16        0        0),     NA,     NA,     NA)        0
VHT2SMCS0 2T  19.01 =    (( 19.01        0        0),     NA,     NA,     NA)        0
VHT2SMCS1 2T  19.01 =    (( 19.01        0        0),     NA,     NA,     NA)        0
VHT2SMCS2 2T  19.01 =    (( 19.01        0        0),     NA,     NA,     NA)        0
VHT2SMCS3 2T  19.01 =    (( 19.01        0        0),     NA,     NA,     NA)        0
VHT2SMCS4 2T  19.01 =    (( 19.01        0        0),     NA,     NA,     NA)        0
VHT2SMCS5 2T  19.01 =    (( 19.01        0        0),     NA,     NA,     NA)        0
VHT2SMCS6 2T  19.01 =    (( 19.01        0        0),     NA,     NA,     NA)        0
VHT2SMCS7 2T  19.01 =    (( 19.01        0        0),     NA,     NA,     NA)        0
VHT2SMCS8 2T  19.01 =    (( 19.01        0        0),     NA,     NA,     NA)        0
VHT2SMCS9 2T  19.01 =    (( 19.01        0        0),     NA,     NA,     NA)        0

```

libc0607, 2024-12-18:
> I got this from 8812EU just right after `modprobe`. Right now, I don't have any idea about these numbers.
> 
> @libc0607 How to understand the relationship between the parameters and the command `iw dev <wlan0> set txpower fixed <mBm>`?

The chipset needs a table to find the corresponding TX power index for different modulations (CCK/OFDM, MCS index, VHT NSS, etc.) because of the difference in the Peak-to-Average Power Ratio (PAPR). 

The first column equals <mBm> you set divided by 100. So, the second table shows that ```driver_txpower_override``` in ```/etc/wfb.conf``` is 50.  

> EDIT2: I'm very interested in RTL8812CU output, which gaves freq/bandwidth/modulation/mbm/evm https://github.com/JasonFreeLab/rtl8812cu?tab=readme-ov-file#ant0_5825mhz_20m_11n_mcs1_mbm1700
> 
> Is there any 8812 EU test output, like RTL8812CU?

... Don't know. I'm also curious about that, will keep focusing.
The Chinese fork of PixelPilot "DigitalFPV牛逼版" gives RX EVM on OSD, which can be used to evaluate PA saturation.

(BTW I've heard that the RunCam tested the RTL8812EU before selling their WiFiLink, but they recommended txpower=58, which will definitely cause PA saturation and TX deterioration. So I'm not trusting that)