OpenIPC/firmware #1352: Hi3518CV100: I²C not working

jp112sdl, 2024-03-17:
Hey guys!

I have an old ZMODO ZP-IBH15-S camera with a 3518CV100 and (possibly) a JXH22

Running `ipctool` on the **original** firmware shows this output:
```
# ./ipctool
---
chip:
  vendor: HiSilicon
  model: 3518CV100
board:
  possible-IR-cut-GPIO: 18
...
firmware:
  u-boot: "2010.06 (Nov 15 2013 - 19:54:17)"
  kernel: "3.0.8 (Wed Apr 10 15:47:48 CST 2013)"
  toolchain: gcc version 4.4.1 (Hisilicon_v100(gcc4.4-290+uclibc_0.9.32.1+eabi+linuxpthread))
  libc: uClibc 0.9.32.1
  sdk: "Hi3518_MPP_V1.0.7.0 Debug (Sep 23 2013, 17:48:02)"
  main-app: ./App3518
sensors:
- vendor: Silicon Optronics
  model: JXH22
  control:
    bus: 0
    type: i2c
    addr: 0x60
  data:
    type: DC
  clock: 24MHz

# ./ipctool i2cdump 0x60 0x00 0xff
       0  1  2  3  4  5  6  7   8  9  A  B  C  D  E  F
    : 00 F6 00 FF FF FF FF FF  FF FF A0 22 40 00 1D 0B  |  ..........."@...
  10: 26 80 30 87 80 25 80 20  D5 10 80 4F 21 FF 9F 00  |  &.0..%. ...O!...
  20: 72 06 FF 02 00 E0 25 E9  0D 00 D4 10 00 08 C4 20  |  r.....%........
  30: 90 10 10 10 32 00 40 4A  70 70 70 70 70 70 70 70  |  ....2.@Jpppppppp
  40: 70 70 70 70 70 70 70 10  40 1A 03 E3 E1 E1 E3 55  |  ppppppp.@......U
  50: 55 55 55 55 55 55 55 55  55 55 55 55 55 55 55 55  |  UUUUUUUUUUUUUUUU
  60: 28 00 11 00 15 40 A0 31  0C 74 41 00 90 02 80 80  |  (....@.1.tA.....
  70: 49 8A 68 33 B8 2B 40 06  14 14 14 14 14 14 14 14  |  I.h3.+@.........
  80: 01 01 01 01 01 01 01 01  01 01 01 01 01 01 01 01  |  ................
  90: 01 01 01 01 01 01 01 01  01 01 01 01 01 01 01 01  |  ................
  a0: 01 01 01 01 01 01 01 01  01 01 01 01 01 01 01 01  |  ................
  b0: 01 01 01 01 01 01 01 01  01 01 01 01 01 01 01 01  |  ................
  c0: 0A 0A 0A 0A 0A 0A 0A 0A  0A 0A 0A 0A 0A 0A 0A 0A  |  ................
  d0: 0A 0A 0A 0A 0A 0A 0A 0A  0A 0A 0A 0A 0A 0A 0A 0A  |  ................
  e0: 0A 0A 0A 0A 0A 0A 0A 0A  0A 0A 0A 0A 0A 0A 0A 0A  |  ................
  f0: 0A 0A 0A 0A 0A 0A 0A 0A  0A 0A 0A 0A 0A 0A 0A     |  ............... 
  ```

After flashing the **OpenIPC** firmware it shows
```
---
chip:
  vendor: HiSilicon
  model: 3518CV100
board:
  vendor: OpenIPC
  version: 2.4.03.16
...
firmware:
  u-boot: "2010.06 (Nov 14 2022 - 19:25:29)"
  kernel: "3.0.8 (Sat Mar 16 13:49:40 UTC 2024)"
  toolchain: buildroot-gcc-12.3.0
  sdk: "Hi3518_MPP_V1.0.B.0  (Nov 19 2015, 16:37:04)"
sensors:
- vendor: TechPoint
  model: TPffff
  control:
    bus: 0
    type: i2c
    addr: 0x88
  vicap-state: down 
```

So now the sensor points to i2c address **0x88** instead of **0x60** !?

When I try to run `i2cdump` within the OpenIPC firmware, there is only: 

```
ipctool i2cdump 0x88 0x00 0xff
       0  1  2  3  4  5  6  7   8  9  A  B  C  D  E  F
    : FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF  |  ................
  10: FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF  |  ................
  20: FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF  |  ................
  30: FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF  |  ................
  40: FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF  |  ................
  50: FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF  |  ................
  60: FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF  |  ................
  70: FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF  |  ................
  80: FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF  |  ................
  90: FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF  |  ................
  a0: FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF  |  ................
  b0: FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF  |  ................
  c0: FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF  |  ................
  d0: FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF  |  ................
  e0: FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF  |  ................
  f0: FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF     |  ...............
FF  |  ................  
```

even on address 0x60 (where the sensor was found within the original firmware):
```
ipctool i2cdump 0x60 0x00 0xff
       0  1  2  3  4  5  6  7   8  9  A  B  C  D  E  F
    : FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF  |  ................
  10: FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF  |  ................
  20: FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF  |  ................
  30: FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF  |  ................
  40: FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF  |  ................
  50: FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF  |  ................
  60: FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF  |  ................
  70: FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF  |  ................
  80: FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF  |  ................
  90: FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF  |  ................
  a0: FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF  |  ................
  b0: FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF  |  ................
  c0: FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF  |  ................
  d0: FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF  |  ................
  e0: FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF FF  |  ................
  f0: FF FF FF FF FF FF FF FF  FF FF FF FF FF FF FF     |  ...............
FF  |  ................  
```

`i2cdetect -l` only shows `i2cdetect: can't open '/sys/class/i2c-dev': No such file or directory `

The only place I can find something with i2c is in `/dev/hi_i2c`.

And here is the output of `lsmod`:
```
lsmod
Module                  Size  Used by    Tainted: P
gpioi2c_ex              5926  0
gpioi2c                 3018  0
hi3518_isp             26916  1
hi3518_base            44314  1 hi3518_isp
mmz                    19773  1 hi3518_base
vfat                    8052  0
fat                    40796  1 vfat
```

All in all, majestic is not working and reports the following error continously:
```
20:46:54  <       majestic> [     sdk] venc_read@903                 Timeout get any data from venc channel 0
20:46:54  <       majestic> [     sdk] venc_read@923                 NOTE: Current frame is NULL! 
```

Are there any ideas, what the problem is or are there any suggestions for further debugging?
Thanks!

impala3525, 2024-03-17:
I was having same issue with goke I had t change BlkCNT to 4 mine was set to 1 for some reason and 1 only works for ingenics. 4 is for Hisi/Goke

jp112sdl, 2024-03-18:
>  had t change BlkCNT to 4 mine was set to 1 for some reason and 1 only works for ingenics

@impala3525 _Where_ can I set the `BlkCNT` value?

impala3525, 2024-03-18:
Majestic>ISP>Block Count

Andrii2313, 2024-04-26:
I have a similar problem. I tried BlkCnt from 1 to 10 and no result. How can I find out the correct value?

Andrii2313, 2024-04-26:
The original firmware was damaged and the camera did not start. After installing OpenIpc, the ipctool utility produced the following result
`root@openipc-hi3518cv100:~# ipctool
The ipctool installed as remote GitHub plugin
---
chip:
  vendor: HiSilicon
  model: 3518CV100
board:
  vendor: OpenIPC
  version: 2.4.04.24
  possible-IR-cut-GPIO: 0,1
ethernet:
  mac: "36:bd:b7:f5:94:8d"
  u-mdio-phyaddr: 1
  phy-id: 0xffffffff
  d-mdio-phyaddr: 0
  phy-mode: mii
rom:
- type: nor
  block: 64K
  partitions:
    - name: boot
      size: 0x40000
      sha1: d359fac3
    - name: env
      size: 0x10000
      sha1: 31f83bef
      contains:
        - name: uboot-env
          offset: 0x0
    - name: kernel
      size: 0x300000
      sha1: 357bb792
    - name: rootfs
      size: 0xa00000
      path: /,squashfs
      sha1: ae52dfdc
    - name: rootfs_data
      size: 0x2b0000
      path: /overlay,jffs2,rw
  size: 16M
  addr-mode: 3-byte
ram:
  total: 128M
  media: 89M
firmware:
  u-boot: "2010.06 (Nov 14 2022 - 19:25:29)"
  kernel: "3.0.8 (Wed Apr 24 02:16:32 UTC 2024)"
  toolchain: buildroot-gcc-13.2.0
  sdk: "Hi3518_MPP_V1.0.B.0  (Nov 19 2015, 16:37:04)"
sensors:
- vendor: ON Semiconductor
  model: AR0130
  control:
    bus: 0
    type: i2c
    addr: 0x20
  vicap-state: down
`

Andrii2313, 2024-04-26:
at address 0x20 this is the result
`root@openipc-hi3518cv100:~# /utils/ipctool i2cdump 0x20 0x00 0xff
       0  1  2  3  4  5  6  7   8  9  A  B  C  D  E  F
    : 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |  ................
  10: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |  ................
  20: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |  ................
  30: 00 1F 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |  ................
  40: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |  ................
  50: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |  ................
  60: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |  ................
  70: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |  ................
  80: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |  ................
  90: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |  ................
  a0: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |  ................
  b0: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |  ................
  c0: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |  ................
  d0: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |  ................
  e0: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |  ................
  f0: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |  ................
`

closed 2026-11-16 by openipc-ai