OpenIPC/firmware #1363: hi3536dv100—OSD delay in video

ZCSHANLIN, 2024-03-28:
NVR(hi3536dv100)use the latest firmware, the OSD in the video will freeze. The old version of the firmware will not.

https://github.com/OpenIPC/firmware/assets/165228626/2eb713ed-41aa-464f-ab64-22d95554ce38

Qqww4599, 2024-06-22:
I have the same issue. I use hi3536dv100 NVR as GS, and hi3536dv100 as Drone side. The latency about 2~3 secs. Firmware is latest fw.

widgetii, 2026-10-20:
Found it, and it is fixed in vdec: OpenIPC/research#58.

**Cause.** vdec's MAVLink thread slept for `usleep(1)` after every UDP datagram. The Hi3536DV100 kernel runs at HZ=100 without high-resolution timers, so that "1 µs" is a full 10 ms tick, and vdec could parse at most 100 MAVLink messages per second. Each message arrives in its own datagram. As soon as the air side sends more than that, the socket buffer fills up and two things happen:
- The OSD shows telemetry that is 1–2+ seconds old.
- Datagrams get dropped, so some fields freeze and update only now and then. That is the "updated slowly" part.

A Raspberry Pi has high-resolution timers, which is why a Pi ground station on the same link looks fine. There was a second, smaller stall too: while RC channel 8 was above 1700, vdec ran `/root/resolution.sh` on every RC message. It now runs only when the switch flips.

Measured on a Hi3536DV100 NVR, with MAVLink sent one message per datagram:

| messages/s | before | after |
|---|---|---|
| 100 | 22 ms | 11 ms |
| 200 | 1.0–2.0 s, drops | 11 ms |
| 400 | 0.9–2.3 s, drops | 11 ms |
| 1000 | – | 3 ms |

The firmware now pins vdec to this commit (#2642). Any image that includes vdec and is built from master after that merge gets the fix; on the ground-station side that is builder's hi3536dv100 wfbng device. Until you have such an image, you can work around it by keeping the total telemetry rate from the flight controller under ~100 messages/s, for example by lowering the MAVLink stream rates.

This was tested with synthetic telemetry on a LAN, not with a real air unit over wfb-ng. If one of you can try a build with the new vdec on your setup and confirm, that would close this out.